OpenDroneMap/ODM #1385: problem on windows10

tangjj110, 2021-12-19:
### How did you install OpenDroneMap? (Docker, natively, ...)?

ODM_Setup_2.6.7.exe

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

windows10

### What is the problem?
ODM can be installed natively on Windows,but I I have a problem when i tried it.

[INFO]    Initializing ODM - Sun Dec 19 23:54:17  2021
[INFO]    ==============
[INFO]    auto_boundary: False
[INFO]    boundary: {}
[INFO]    build_overviews: False
[INFO]    camera_lens: auto
[INFO]    cameras: {}
[INFO]    cog: False
[INFO]    copy_to: None
[INFO]    crop: 3
[INFO]    debug: False
[INFO]    dem_decimation: 1
[INFO]    dem_euclidean_map: False
[INFO]    dem_gapfill_steps: 3
[INFO]    dem_resolution: 5
[INFO]    depthmap_resolution: 640
[INFO]    dsm: False
[INFO]    dtm: False
[INFO]    end_with: odm_postprocess
[INFO]    fast_orthophoto: False
[INFO]    feature_quality: high
[INFO]    feature_type: sift
[INFO]    force_gps: False
[INFO]    gcp: None
[INFO]    geo: None
[INFO]    gps_accuracy: 10
[INFO]    ignore_gsd: False
[INFO]    matcher_distance: 0
[INFO]    matcher_neighbors: 8
[INFO]    matcher_type: flann
[INFO]    max_concurrency: 24
[INFO]    merge: all
[INFO]    mesh_octree_depth: 11
[INFO]    mesh_size: 200000
[INFO]    min_num_features: 8000
[INFO]    name: test
[INFO]    optimize_disk_space: False
[INFO]    orthophoto_compression: DEFLATE
[INFO]    orthophoto_cutline: False
[INFO]    orthophoto_kmz: False
[INFO]    orthophoto_no_tiled: False
[INFO]    orthophoto_png: False
[INFO]    orthophoto_resolution: 5
[INFO]    pc_classify: False
[INFO]    pc_csv: False
[INFO]    pc_ept: False
[INFO]    pc_filter: 2.5
[INFO]    pc_geometric: False
[INFO]    pc_las: False
[INFO]    pc_quality: medium
[INFO]    pc_rectify: False
[INFO]    pc_sample: 0
[INFO]    pc_tile: False
[INFO]    primary_band: auto
[INFO]    project_path: /
[INFO]    radiometric_calibration: none
[INFO]    rerun: None
[INFO]    rerun_all: False
[INFO]    rerun_from: None
[... 241 lines cut ...]
    self.next_stage.run(outputs)
  [Previous line repeated 2 more times]
  File "D:\ODM\opendm\types.py", line 327, in run
    self.process(self.args, outputs)
  File "D:\ODM\stages\odm_filterpoints.py", line 39, in process
    point_cloud.filter(inputPointCloud, tree.filtered_point_cloud,
  File "D:\ODM\opendm\point_cloud.py", line 176, in filter
    run_pipeline(pipeline, verbose=verbose)
  File "D:\ODM\opendm\dem\pdal.py", line 161, in run_pipeline
    system.run(' '.join(cmd))
  File "D:\ODM\opendm\system.py", line 106, in run
    raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 1

===== Done, human-readable information to follow... =====

[ERROR]   Uh oh! Processing stopped because of strange values in the reconstruction. This is often a sign that the input data has some issues or the software cannot deal with it. Have you followed best practices for data acquisition? See https://docs.opendronemap.org/flying/

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

none

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

none

pierotofy, 2021-12-19:
>  If possible please include a copy of your dataset uploaded on Google Drive or Dropbox.

pierotofy, 2021-12-23:
We cannot fix this unless we can access a dataset. Can you share a dataset?

pierotofy, 2022-01-04:
Closing since a dataset cannot be accessed and that code path has been removed in `2.7.0` anyway (we now invoke the `FPCFilter` module for odm_filterpoints). 🙏

closed 2022-01-04 by pierotofy