OpenDroneMap/ODM #1403: Possible Bug: SIFT_GPU processing seems to vail (v1.9.11 B49)

Saijin-Naib, 2022-01-13:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Windows Native

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

NA

### What is the problem?

```
[INFO]    Fast orthophoto is turned on, automatically setting --skip-3dmodel
[INFO]    Initializing ODM 2.7.0 - Wed Jan 12 19:18:11  2022
[INFO]    ==============
[INFO]    auto_boundary: True
[INFO]    boundary: {}
[INFO]    build_overviews: False
[INFO]    camera_lens: auto
[INFO]    cameras: {}
[INFO]    cog: True
[INFO]    copy_to: None
[INFO]    crop: 3
[INFO]    debug: False
[INFO]    dem_decimation: 1
[INFO]    dem_euclidean_map: False
[INFO]    dem_gapfill_steps: 3
[INFO]    dem_resolution: 5
[INFO]    depthmap_resolution: 640
[INFO]    dsm: False
[INFO]    dtm: False
[INFO]    end_with: odm_postprocess
[INFO]    fast_orthophoto: True
[INFO]    feature_quality: high
[INFO]    feature_type: sift
[INFO]    force_gps: False
[INFO]    gcp: None
[INFO]    geo: None
[INFO]    gps_accuracy: 10
[INFO]    ignore_gsd: False
[INFO]    matcher_neighbors: 8
[INFO]    matcher_type: flann
[INFO]    max_concurrency: 6
[INFO]    merge: all
[INFO]    mesh_octree_depth: 11
[INFO]    mesh_size: 200000
[INFO]    min_num_features: 8000
[INFO]    name: 951218a1-b597-450e-8144-5ecd979cbeda
[INFO]    optimize_disk_space: False
[INFO]    orthophoto_compression: DEFLATE
[INFO]    orthophoto_cutline: False
[INFO]    orthophoto_kmz: False
[INFO]    orthophoto_no_tiled: False
[INFO]    orthophoto_png: False
[INFO]    orthophoto_resolution: 5
[INFO]    pc_classify: False
[INFO]    pc_csv: False
[INFO]    pc_ept: True
[INFO]    pc_filter: 2.5
[INFO]    pc_geometric: False
[INFO]    pc_las: False
[INFO]    pc_quality: medium
[INFO]    pc_rectify: False
[INFO]    pc_sample: 0
[INFO]    pc_tile: False
[INFO]    primary_band: auto
[INFO]    project_path: D:\WebODM\resources\app\apps\NodeODM\data
[INFO]    radiometric_calibration: none
[INFO]    rerun: None
[INFO]    rerun_all: False
[INFO]    rerun_from: None
[... 137 lines cut ...]
commands.command_runner(
File "D:\WebODM\resources\app\apps\ODM\SuperBuild\install\bin\opensfm\opensfm\commands\command_runner.py", line 37, in command_runner
command.run(data, args)
File "D:\WebODM\resources\app\apps\ODM\SuperBuild\install\bin\opensfm\opensfm\commands\command.py", line 12, in run
self.run_impl(data, args)
File "D:\WebODM\resources\app\apps\ODM\SuperBuild\install\bin\opensfm\opensfm\commands\reconstruct.py", line 11, in run_impl
reconstruct.run_dataset(dataset)
File "D:\WebODM\resources\app\apps\ODM\SuperBuild\install\bin\opensfm\opensfm\actions\reconstruct.py", line 9, in run_dataset
tracks_manager = data.load_tracks_manager()
File "D:\WebODM\resources\app\apps\ODM\SuperBuild\install\bin\opensfm\opensfm\dataset.py", line 381, in load_tracks_manager
return pymap.TracksManager.instanciate_from_file(self._tracks_manager_file(filename))
RuntimeError: Can't read tracks manager file
[ERROR]   The program could not process this dataset using the current settings. Check that the images have enough overlap, that there are enough recognizable features and that the images are in focus. You could also try to increase the --min-num-features parameter.The program will now exit.
```

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

This should process normally and does when I disable my nvidia GPU.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
```
Created on: 1/12/2022, 7:17:19 PM
Processing Node: node-odm-1 (manual)
Options: auto-boundary: true, fast-orthophoto: true, max-concurrency: 6, skip-3dmodel: true
```

pierotofy, 2022-01-13:
Oh interesting; perhaps we need to multiply the image size by a factor to account for the intermediate octave layers.

Could you try to tweak this line?

https://github.com/OpenDroneMap/ODM/blob/66f32588c66c5b84a0d81e044645778cd256e2d2/opendm/gpu.py#L14

By adding a multiplication? (e.g. `fits = pypopsift.fits_texture(width * 1.2, height * 1.2)` which adds a margin of 20%). I'd be curious to see what's the minimum factor that allows you to get this processed.

Saijin-Naib, 2022-01-13:
Trying with:  
```
@lru_cache(maxsize=None)
def has_popsift_and_can_handle_texsize(width, height):
    try:
        from opensfm import pypopsift
        fits = pypopsift.fits_texture(width * 2.0, height * 2.0)
        if not fits:
            log.ODM_WARNING("Image size (%sx%spx) would not fit in GPU memory, falling back to CPU" % (width * 2.0, height * 2.0))
        return fits
    except (ModuleNotFoundError, ImportError):
        return False
    except Exception as e:
        log.ODM_WARNING(str(e))
        return False
```

Returns:  
```
[INFO]    ==============
[INFO]    Running dataset stage
[INFO]    Loading dataset from: D:\WebODM\resources\app\apps\NodeODM\data\951218a1-b597-450e-8144-5ecd979cbeda\images
[INFO]    Loading 201 images
[INFO]    Wrote images database: D:\WebODM\resources\app\apps\NodeODM\data\951218a1-b597-450e-8144-5ecd979cbeda\images.json
[INFO]    Found 201 usable images
[INFO]    Parsing SRS header: WGS84 UTM 32N
[INFO]    Finished dataset stage
[INFO]    Running split stage
[INFO]    Normal dataset, will process all at once.
[INFO]    Finished split stage
[INFO]    Running merge stage
[INFO]    Normal dataset, nothing to merge.
[INFO]    Finished merge stage
[INFO]    Running opensfm stage
[INFO]    Photo dimensions for feature extraction: 4000px
[WARNING] fits_texture(): incompatible function arguments. The following argument types are supported:
1. (width: int, height: int, downsampling: float = -1) -> bool

Invoked with: 8000.0, 6000.0[INFO]    Altitude data detected, enabling it for GPS alignment
[INFO]    ['use_exif_size: no', 'flann_algorithm: KDTREE', 'feature_process_size: 4000', 'feature_min_frames: 8000', 'processes: 6', 'matching_gps_neighbors: 8', 'matching_gps_distance: 0', 'matching_graph_rounds: 50', 'optimize_camera_parameters: yes', 'reconstruction_algorithm: incremental', 'undistorted_image_format: tif', 'bundle_outlier_filtering_type: AUTO', 'sift_peak_threshold: 0.066', 'align_orientation_prior: vertical', 'triangulation_type: ROBUST', 'retriangulation_ratio: 2', 'matcher_type: FLANN', 'feature_type: SIFT', 'use_altitude_tag: yes', 'align_method: auto', 'local_bundle_radius: 0']
[INFO]    Wrote reference_lla.json
[INFO]    running "D:\WebODM\resources\app\apps\ODM\SuperBuild\install\bin\opensfm\bin\opensfm" detect_features "D:\WebODM\resources\app\apps\NodeODM\data\951218a1-b597-450e-8144-5ecd979cbeda\opensfm"
2022-01-13 09:06:52,322 INFO: Capping memory usage to ~ 11335.6953125 MB
2022-01-13 09:06:52,322 INFO: Expecting to process 61 images.
2022-01-13 09:06:52,322 INFO: Reading data for image DJI_0002.JPG (queue-size=0
2022-01-13 09:06:52,322 INFO: Reading data for image DJI_0136.JPG (queue-size=0
2022-01-13 09:06:52,338 INFO: Reading data for image DJI_0867.JPG (queue-size=0
2022-01-13 09:06:53,416 INFO: Reading data for image DJI_0869.JPG (queue-size=1
2022-01-13 09:06:53,416 INFO: Extracting ROOT_SIFT features for image DJI_0867.JPG
2022-01-13 09:06:53,431 INFO: Reading data for image DJI_0138.JPG (queue-size=1
2022-01-13 09:06:53,447 INFO: Reading data for image DJI_0004.JPG (queue-size=1
2022-01-13 09:06:53,447 INFO: Extracting ROOT_SIFT features for image DJI_0136.JPG
2022-01-13 09:06:53,447 INFO: Extracting ROOT_SIFT features for image DJI_0002.JPG
2022-01-13 09:06:53,478 DEBUG: Computing sift with threshold 0.066
2022-01-13 09:06:53,494 DEBUG: Computing sift with threshold 0.066
2022-01-13 09:06:53,494 DEBUG: Computing sift with threshold 0.066
2022-01-13 09:06:54,900 INFO: Reading data for image DJI_0871.JPG (queue-size=1
2022-01-13 09:06:55,135 INFO: Reading data for image DJI_0006.JPG (queue-size=2
2022-01-13 09:06:55,166 INFO: Reading data for image DJI_0140.JPG (queue-size=3
2022-01-13 09:06:56,541 INFO: Reading data for image DJI_0873.JPG (queue-size=4
2022-01-13 09:06:56,994 INFO: Reading data for image DJI_0008.JPG (queue-size=5
2022-01-13 09:06:57,103 INFO: Reading data for image DJI_0737.JPG (queue-size=6
2022-01-13 09:06:58,275 DEBUG: Found 1045 points in 4.796880006790161s
2022-01-13 09:06:58,306 DEBUG: Found 65 points in 4.8125059604644775s
2022-01-13 09:06:58,306 DEBUG: reducing threshold
2022-01-13 09:06:58,306 INFO: Reading data for image DJI_0875.JPG (queue-size=7
2022-01-13 09:06:58,306 DEBUG: reducing threshold
2022-01-13 09:06:58,306 DEBUG: Computing sift with threshold 0.044000000000000004
2022-01-13 09:06:58,306 DEBUG: Computing sift with threshold 0.044000000000000004
2022-01-13 09:06:58,385 DEBUG: Found 723 points in 4.89062762260437s
[... 66 lines cut ...]
2022-01-13 09:07:15,629 INFO: Reading data for image DJI_0763.JPG (queue-size=39
2022-01-13 09:07:15,970 INFO: Reading data for image DJI_0899.JPG (queue-size=40
2022-01-13 09:07:16,973 INFO: Reading data for image DJI_0034.JPG (queue-size=41
2022-01-13 09:07:16,995 INFO: Reading data for image DJI_0765.JPG (queue-size=42
2022-01-13 09:07:17,026 DEBUG: Found 1370 points in 4.0355494022369385s
2022-01-13 09:07:17,026 DEBUG: reducing threshold
2022-01-13 09:07:17,027 DEBUG: Computing sift with threshold 0.044000000000000004
2022-01-13 09:07:17,260 INFO: Reading data for image DJI_0901.JPG (queue-size=43
2022-01-13 09:07:18,324 INFO: Reading data for image DJI_0036.JPG (queue-size=44
2022-01-13 09:07:18,324 INFO: Reading data for image DJI_0767.JPG (queue-size=45
2022-01-13 09:07:18,701 INFO: Reading data for image DJI_0903.JPG (queue-size=46
2022-01-13 09:07:19,738 DEBUG: Found 30466 points in 4.548115968704224s
2022-01-13 09:07:19,738 DEBUG: done
[... cut ...] ding data for image DJI_0038.JPG (queue-size=47
2022-01-13 09:07:19,917 INFO: Reading data for image DJI_0769.JPG (queue-size=48
```

So it appears to be processing properly, though it has not touched my GPU and it threw this exception:  
```
[INFO]    Photo dimensions for feature extraction: 4000px
[WARNING] fits_texture(): incompatible function arguments. The following argument types are supported:
1. (width: int, height: int, downsampling: float = -1) -> bool

Invoked with: 8000.0, 6000.0
```

pierotofy, 2022-01-13:
Oh, sorry, you need to cast to int:

```
fits = pypopsift.fits_texture(int(width * 1.2), int(height * 1.2))
```

Saijin-Naib, 2022-01-13:
Retried:  
```
@lru_cache(maxsize=None)
def has_popsift_and_can_handle_texsize(width, height):
    try:
        from opensfm import pypopsift
        fits = pypopsift.fits_texture(int(width * 1.25), int(height * 1.25))
        if not fits:
            log.ODM_WARNING("Image size (%sx%spx) would not fit in GPU memory, falling back to CPU" % (width, height))
        return fits
    except (ModuleNotFoundError, ImportError):
        return False
    except Exception as e:
        log.ODM_WARNING(str(e))
        return False
```
Errored, but successfully fell back to CPU pathway:  
```
[INFO]    ==============
[INFO]    Running dataset stage
[INFO]    Loading dataset from: D:\WebODM\resources\app\apps\NodeODM\data\951218a1-b597-450e-8144-5ecd979cbeda\images
[INFO]    Loading 201 images
[INFO]    Wrote images database: D:\WebODM\resources\app\apps\NodeODM\data\951218a1-b597-450e-8144-5ecd979cbeda\images.json
[INFO]    Found 201 usable images
[INFO]    Parsing SRS header: WGS84 UTM 32N
[INFO]    Finished dataset stage
[INFO]    Running split stage
[INFO]    Normal dataset, will process all at once.
[INFO]    Finished split stage
[INFO]    Running merge stage
[INFO]    Normal dataset, nothing to merge.
[INFO]    Finished merge stage
[INFO]    Running opensfm stage
[INFO]    Maximum photo dimensions: 8000px
[INFO]    Photo dimensions for feature extraction: 4000px
[WARNING] Image size (4000x3000px) would not fit in GPU memory, falling back to CPU
[INFO]    Altitude data detected, enabling it for GPS alignment
[INFO]    ['use_exif_size: no', 'flann_algorithm: KDTREE', 'feature_process_size: 4000', 'feature_min_frames: 8000', 'processes: 6', 'matching_gps_neighbors: 8', 'matching_gps_distance: 0', 'matching_graph_rounds: 50', 'optimize_camera_parameters: yes', 'reconstruction_algorithm: incremental', 'undistorted_image_format: tif', 'bundle_outlier_filtering_type: AUTO', 'sift_peak_threshold: 0.066', 'align_orientation_prior: vertical', 'triangulation_type: ROBUST', 'retriangulation_ratio: 2', 'matcher_type: FLANN', 'feature_type: SIFT', 'use_altitude_tag: yes', 'align_method: auto', 'local_bundle_radius: 0']
[INFO]    Wrote reference_lla.json
[INFO]    running "D:\WebODM\resources\app\apps\ODM\SuperBuild\install\bin\opensfm\bin\opensfm" detect_features "D:\WebODM\resources\app\apps\NodeODM\data\951218a1-b597-450e-8144-5ecd979cbeda\opensfm"
2022-01-13 15:12:47,100 INFO: Expecting to process 63 images.
```

Confirmed that images are actually 8000x6000, so it looks like improper dimensions are being passed to width/height somehow.
[image]

pierotofy, 2022-01-13:
Right, graceful failover is what we're going after (which is better than just crashing).

So what's the minimum scalar value that is able to make the process fallback gracefully?

Saijin-Naib, 2022-01-13:
I'm iterate and get back on that. I wasn't testing for that but rather trying to get it to process via GPU, hence going up to 2.25

pierotofy, 2022-01-13:
Yeah there's no processing on the GPU if the card doesn't have sufficient memory; you need to reduce --feature-quality to use lower size images. But we should fail gracefully when a card doesn't have sufficient memory.

Saijin-Naib, 2022-01-13:
What's the maths on that? I would assume 4GB dedicated and something like an extra 16GB shared would be sufficient.

pierotofy, 2022-01-13:
No idea! :man_shrugging: :)

Saijin-Naib, 2022-01-13:
What's odd is that I was processing via GPU earlier this week, so I know my GTX 1050TI OC can swing it... Hmm 🤔

Saijin-Naib, 2022-01-14:
Scalar of ```1.025``` seems to be the minimum viable/reliable number here. Don't have any other GPUs on-hand to test, however (my wife's machine is identical to mine with GTX 1050TI OC 4GB).

pierotofy, 2022-01-14:
Excellent! Would you mind opening a pull request with this change? 🙏

closed 2022-01-14 by pierotofy

sbonaime, 2022-02-08:
Hi,
I would like to understand what  pypopsift.fits_texture function is doing but I can't find where it come from. 
fits_texture  is not in https://github.com/mapillary/OpenSfM

Thanks