OpenDroneMap/ODM #1409: Failing process during sift I believe

DarkSlaayer, 2022-01-24:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 97 on Windows 11 
Does it on Windows 10 also

### What is the problem?

128GB Swap
Fails when its around 60gb usually

Stopping during sift

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

`[INFO]    Initializing ODM 2.7.2 - Mon Jan 24 04:31:29  2022
[INFO]    ==============
[INFO]    auto_boundary: True
[INFO]    boundary: {}
[INFO]    build_overviews: False
[INFO]    camera_lens: auto
[INFO]    cameras: {}
[INFO]    cog: True
[INFO]    copy_to: None
[INFO]    crop: 3
[INFO]    debug: True
[INFO]    dem_decimation: 1
[INFO]    dem_euclidean_map: False
[INFO]    dem_gapfill_steps: 3
[INFO]    dem_resolution: 5
[INFO]    depthmap_resolution: 640
[INFO]    dsm: True
[INFO]    dtm: True
[INFO]    end_with: odm_postprocess
[INFO]    fast_orthophoto: False
[INFO]    feature_quality: ultra
[INFO]    feature_type: sift
[INFO]    force_gps: False
[INFO]    gcp: None
[INFO]    geo: None
[INFO]    gps_accuracy: 10
[INFO]    ignore_gsd: True
[INFO]    matcher_neighbors: 36
[INFO]    matcher_type: flann
[INFO]    max_concurrency: 16
[INFO]    merge: all
[INFO]    mesh_octree_depth: 12
[INFO]    mesh_size: 200000
[INFO]    min_num_features: 18000
[INFO]    name: df6b7d2b-1e44-4f46-bd98-d2524e8aa4fc
[INFO]    optimize_disk_space: False
[INFO]    orthophoto_compression: DEFLATE
[INFO]    orthophoto_cutline: False
[INFO]    orthophoto_kmz: False
[INFO]    orthophoto_no_tiled: False
[INFO]    orthophoto_png: False
[INFO]    orthophoto_resolution: 1.0
[INFO]    pc_classify: True
[INFO]    pc_csv: False
[INFO]    pc_ept: True
[INFO]    pc_filter: 0.0
[INFO]    pc_geometric: True
[INFO]    pc_las: False
[INFO]    pc_quality: ultra
[INFO]    pc_rectify: True
[INFO]    pc_sample: 0
[INFO]    pc_tile: False
[... 263 lines cut ...]
Child returned 1
Traceback (most recent call last):
File "/code/stages/odm_app.py", line 94, in execute
self.first_stage.run()
File "/code/opendm/types.py", line 346, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 346, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 346, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 327, in run
self.process(self.args, outputs)
File "/code/stages/run_opensfm.py", line 35, in process
octx.feature_matching(self.rerun())
File "/code/opendm/osfm.py", line 321, in feature_matching
self.run('detect_features')
File "/code/opendm/osfm.py", line 34, in run
system.run('"%s" %s "%s"' %
File "/code/opendm/system.py", line 106, in run
raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 1

===== Done, human-readable information to follow... =====

[ERROR]   Uh oh! Processing stopped because of strange values in the reconstruction. This is often a sign that the input data has some issues or the software cannot deal with it. Have you followed best practices for data acquisition? See https://docs.opendronemap.org/flying/`

pierotofy, 2022-01-24:
> If possible please include a copy of your dataset uploaded on Google Drive or Dropbox

DarkSlaayer, 2022-01-27:
Heres another pretty much any larger dataset its happening to.

Seems like a python issue?

[INFO]    Wrote images database: /var/www/data/f30077d2-6638-49ac-95a8-427f709ecc12/images.json
[INFO]    Found 606 usable images
[INFO]    Parsing SRS header: WGS84 UTM 14N
[INFO]    Finished dataset stage
[INFO]    Running split stage
[INFO]    Large dataset detected (606 photos) and split set at 20. Preparing split merge.
[INFO]    Maximum photo dimensions: 4864px
[INFO]    Photo dimensions for feature extraction: 4864px
[WARNING] Image size (4864x3648px) would not fit in GPU memory, falling back to CPU
[INFO]    Altitude data detected, enabling it for GPS alignment
[INFO]    ['use_exif_size: no', 'flann_algorithm: KDTREE', 'feature_process_size: 4864', 'feature_min_frames: 30000', 'processes: 16', 'matching_gps_neighbors: 36', 'matching_gps_distance: 0', 'matching_graph_rounds: 50', 'optimize_camera_parameters: yes', 'reconstruction_algorithm: incremental', 'undistorted_image_format: tif', 'bundle_outlier_filtering_type: AUTO', 'sift_peak_threshold: 0.066', 'align_orientation_prior: vertical', 'triangulation_type: ROBUST', 'retriangulation_ratio: 2', 'matcher_type: FLANN', 'feature_type: SIFT', 'use_altitude_tag: yes', 'align_method: auto', 'local_bundle_radius: 0', 'submodels_relpath: ../submodels/opensfm', 'submodel_relpath_template: ../submodels/submodel_%04d/opensfm', 'submodel_images_relpath_template: ../submodels/submodel_%04d/images', 'submodel_size: 20', 'submodel_overlap: 150']
[INFO]    Wrote reference_lla.json
[INFO]    running "/code/SuperBuild/install/bin/opensfm/bin/opensfm" detect_features "/var/www/data/f30077d2-6638-49ac-95a8-427f709ecc12/opensfm"
2022-01-27 04:29:32,631 INFO: Expecting to process 65 images.
2022-01-27 04:29:32,637 INFO: Reading data for image DJI_0669.JPG (queue-size=0
2022-01-27 04:29:32,637 INFO: Reading data for image DJI_0054-edit.JPG (queue-size=0
2022-01-27 04:29:32,637 INFO: Reading data for image DJI_0255.JPG (queue-size=0
2022-01-27 04:29:32,637 INFO: Reading data for image DJI_0417.JPG (queue-size=0
2022-01-27 04:29:33,076 INFO: Reading data for image DJI_0071-edit.JPG (queue-size=1
2022-01-27 04:29:33,077 INFO: Extracting ROOT_SIFT features for image DJI_0255.JPG
2022-01-27 04:29:33,083 INFO: Reading data for image DJI_0325.JPG (queue-size=1
2022-01-27 04:29:33,084 INFO: Extracting ROOT_SIFT features for image DJI_0054-edit.JPG
2022-01-27 04:29:33,091 DEBUG: Computing sift with threshold 0.066
2022-01-27 04:29:33,096 DEBUG: Computing sift with threshold 0.066
2022-01-27 04:29:33,103 INFO: Reading data for image DJI_0678.JPG (queue-size=1
2022-01-27 04:29:33,103 INFO: Extracting ROOT_SIFT features for image DJI_0417.JPG
2022-01-27 04:29:33,110 INFO: Reading data for image DJI_0448.JPG (queue-size=1
2022-01-27 04:29:33,115 INFO: Extracting ROOT_SIFT features for image DJI_0669.JPG
2022-01-27 04:29:33,127 DEBUG: Computing sift with threshold 0.066
2022-01-27 04:29:33,163 DEBUG: Computing sift with threshold 0.066
2022-01-27 04:29:33,629 INFO: Reading data for image DJI_0430.JPG (queue-size=1
2022-01-27 04:29:33,630 INFO: Extracting ROOT_SIFT features for image DJI_0325.JPG
2022-01-27 04:29:33,643 INFO: Reading data for image DJI_0450.JPG (queue-size=1
2022-01-27 04:29:33,644 INFO: Extracting ROOT_SIFT features for image DJI_0071-edit.JPG
2022-01-27 04:29:33,657 DEBUG: Computing sift with threshold 0.066
2022-01-27 04:29:33,693 INFO: Reading data for image DJI_0590.JPG (queue-size=1
2022-01-27 04:29:33,693 INFO: Extracting ROOT_SIFT features for image DJI_0678.JPG
2022-01-27 04:29:33,694 DEBUG: Computing sift with threshold 0.066
2022-01-27 04:29:33,756 DEBUG: Computing sift with threshold 0.066
2022-01-27 04:29:33,790 INFO: Reading data for image DJI_0543.JPG (queue-size=1
2022-01-27 04:29:33,791 INFO: Extracting ROOT_SIFT features for image DJI_0448.JPG
2022-01-27 04:29:33,851 DEBUG: Computing sift with threshold 0.066
2022-01-27 04:29:34,440 INFO: Reading data for image DJI_0494.JPG (queue-size=1
2022-01-27 04:29:34,443 INFO: Extracting ROOT_SIFT features for image DJI_0450.JPG
2022-01-27 04:29:34,441 INFO: Reading data for image DJI_0041-edit.JPG (queue-size=2
2022-01-27 04:29:34,444 INFO: Extracting ROOT_SIFT features for image DJI_0430.JPG
2022-01-27 04:29:34,507 INFO: Reading data for image DJI_0007-edit.JPG (queue-size=1
2022-01-27 04:29:34,509 INFO: Extracting ROOT_SIFT features for image DJI_0590.JPG
2022-01-27 04:29:34,517 DEBUG: Computing sift with threshold 0.066
2022-01-27 04:29:34,517 DEBUG: Computing sift with threshold 0.066
2022-01-27 04:29:34,606 DEBUG: Computing sift with threshold 0.066
2022-01-27 04:29:34,665 INFO: Reading data for image DJI_0015-edit.JPG (queue-size=1
2022-01-27 04:29:34,667 INFO: Extracting ROOT_SIFT features for image DJI_0543.JPG
2022-01-27 04:29:34,776 DEBUG: Computing sift with threshold 0.066
2022-01-27 04:29:35,485 INFO: Reading data for image DJI_0419.JPG (queue-size=1
2022-01-27 04:29:35,658 INFO: Reading data for image DJI_0504.JPG (queue-size=2
2022-01-27 04:29:35,717 INFO: Reading data for image DJI_0588.JPG (queue-size=3
2022-01-27 04:29:36,004 INFO: Reading data for image DJI_0349.JPG (queue-size=4
2022-01-27 04:29:37,283 INFO: Reading data for image DJI_0463.JPG (queue-size=5
2022-01-27 04:29:37,310 INFO: Reading data for image DJI_0401.JPG (queue-size=6
2022-01-27 04:29:37,340 INFO: Reading data for image DJI_0248.JPG (queue-size=7
[... cut ...]
File "/code/opendm/system.py", line 106, in run
raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 134

===== Done, human-readable information to follow... =====

[ERROR]   Uh oh! Processing stopped because of strange values in the reconstruction. This is often a sign that the input data has some issues or the software cannot deal with it. Have you followed best practices for data acquisition? See https://docs.opendronemap.org/flying/

markFieldman, 2022-01-27:
Similar behavior in my case too :(

DarkSlaayer, 2022-01-28:
I increased full_queue_timeout = 120 to 120000 and it seems to be working. 

Now I don’t have a full grasp on the inner workings of odm and I never had this issue until recently even tried native windows odm and same issue. But I think part of the problem is when using page filing/swap file and it times out before it finishes due to the slow nature of page filing.

PR #1419 (2022-02-25): Bug fixes and misc improvements
Closes #1409﻿
closed 2022-02-25 by pierotofy