OpenDroneMap/ODM #1410: Possible Bug: ERROR Could not execute GRASS script /webodm/coreplugins/elevationmap/elevationmap.py

Saijin-Naib, 2022-01-26:
### How did you install OpenDroneMap? (Docker, natively, ...)?
Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Microsoft Edge Beta, Flatpak
Alpine Linux Edge

### What is the problem?
Running the Elevation Map plugin does not succeed.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.
This plugin should process without errors.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
```
$ docker images
REPOSITORY                   TAG       IMAGE ID       CREATED        SIZE
opendronemap/webodm_webapp   latest    97ff1bc45181   11 hours ago   1.6GB
opendronemap/nodeodm         latest    1dd539fb30e0   6 days ago     1.76GB
redis                        latest    7614ae9453d1   5 weeks ago    113MB
opendronemap/odm             latest    e890026fc5af   5 weeks ago    1.43GB
ubuntu                       20.04     ba6acccedd29   3 months ago   72.8MB
ubuntu                       21.04     de6f83bfe0b6   3 months ago   80MB
opendronemap/webodm_db       latest    e40c0f274bba   4 months ago   695MB
```
```
worker      | INFO Executing grass script from /webodm/app/media/tmp/tmpnyjynk4h_grass_engine: /usr/bin/grass78 -c /webodm/app/media/project/1/task/4b9abf46-1add-479d-bc3f-2c10f7aa5499/assets/odm_dem/dsm.tif location --exec python3 /webodm/coreplugins/elevationmap/elevationmap.py dsm=/webodm/app/media/project/1/task/4b9abf46-1add-479d-bc3f-2c10f7aa5499/assets/odm_dem/dsm.tif interval=5 format=GeoJSON noise_filter_size=3.0 epsg=4326
worker      | [2022-01-26 02:33:54,504: INFO/ForkPoolWorker-1] worker.tasks.execute_grass_script[78364cff-e752-4fc3-a8c2-4752adb32afa]: Executing grass script from /webodm/app/media/tmp/tmpnyjynk4h_grass_engine: /usr/bin/grass78 -c /webodm/app/media/project/1/task/4b9abf46-1add-479d-bc3f-2c10f7aa5499/assets/odm_dem/dsm.tif location --exec python3 /webodm/coreplugins/elevationmap/elevationmap.py dsm=/webodm/app/media/project/1/task/4b9abf46-1add-479d-bc3f-2c10f7aa5499/assets/odm_dem/dsm.tif interval=5 format=GeoJSON noise_filter_size=3.0 epsg=4326
worker      | ERROR Could not execute GRASS script /webodm/coreplugins/elevationmap/elevationmap.py from /webodm/app/media/tmp/tmpnyjynk4h_grass_engine: Starting GRASS GIS...
worker      | Creating new GRASS GIS location <location>...
worker      | Cleaning up temporary files...
worker      | Executing <python3 /webodm/coreplugins/elevationmap/elevationmap.py dsm=/webodm/app/media/project/1/task/4b9abf46-1add-479d-bc3f-2c10f7aa5499/assets/odm_dem/dsm.tif interval=5 format=GeoJSON noise_filter_size=3.0 epsg=4326> ...
worker      | Traceback (most recent call last):
worker      |   File "/webodm/coreplugins/elevationmap/elevationmap.py", line 48, in <module>
worker      |     import cv2, math, argparse
worker      | ModuleNotFoundError: No module named 'cv2'
worker      | Execution of <python3 /webodm/coreplugins/elevationmap/elevationmap.py dsm=/webodm/app/media/project/1/task/4b9abf46-1add-479d-bc3f-2c10f7aa5499/assets/odm_dem/dsm.tif interval=5 format=GeoJSON noise_filter_size=3.0 epsg=4326> finished.
worker      | Cleaning up temporary files...
worker      | [2022-01-26 02:33:55,301: ERROR/ForkPoolWorker-1] worker.tasks.execute_grass_script[78364cff-e752-4fc3-a8c2-4752adb32afa]: Could not execute GRASS script /webodm/coreplugins/elevationmap/elevationmap.py from /webodm/app/media/tmp/tmpnyjynk4h_grass_engine: Starting GRASS GIS...
worker      | Creating new GRASS GIS location <location>...
worker      | Cleaning up temporary files...
worker      | Executing <python3 /webodm/coreplugins/elevationmap/elevationmap.py dsm=/webodm/app/media/project/1/task/4b9abf46-1add-479d-bc3f-2c10f7aa5499/assets/odm_dem/dsm.tif interval=5 format=GeoJSON noise_filter_size=3.0 epsg=4326> ...
worker      | Traceback (most recent call last):
worker      |   File "/webodm/coreplugins/elevationmap/elevationmap.py", line 48, in <module>
worker      |     import cv2, math, argparse
worker      | ModuleNotFoundError: No module named 'cv2'
worker      | Execution of <python3 /webodm/coreplugins/elevationmap/elevationmap.py dsm=/webodm/app/media/project/1/task/4b9abf46-1add-479d-bc3f-2c10f7aa5499/assets/odm_dem/dsm.tif interval=5 format=GeoJSON noise_filter_size=3.0 epsg=4326> finished.
worker      | Cleaning up temporary files...
```
[image]

Forum thread with original reporter:  
https://community.opendronemap.org/t/elevation-map-preview-error-windows-10-pro/10282?u=saijin_naib

rumenmitrev, 2022-01-27:
As a logs suggests , there is no opencv(cv2) installed inside worker container , alongside with scipy , numpy
Here is the solution/fix:
https://github.com/OpenDroneMap/WebODM/issues/1059#issuecomment-986791763

pierotofy, 2022-02-17:
This should now be fixed in the latest docker images. :+1:

closed 2022-02-17 by pierotofy

Saijin-Naib, 2022-02-17:
Indeed! Thanks so much!
[image]

```
$ docker images
REPOSITORY                   TAG       IMAGE ID       CREATED        SIZE
opendronemap/nodeodm         latest    264ff6a37d8d   40 hours ago   1.77GB
opendronemap/webodm_webapp   latest    0924de891f48   9 days ago     1.6GB
redis                        latest    f1b6973564e9   3 weeks ago    113MB
opendronemap/webodm_db       latest    e40c0f274bba   5 months ago   695MB
```