OpenDroneMap/ODM #1412: Warn of or skip images with corrupt EXIF coordinate info

ninsbl, 2022-01-30:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

MS Edge

### What is the problem?

Some images in my dataset contain the following coordinates in the exif metadata:

```
    exif:GPSLatitude: 0/1, 0/1, 0/10000
    exif:GPSLatitudeRef: N
    exif:GPSLongitude: 0/1, 0/1, 0/10000
    exif:GPSLongitudeRef: E
```
Causing ODM to fail after quite some time of processing. For novice users error messages suggestion to check the flight planning or increase the min-num-features parameter are not very helpful to spot the problem with a particular file.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

It would be great if ODM could check the input data for consistency and skip images with invalid exif data, or at least pring a clear warning. Ideally, the error message points to the file causing trouble.

Alternatively, ODM could provide a "testrun" option, where data is checked for consistency and validity (corrupt images, inconsisten exif as here).

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Try to process a dataset where one image contains the above lat/lon metadata in EXIF ...

pierotofy, 2022-02-17:
Could you share some of these corrupted images @ninsbl :pray: ?

ninsbl, 2022-02-17:
Will ask the data owner if that is OK. Do you need a full dataset or a single corrupt image?

pierotofy, 2022-02-17:
Just a few images (12-16).

pierotofy, 2022-03-08:
Closing due to inactivity, feel free to re-open with the necessary test images. :pray:

ninsbl, 2022-03-11:
After getting permission to share the images I now uploaded a sample of 13 images to:
https://transfer.sh/pRe31o/corrput_images.zip
You can download with wget or similar.

The corrupt image is `DJI_0709.JPG` and you can check the problematic header with:
`identify -verbose DJI_0709.JPG`

Data will be available for 14 days from now.

ninsbl, 2022-03-11:
P.S.: I cannot re-open...

pierotofy, 2022-03-11:
Thanks! I've added a warning for images that have a likely invalid GPS value.

PR #1434 (2022-03-11): Warn if images have invalid GPS position
Closes #1412

We cannot just skip these images because some vendors might write 0 in images that are non-georeferenced. But a warning should at least help identify them.
closed 2022-03-11 by pierotofy