OpenDroneMap/ODM #1415: Possible Bug: geo location file always overwrites opk

originlake, 2022-02-03:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

ubuntu

### What is the problem?

The file format of geo location file given in documents is below
```
<projection>
image_name geo_x geo_y [geo_z] [omega (degrees)] [phi (degrees)] [kappa (degrees)] [horz accuracy (meters)] [vert accuracy (meters)] [extras...]
...
```
The current way of updating will set all the optional values to `None` in the ODM_Photo object. There is no way to provide different GPS xyz data with geo.txt but still use the opk values extracted from image metadata. 

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

When this feature is implemented, the opk value is not used, so this behavior is fine. But after adding the triangulation algorithm, this might need to be carefully handled. It's also not possible to provide horizontal/vertical accuracy without overwriting the opk value.

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

[Type answer here]

originlake, 2022-02-04:
I can help with this, but need to understand what's the expected behavior. Should we overwrite the optional values to None if they are not given?  If only x and y are given, should we use z from the image or set it to none? What about opk, and horizontal/vertical accuracy. I think x,y,z, and horizontal/vertical accuracy are tied together, so it makes sense to overwrite all of them together, but for opk, I guess they need to be considered separately.
Also maybe we can add a specific value that indicates I don't want to overwrite the opk, something like none for example
```
EPSG:4326
image_name -89 32 256 None None None 0.01 0.01 
...
```

pierotofy, 2022-02-04:
I think that would be a good approach (just use `NaN` instead of `None`, which is the same behavior we have with GCP files).

closed 2022-06-20 by pierotofy