OpenDroneMap/ODM #1417: For Better Banana (feature request)

ichsan2895, 2022-02-12:
I have tested this [Banana Dataset](https://github.com/pierotofy/dataset_banana/tree/master)

These settings are
`run --project-path “D:\qwerty\Banana_ODM” --mesh-octree-depth 14 --mesh-size 300000 --pc-quality ultra --pc-geometric `

I got a banana with very sharp texture which is really great (IMO better than Reality Capture texture with same dataset)
[image]

But unfortunatelly, I can't eat this banana since there big hole in it
[image]

Why the hole appears? My hypothesis, it was appear at texturing stage since ODM remove the faces which ODM can't associated/interpolated from input images texture.

I checked `/code/odm_meshing/odm_mesh.ply` and it doesn't any hole in it, but it was uncolored.
[image]

I also try to add `--texturing-keep-unseen-faces`, The hole was dissapear ! but, the hole was filled by pure white faces.
[image]

Now the suggestion is, **The texturing stage can interpolate the color of that faces instead of removing the faces or instead of fill it by pure white faces**

ichsan2895, 2022-02-12:
Lets try with meshlab

`/code/odm_filterpoints/point_cloud.ply` has colored point cloud
[image]

I use **meshlab's screened poisson** with `Resolution depth` is 11 and `Adaptive Octree Depth` is 8. The result is good enough, the color of the hole was filled by interpolated point cloud colors
[image]
But the texture is not sharp as ODM
[image]

My idea was to sharper the colored point cloud with ODM's texturing stage. I rename that mesh to `odm_mesh.ply`, then put that mesh to `/code/odm_meshing`

But I was failed. The faces become dissapear again and get the hole

Still figure it how to fix that.

ichsan2895, 2022-02-12:
BONUS :
Another photogrammetry software result

IMO, I have said the **ODM** texture was better than **Reality Capture**, here is the screenshot
[image]

Keep good works !

But, **Reality Capture** tends to give color to the faces rather that removing it.

This is Reality capture with high quality settings for Reconstruction
[image]

pierotofy, 2022-02-17:
The key problem is that currently mvs-texturing does not support texture interpolation.

Would you be interested in adding support for it?

ichsan2895, 2022-02-18:
> The key problem is that currently mvs-texturing does not support texture interpolation.
> 
> Would you be interested in adding support for it?

Thanks for your information that mvs-texturing doesn't support texture interpolation. What can I do beside coding? Maybe I have plan to buy native windows WebODM for fund support

I also tried to increase this value which located in [this file](https://github.com/OpenDroneMap/mvs-texturing/blob/master/libs/tex/generate_texture_patches.cpp)
```
#define MAX_HOLE_NUM_FACES 100
#define MAX_HOLE_PATCH_SIZE 100
```

But since I don't know coding, I confuse how to change that file which is inside Docker environment. I use MXLinux 19.4 and WebODM Docker version.

pierotofy, 2022-03-08:
See https://opendronemap.org/get-involved/ for ways to help the project!

This is something tricky to add, since Reality Capture derives its point cloud from some type of mesh-like interpolation, it works quite differently from ODM.