OpenDroneMap/ODM #1427: GPU based docker image getting stuck in feature extraction on GPU machine

vdharankar, 2022-03-02:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Ubuntu 20.04

### What is the problem?

GPU based image gets stuck while extracting features

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

In old images it used to work fine

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

The image set recommended in example on readme and the command for GPU based ODM is what was used

pierotofy, 2022-03-08:
Can you share the images that are getting stuck?

vdharankar, 2022-03-11:
All type of images there is nothing specific here as I have tested , same image set used to get processed

closed 2022-03-15 by vdharankar