OpenDroneMap/ODM #1428: Can't build docker image from latest master

rumenmitrev, 2022-03-04:
Hello there , 

### How did you install OpenDroneMap? (Docker, natively, ...)?
Docker
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
N/A
### What is the problem?
Can't build docker image from latest master
### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.
Building image without issue
### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)
```
git clone https://github.com/OpenDroneMap/ODM.git odm
cd odm
docker build -t rmitrev/odm -f ./Dockerfile . > out.txt 2>&1
```

Will attach building logs later today.

ckato3, 2022-03-04:
Workflow (Publish Docker and WSL Images ) also has the same error and is not updating the docker image at DockerHub.

pierotofy, 2022-03-05:
Should be fixed, thanks for the report!

PR #1429 (2022-03-05): Fix OpenMVS build
Fixes #1428
﻿
closed 2022-03-05 by pierotofy