OpenDroneMap/ODM #1438: SNAP: ModuleNotFoundError: No module named 'encodings'

sysdef, 2022-03-19:
_This error was ignored in http://community.opendronemap.org/c/opendronemap so I open an issue here_ 

```
$ sudo /snap/bin/opendronemap 
Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = (not set)
  program name = 'python3'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = '/usr/bin/python3'
  sys.base_prefix = '/usr'
  sys.base_exec_prefix = '/usr'
  sys.executable = '/usr/bin/python3'
  sys.prefix = '/usr'
  sys.exec_prefix = '/usr'
  sys.path = [
    '/usr/lib/python38.zip',
    '/usr/lib/python3.8',
    '/usr/lib/python3.8/lib-dynload',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00007f66cf4d6740 (most recent call first):
<no Python frame>
```
```
$ uname -a; cat /etc/debian_version 
Linux can 5.10.0-12-amd64 #1 SMP Debian 5.10.103-1 (2022-03-07) x86_64 GNU/Linux
11.2
```

pierotofy, 2022-03-20:
Thanks for reporting the issue!

Currently our snap image hasn't been updated in a while due a build issue with https://github.com/diddlesnaps/snapcraft-container/issues/8 (not sure if this is related, but it could be).

Camilochiang, 2022-08-09:
Til today, August 9, this problem persist.

diddledani, 2022-08-26:
Hi, @pierotofy , I have good news, I believe I've fixed the container image so you can re-enable the snap build.

pierotofy, 2022-08-26:
Thanks @diddledani ! 🙏 I've seen the changes in the container image, but haven't tested it. I will sometimes this weekend (unless somebody beats me to it).

pierotofy, 2022-08-28:
It built! 🎉

JeroenDelcour, 2022-08-29:
I'm still getting the same `No module named 'encodings'` error on a fresh install through snap. I see the snap image has been updated yesterday so I assumed the fixed version had been deployed. Am I missing something?

JeroenDelcour, 2022-09-05:
Thanks for the fix, but it seems the [snap publishing pipeline is broken](https://github.com/OpenDroneMap/ODM/runs/8079544892?check_suite_focus=true) so I'm unable to test it.

pierotofy, 2022-09-05:
Mm, something else seems to be breaking the build, as you pointed out. I'm not sure of the cause.

pierotofy, 2022-09-06:
I made some changes, which seemed to have worked (?) https://github.com/OpenDroneMap/ODM/runs/8210390371?check_suite_focus=true

I haven't tested as I don't have snap installed locally.

diddledani, 2022-09-07:
oh yes, that was my bad. I forgot to remove the `part` that tries to use `snap/local` after removing all the files from the directory. Well done, you made the correct fix that I would have if I had got here earlier :-)

JeroenDelcour, 2022-09-07:
Thanks for the fix! The `No module named 'encodings'` error seems fixed. However, I'm now getting another missing dependency:

```
Traceback (most recent call last):
  File "/snap/opendronemap/177/odm/run.py", line 15, in <module>
    from opendm.utils import get_processing_results_paths, rm_r
  File "/snap/opendronemap/177/odm/opendm/utils.py", line 3, in <module>
    from opendm.photo import find_largest_photo_dims
  File "/snap/opendronemap/177/odm/opendm/photo.py", line 17, in <module>
    from opendm import get_image_size
  File "/snap/opendronemap/177/odm/opendm/get_image_size.py", line 2, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'
```

Should I make a new issue for this?

ArthurTlprt, 2022-09-17:
Same issue than @JeroenDelcour after installing ODM via snap on Ubuntu 20.

pierotofy, 2022-09-17:
No need for a new issue. Mm, something is still broken.

pierotofy, 2022-09-20:
One more attempt: https://github.com/OpenDroneMap/ODM/commit/081d4c89487137512b8f3b0e84c4acf39625af84

pierotofy, 2022-09-28:
I will assume this has been solved, if not please feel free to re-open? 🙏

JeroenDelcour, 2022-09-28:
It has not been solved. Updating the snap package and running `opendronemap` without any arguments still shows the same error. Have you tested your own commit? Are you not getting this error?

You can leave it closed if you like, I'm personally not going to put more time into this. I rely on opendronemap in a production setting, so until the snap release gets better testing to show its reliable, I'm going to stick to the Docker image.

pierotofy, 2022-09-28:
I unfortunately don't have an Ubuntu machine to test this on.

I will also stop putting more time into this; I don't use snap and this will have to wait for a community fix.

pierotofy, 2022-10-16:
Fixed and tested with https://github.com/OpenDroneMap/ODM/commit/61d1865bd3a522106461b8a9d754039eaa122ecc and https://github.com/OpenDroneMap/ODM/commit/093929dd914ee56fbfe6016769eb0bdaa1e62fbf

closed 2022-10-16 by pierotofy

sysdef, 2022-10-16:
Thank you for fixing that!

JeroenDelcour, 2022-10-17:
Thanks for fixing!

It took me a minute to figure out how to use it, so for anyone else wondering: it expects an absolute file path, rather than relative to the present working directory.

PR #1527 (2022-08-29): Fix python modules and library search paths
* Add PYTHONPATH environment variable to `snapcraft.yaml`
* Drop `snapcraft-preload` from snap package
* Update `LD_LIBRARY_PATH` in `snapcraft.yaml`

Fixes #1438
commit 5ab34603 (2022-08-29): Add PYTHONPATH environment variable to snapcraft.yaml

This fixes #1438: `encodings` module not found

The variable replaces the `sitecustomize.py` file from `snap/local` because it is no-longer functional due to changes in snapcraft.