OpenDroneMap/ODM #1441: Non ascii character in ODM Install Path

coumbsek, 2022-03-21:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Natively on Windows 10

### What is the problem?

ODM Console not functionnal if there is a non ascii character in ODM Install Path, for instance Cyrillic characters.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

Console should function 

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

Insert a non ascii character in ODM Install Path, for instance a Cyrillic character.

Here is the error log :
```['C:\\space é folder\\ODM', 'C:\\space Ã© folder\\ODM\\venv\\lib\\site-packages\\cv2\\python-3.8', 'C:\\space é folder\\ODM\\venv', 'C:\\space é folder\\ODM\\python38\\python38.zip', 'C:\\space é folder\\ODM\\python38\\DLLs', 'C:\\space é folder\\ODM\\python38\\lib', 'C:\\space é folder\\ODM\\python38', 'C:\\space é folder\\ODM\\venv\\lib\\site-packages', 'C:\\space é folder\\ODM\\SuperBuild\\install/lib/python3.9/dist-packages', 'C:\\space é folder\\ODM\\SuperBuild\\install/lib/python3.8/dist-packages', 'C:\\space é folder\\ODM\\SuperBuild\\install/lib/python3/dist-packages', 'C:\\space é folder\\ODM\\SuperBuild\\install/bin/opensfm']
Traceback (most recent call last):
  File "C:\space é folder\ODM\\run.py", line 15, in <module>
    from opendm.utils import get_processing_results_paths, rm_r
  File "C:\space é folder\ODM\opendm\utils.py", line 3, in <module>
    from opendm.photo import find_largest_photo_dim
  File "C:\space é folder\ODM\opendm\photo.py", line 16, in <module>
    from opendm import get_image_size
  File "C:\space é folder\ODM\opendm\get_image_size.py", line 2, in <module>
    import cv2
  File "C:\space é folder\ODM\venv\lib\site-packages\cv2\__init__.py", line 99, in <module>
    bootstrap()
  File "C:\space é folder\ODM\venv\lib\site-packages\cv2\__init__.py", line 88, in bootstrap
    import cv2
  File "C:\space é folder\ODM\venv\lib\site-packages\cv2\__init__.py", line 99, in <module>
    bootstrap()
  File "C:\space é folder\ODM\venv\lib\site-packages\cv2\__init__.py", line 23, in bootstrap
    raise ImportError('ERROR: recursion is detected during loading of "cv2" binary extensions. Check OpenCV installation.')
ImportError: ERROR: recursion is detected during loading of "cv2" binary extensions. Check OpenCV installation.```

smathermather, 2022-03-21:
Related: https://community.opendronemap.org/t/non-ascii-character-in-odm-install-path/11020

pierotofy, 2022-03-23:
Should be fixed now :clinking_glasses: !

closed 2022-03-23 by pierotofy