OpenDroneMap/ODM #1442: GPU processing crashes and fallsback to CPU processing

madisi98, 2022-03-21:
### How did you install OpenDroneMap? (Docker, natively, ...)?

Docker GPU image

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Ubuntu 20.04

### What is the problem?

GPU processing crashes and falls back to CPU processing during the feature extraction leaving this message:

> /code/SuperBuild/install/bin/opensfm/bin/opensfm: line 12: 65 Segmentation fault (core dumped) "$PYTHON" "$DIR"/opensfm_main.py "$@"
> [WARNING] GPU SIFT extraction failed, maybe the graphics card is not supported? Attempting fallback to CPU

### How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

I'm using the following command: 

> docker run -ti --rm --gpus all -v /home/manu/project/:/workspace opendronemap/odm:gpu --project-path /workspace datasets --dsm --dtm --min-num-features 15000 --matcher-neighbors 20 --dem-resolution 2.0 --feature-quality high --rerun-all

I'm running the docker in an nvidia rtx 3080, with driver version 510.47.03
The memory of the card is nowhere near full (normally using 2/10 GB)

Sadly I can't share my dataset, but it's about 1700 images (wihch make up for a total of 14GB)

Here I share the configuration printed at the beggining of the process:

> [INFO]    Initializing ODM 2.8.1 - Mon Mar 21 15:20:43  2022
> [INFO]    ==============
> [INFO]    auto_boundary: False
> [INFO]    boundary: {}
> [INFO]    build_overviews: False
> [INFO]    camera_lens: auto
> [INFO]    cameras: {}
> [INFO]    cog: False
> [INFO]    copy_to: None
> [INFO]    crop: 3
> [INFO]    debug: False
> [INFO]    dem_decimation: 1
> [INFO]    dem_euclidean_map: False
> [INFO]    dem_gapfill_steps: 3
> [INFO]    dem_resolution: 5
> [INFO]    depthmap_resolution: 640
> [INFO]    dsm: False
> [INFO]    dtm: False
> [INFO]    end_with: odm_postprocess
> [INFO]    fast_orthophoto: False
> [INFO]    feature_quality: high
> [INFO]    feature_type: sift
> [INFO]    force_gps: False
> [INFO]    gcp: None
> [INFO]    geo: None
> [INFO]    gps_accuracy: 10
> [INFO]    ignore_gsd: False
> [INFO]    matcher_neighbors: 8
> [INFO]    matcher_type: flann
> [INFO]    max_concurrency: 24
> [INFO]    merge: all
> [INFO]    mesh_octree_depth: 11
> [INFO]    mesh_size: 200000
> [INFO]    min_num_features: 10000
> [INFO]    name: datasets
> [INFO]    optimize_disk_space: False
> [INFO]    orthophoto_compression: DEFLATE
> [INFO]    orthophoto_cutline: False
> [INFO]    orthophoto_kmz: False
> [INFO]    orthophoto_no_tiled: False
> [INFO]    orthophoto_png: False
> [INFO]    orthophoto_resolution: 5
> [INFO]    pc_classify: False
> [INFO]    pc_csv: False
> [INFO]    pc_ept: False
> [INFO]    pc_filter: 2.5
> [INFO]    pc_geometric: False
> [INFO]    pc_las: False
> [INFO]    pc_quality: medium
> [INFO]    pc_rectify: False
> [INFO]    pc_sample: 0
> [INFO]    pc_tile: False
> [INFO]    primary_band: auto
> [INFO]    project_path: /un
> [INFO]    radiometric_calibration: none
> [INFO]    rerun: None
> [INFO]    rerun_all: True
> [INFO]    rerun_from: None
> [INFO]    resize_to: 2048
> [INFO]    sfm_algorithm: incremental
> [INFO]    skip_3dmodel: False
> [INFO]    skip_band_alignment: False
> [INFO]    skip_orthophoto: False
> [INFO]    skip_report: False
> [INFO]    sm_cluster: None
> [INFO]    smrf_scalar: 1.25
> [INFO]    smrf_slope: 0.15
> [INFO]    smrf_threshold: 0.5
> [INFO]    smrf_window: 18.0
> [INFO]    split: 999999
> [INFO]    split_image_groups: None
> [INFO]    split_overlap: 150
> [INFO]    texturing_data_term: gmi
> [INFO]    texturing_keep_unseen_faces: False
> [INFO]    texturing_outlier_removal_type: gauss_clamping
> [INFO]    texturing_skip_global_seam_leveling: False
> [INFO]    texturing_skip_local_seam_leveling: False
> [INFO]    texturing_tone_mapping: none
> [INFO]    tiles: False
> [INFO]    time: False
> [INFO]    use_3dmesh: False
> [INFO]    use_exif: False
> [INFO]    use_fixed_camera_params: False
> [INFO]    use_hybrid_bundle_adjustment: False
> [INFO]    verbose: False
> [INFO]    ==============
> [INFO]    Rerun all -- Removing old data
> [INFO]    Running dataset stage
> [INFO]    Loading dataset from: /un/datasets/images
> [INFO]    Loading 1734 images
> [INFO]    Wrote images database: /un/datasets/images.json
> [INFO]    Found 1734 usable images
> [INFO]    Parsing SRS header: WGS84 UTM 35N
> [INFO]    Finished dataset stage
> [INFO]    Running split stage
> [INFO]    Normal dataset, will process all at once.
> [INFO]    Finished split stage
> [INFO]    Running merge stage
> [INFO]    Normal dataset, nothing to merge.
> [INFO]    Finished merge stage
> [INFO]    Running opensfm stage
> [INFO]    Maximum photo dimensions: 4056px
> [INFO]    Photo dimensions for feature extraction: 2028px
> [INFO]    nvidia-smi detected
> [INFO]    Using GPU for extracting SIFT features
> [INFO]    Altitude data detected, enabling it for GPS alignment
> [INFO]    ['use_exif_size: no', 'flann_algorithm: KDTREE', 'feature_process_size: 2028', 'feature_min_frames: 10000', 'processes: 24', 'matching_gps_neighbors: 8', 'matching_gps_distance: 0', 'matching_graph_rounds: 50', 'optimize_camera_parameters: yes', 'reconstruction_algorithm: incremental', 'undistorted_image_format: tif', 'bundle_outlier_filtering_type: AUTO', 'sift_peak_threshold: 0.066', 'align_orientation_prior: vertical', 'triangulation_type: ROBUST', 'retriangulation_ratio: 2', 'bundle_compensate_gps_bias: yes', 'matcher_type: FLANN', 'feature_type: SIFT_GPU', 'use_altitude_tag: yes', 'align_method: auto', 'local_bundle_radius: 0']
> [INFO]    Wrote reference_lla.json
> [INFO]    running "/code/SuperBuild/install/bin/opensfm/bin/opensfm" detect_features "/un/datasets/opensfm"
> 2022-03-21 15:20:51,250 INFO: Capping memory usage to ~ 25373.705078125 MB
> 2022-03-21 15:20:51,250 INFO: Expecting to process 200 images.

smathermather, 2022-03-21:
If it isn't possible to share the dataset, it is much harder to reproduce and apply a patch. But perhaps someone else will have a similar issue with a shareable dataset so we can troubleshoot properly.

madisi98, 2022-03-22:
I tried downloading an open source dataset, I downloaded odm_data_toledo and same thing happened.

Here is the link to the dataset:

> https://github.com/OpenDroneMap/odm_data_toledo

And here is the output in the shell with the command included:

> manu@manuPC:~/projects/smart_camp/datasets$ time docker run -ti --rm --gpus all --ipc=host -v /home/manu/projects/project/datasets/:/workspace opendronemap/odm:gpu --project-path /workspace odm_data_toledo --dsm --dtm --min-num-features 15000 --matcher-neighbors 20 --dem-resolution 2.0 --feature-quality high --rerun-all
> [INFO]    DTM is turned on, automatically turning on point cloud classification
> [INFO]    Initializing ODM 2.8.1 - Tue Mar 22 15:07:29  2022
> [INFO]    ==============
> [INFO]    auto_boundary: False
> [INFO]    boundary: {}
> [INFO]    build_overviews: False
> [INFO]    camera_lens: auto
> [INFO]    cameras: {}
> [INFO]    cog: False
> [INFO]    copy_to: None
> [INFO]    crop: 3
> [INFO]    debug: False
> [INFO]    dem_decimation: 1
> [INFO]    dem_euclidean_map: False
> [INFO]    dem_gapfill_steps: 3
> [INFO]    dem_resolution: 2.0
> [INFO]    depthmap_resolution: 640
> [INFO]    dsm: True
> [INFO]    dtm: True
> [INFO]    end_with: odm_postprocess
> [INFO]    fast_orthophoto: False
> [INFO]    feature_quality: high
> [INFO]    feature_type: sift
> [INFO]    force_gps: False
> [INFO]    gcp: None
> [INFO]    geo: None
> [INFO]    gps_accuracy: 10
> [INFO]    ignore_gsd: False
> [INFO]    matcher_neighbors: 20
> [INFO]    matcher_type: flann
> [INFO]    max_concurrency: 24
> [INFO]    merge: all
> [INFO]    mesh_octree_depth: 11
> [INFO]    mesh_size: 200000
> [INFO]    min_num_features: 15000
> [INFO]    name: odm_data_toledo
> [INFO]    optimize_disk_space: False
> [INFO]    orthophoto_compression: DEFLATE
> [INFO]    orthophoto_cutline: False
> [INFO]    orthophoto_kmz: False
> [INFO]    orthophoto_no_tiled: False
> [INFO]    orthophoto_png: False
> [INFO]    orthophoto_resolution: 5
> [INFO]    pc_classify: True
> [INFO]    pc_csv: False
> [INFO]    pc_ept: False
> [INFO]    pc_filter: 2.5
> [INFO]    pc_geometric: False
> [INFO]    pc_las: False
> [INFO]    pc_quality: medium
> [INFO]    pc_rectify: False
> [INFO]    pc_sample: 0
> [INFO]    pc_tile: False
> [INFO]    primary_band: auto
> [INFO]    project_path: /workspace
> [INFO]    radiometric_calibration: none
> [INFO]    rerun: None
> [INFO]    rerun_all: True
> [INFO]    rerun_from: None
> [INFO]    resize_to: 2048
> [INFO]    sfm_algorithm: incremental
[... 131 lines cut ...]
> 2022-03-22 15:08:07,016 INFO: Extracting ROOT_SIFT features for image DJI_0061.JPG
> 2022-03-22 15:08:07,043 DEBUG: Computing sift with threshold 0.066
> 2022-03-22 15:08:07,179 INFO: Reading data for image DJI_0095.JPG (queue-size=1
> 2022-03-22 15:08:07,180 INFO: Extracting ROOT_SIFT features for image DJI_0087.JPG
> 2022-03-22 15:08:07,204 DEBUG: Computing sift with threshold 0.066
> 2022-03-22 15:08:07,231 INFO: Reading data for image 1JI_0072.JPG (queue-size=1
> 2022-03-22 15:08:07,232 INFO: Extracting ROOT_SIFT features for image DJI_0064.JPG
> 2022-03-22 15:08:07,255 DEBUG: Computing sift with threshold 0.066
> 2022-03-22 15:08:07,307 INFO: Reading data for image 1JI_0088.JPG (queue-size=1
> 2022-03-22 15:08:07,307 INFO: Extracting ROOT_SIFT features for image DJI_0099.JPG
> 2022-03-22 15:08:07,321 DEBUG: Found 10620 points in 1.2514674663543701s
> 2022-03-22 15:08:07,321 DEBUG: reducing threshold
> 2022-03-22 15:08:07,322 DEBUG: Computing sift with threshold 0.044000000000000004
> 2022-03-22 15:08:07,338 INFO: Reading data for image 1JI_0082.JPG (queue-size=1
> 2022-03-22 15:08:07,339 INFO: Extracting ROOT_SIFT features for image DJI_0082.JPG
> 2022-03-22 15:08:07,345 DEBUG: Computing sift with threshold 0.066
> 2022-03-22 15:08:07,381 DEBUG: Found 21491 points in 1.3121676445007324s
> 2022-03-22 15:08:07,382 DEBUG: done
> 2022-03-22 15:08:07,394 DEBUG: Computing sift with threshold 0.066
> 2022-03-22 15:08:07,595 INFO: Reading data for image DJI_0060.JPG (queue-size=1
> 2022-03-22 15:08:07,598 INFO: Reading data for image DJI_0093.JPG (queue-size=2
> 2022-03-22 15:08:07,737 INFO: Reading data for image 1JI_0062.JPG (queue-size=3
> 2022-03-22 15:08:07,770 INFO: Reading data for image DJI_0081.JPG (queue-size=4
> ^C[WARNING] Caught TERM/INT signal, attempting to exit gracefully...
> [WARNING] Sending TERM signal to PID 148...

pierotofy, 2022-07-25:
This *should* be fixed with https://github.com/OpenDroneMap/ODM/commit/55a7821058229cab735e2447daa83a86ff65e7cd. :clinking_glasses: 

Try to update, reprocess and re-open if the issue persists? :pray:

closed 2022-07-25 by pierotofy