OpenDroneMap/ODM #1468: Docker Image build fails

NightRaven-spec, 2022-06-06:
Running docker build -t [tags] . using an unmodified clone of ODM (https://github.com/OpenDroneMap/ODM) failed setting up GDAL with pip.

Building a very slightly modified ODM the same way resulted in a lot of errors, while the docker pull... command works just fine.

Hardware:
Apple : M1 MacBook Air, 8gb ram, 512gb SSD
Windows11 : Intel i5 7th gen 7300HQ, NVIDIA GTX 1050 4gb, 16gb Ram (Laptop), 1tb Storage(non SSD)
Ubuntu : Running Virtual machine on Mac M1

I have tried running the clone on all the above machines. 

This issue can be reproduced while building an Unmodified clone of the ODM using the DockerFile provided using the docker build -t . The issue every time is a Different one, I will try uploading the logs later on. Sometimes it is an OpenCv2 error or a PCL error, Make Build Failed.

pierotofy, 2022-06-16:
> Building a very slightly modified ODM

Are you able to build a vanilla clone of ODM? Perhaps there's an issue with your fork.

pierotofy, 2022-06-16:
Maybe a network issue? pip downloads several requirements from the internet.

rrowlands, 2022-06-16:
I'm having an issue when building with Docker pretty much immediately. None of the "command -v" checks are working and it attempts (and fails) to install sudo, among other things.

```
Sending build context to Docker daemon  1.155MB
Step 1/13 : FROM ubuntu:21.04 AS builder
 ---> 7cc39f89fa58
Step 2/13 : ENV DEBIAN_FRONTEND=noninteractive     PYTHONPATH="$PYTHONPATH:/code/SuperBuild/install/lib/python3.9/dist-packages:/code/SuperBuild/install/lib/python3.8/dist-packages:/code/SuperBuild/install/bin/opensfm"     LD_LIBRARY_PATH="$LD_LIBRARY_PATH:/code/SuperBuild/install/lib"
 ---> Using cache
 ---> 02a129dbbed4
Step 3/13 : WORKDIR /code
 ---> Using cache
 ---> 498414bee0de
Step 4/13 : COPY . ./
 ---> Using cache
 ---> 19b57014737a
Step 5/13 : RUN bash configure.sh install
 ---> Running in 8239d6b552de
Updating the system
Installing sudo
Get:1 http://security.ubuntu.com/ubuntu hirsute-security InRelease [110 kB]
Get:2 http://archive.ubuntu.com/ubuntu hirsute InRelease [269 kB]
Err:1 http://security.ubuntu.com/ubuntu hirsute-security InRelease
  gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
Err:2 http://archive.ubuntu.com/ubuntu hirsute InRelease
  gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
Get:3 http://archive.ubuntu.com/ubuntu hirsute-updates InRelease [115 kB]
Err:3 http://archive.ubuntu.com/ubuntu hirsute-updates InRelease
  gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
Get:4 http://archive.ubuntu.com/ubuntu hirsute-backports InRelease [101 kB]
Err:4 http://archive.ubuntu.com/ubuntu hirsute-backports InRelease
  gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
Reading package lists...
W: GPG error: http://security.ubuntu.com/ubuntu hirsute-security InRelease: gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
E: The repository 'http://security.ubuntu.com/ubuntu hirsute-security InRelease' is not signed.
W: GPG error: http://archive.ubuntu.com/ubuntu hirsute InRelease: gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
E: The repository 'http://archive.ubuntu.com/ubuntu hirsute InRelease' is not signed.
W: GPG error: http://archive.ubuntu.com/ubuntu hirsute-updates InRelease: gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
E: The repository 'http://archive.ubuntu.com/ubuntu hirsute-updates InRelease' is not signed.
W: GPG error: http://archive.ubuntu.com/ubuntu hirsute-backports InRelease: gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
E: The repository 'http://archive.ubuntu.com/ubuntu hirsute-backports InRelease' is not signed.
E: Unable to locate package sudo
Installing lsb_release
configure.sh: line 46: sudo: command not found
Installing pkg-config
configure.sh: line 51: sudo: command not found
Installing tzdata
configure.sh: line 55: sudo: command not found
configure.sh: line 57: lsb_release: command not found
Installing Python PIP
configure.sh: line 66: sudo: command not found
configure.sh: line 69: sudo: command not found
configure.sh: line 70: sudo: command not found
configure.sh: line 7: lsb_release: command not found
You are not on Ubuntu 21.04 (detected: )
It might be possible to run ODM on a newer version of Ubuntu, however, you cannot rely on this script.
The command '/bin/sh -c bash configure.sh install' returned a non-zero code: 1
```

rrowlands, 2022-06-16:
I apparently have run into a bug in the Docker runtime itself, as per:

https://stackoverflow.com/questions/66319610/gpg-error-in-ubuntu-21-04-after-second-apt-get-update-during-docker-build

pierotofy, 2022-07-21:
I'll close this for the time being, feel free to re-open if the issue persists.

closed 2022-07-21 by pierotofy