OpenDroneMap/ODM #1473: geo.txt should accept yaw/pitch/roll values, not omega/phi/kappa

pierotofy, 2022-06-17:
We currently set omega/phi/kappa values when reading from `geo.txt` files, but that's incorrect. We should set ypr angles and convert them for end users.

originlake, 2022-06-20:
@pierotofy I'm working on #1415(finally get a chance to do it), figure I could add this as well. One problem is that will the ypr value in geo.txt take the camera make into consideration? As the ypr is treated differently in different make. If not need to consider this, call `compute_opk` after updating the geo entry should do the work.

https://github.com/OpenDroneMap/ODM/blob/157a2eb90a652f33fa8c91f4c0bb6c06ecb20388/opendm/photo.py#L426-L431

pierotofy, 2022-06-20:
> If not need to consider this, call compute_opk after updating the geo entry should do the work. 

Yep! :+1:

pierotofy, 2022-06-20:
Thanks for looking into this!

closed 2022-06-20 by pierotofy