OpenDroneMap/ODM #1479: Possible Bug: Rolling Shutter Distortion Correction applied only to primary band image

Saijin-Naib, 2022-06-21:
As per ckato3 here: 
https://community.opendronemap.org/t/important-feature-update-rolling-shutter-distortion-correction/12032/3?u=saijin_naib

It seems like only the primary band image from a set of mult-band images (one image per channel) gets corrected, leading to compounded misalignment between bands in the dataset.

pierotofy, 2022-06-21:
I'm not sure this is really a bug; we don't perform correction on the image pixels, just on the feature extraction process, so the warp matrix calculation should not be affected at all.

A dataset to test issues with multispectral issues and rolling shutter (with before and after screenshots) would be nice to see any issues.

Saijin-Naib, 2022-06-21:
Sounds good. I'll follow up with the reporter and ask them to give test data and results here.

ckato3, 2022-06-22:
Stripe-like borders appear on the NDVI image from a grid pattern flight with multiple turns to the north and south.
This phenomenon does not appear in subsets of images taken only northward or only southward.
So I suppose this come from that inter-band warp matrices are different for northward or southward due to rolling shutter distortion.

This boundary is barely noticeable in the red and near-infrared images.
It is noticeable only  calculated NDVI images.

Also, this phenomenon is not seen with Sequoia cameras using GLOBAL shutter multi-band cameras.
It is noticeable on the DJI P4 Multispectral which I believe use ROLLING shutter cameras.

I will upload sample images, probably NEXT week.

pierotofy, 2022-06-22:
But is this an issue related to turning on `--rolling-shutter` (e.g. does it work better if you don't turn on `--rolling-shutter`?)

ckato3, 2022-06-22:
I'm a bit confused.

Looking at the ODM log file, if the --rolling-shutter option is enabled, I see
     [INFO]   running "/code/SuperBuild/   ...   /opensfm"   rs_correct   "/code/opensfm"
     ......
     INFO: Correcting observations...
     INFO: Writing corrected and trimmed features for DJI_0873.TIFF

It reads as if an image file with rolling shutter correction is being written out at this opensfm stage.

If that is true, then changing "rs_correct" to apply the correction to all band images, 
Then, at the later stage, the band alignment will be performed using the rolling shutter corrected images, and the warp matrix will be obtained.
(The band alignment can be obtained without the effects of rolling shutter distortion.)

Or, is the image file itself unaltered at "opensfm rs_correcrt" stage, and does not affect subsequent calculations ?

pierotofy, 2022-06-22:
> It reads as if an image file with rolling shutter correction is being written out at this opensfm stage.

It writes *features*, not *images*.

>  is the image file itself unaltered

Correct!

closed 2022-06-22 by pierotofy