OpenDroneMap/ODM #1481: No report

sbonaime, 2022-06-22:
### How did you install OpenDroneMap? (Docker, natively, ...)?
docker image opendronemap/odm             latest    0b4f762b980e
debian 10.12

### What is the problem?

No report is generated with this docker image. With last image  of last month, I did get a report. I did not change my worflow :

````
/usr/bin/docker run  --rm -v  "$input_image_dir/:/code/images" \
-v "$output_dir/odm_georeferencing:/code/odm_georeferencing"  \
-v "$output_dir/odm_meshing:/code/odm_meshing"  \
-v "$output_dir/odm_dem:/code/odm_dem" \
-v "$output_dir/odm_orthophoto:/code/odm_orthophoto" \
-v "$output_dir/odm_texturing:/code/odm_texturing" \
-v "$output_dir/odm_report:/code/odm_report" \
opendronemap/odm --orthophoto-png --mesh-octree-depth 13 \
--auto-boundary    --pc-quality ultra --matcher-neighbors 40\
 --dsm   --orthophoto-kmz  --camera-lens brown  --orthophoto-cutline \
 --mesh-size 300000  \
 --feature-quality ultra --max-concurrency 16  --use-hybrid-bundle-adjustment \
 --build-overviews --time --min-num-features 64000 --feature-type $feature_type
````

sbonaime, 2022-06-22:
this issue can be closed ! I was not looking in the correct directory ... :upside_down_face:

closed 2022-06-22 by pierotofy

Saijin-Naib, 2022-06-22:
[image]