OpenDroneMap/ODM #1483: Add support for P4 camera+sun calibration

pierotofy, 2022-06-23:
Currently the Phantom 4 multispectral sun sensor calibration does not work properly. Some modifications to the code need to be implemented to support it.

Reference: https://dl.djicdn.com/downloads/p4-multispectral/20200717/P4_Multispectral_Image_Processing_Guide_EN.pdf

Community thread: https://community.opendronemap.org/t/dji-p4m-segmented-orthophoto/12039/15

Christoffer-Svenningsson, 2022-07-06:
I was curious whether ODM actually was reading all the meta-data from images coming from P4 correctly and noticed that there are several members in a loaded ODM_photo instance set to None or 0 which probably should not be. I am trying to add XMP-tags where i see fit in ODM_photo but I am very new to working with this kind of data and I struggle to translate what everything is. 

For instance: 
Is DLS (yaw pitch roll) same as Irrradiance (yaw pitch roll)?
Is DarkLevel same as BlackLevel?
What is horizontal irradiance and how does it compare to irradiance?

I am searching for information but if anyone has input it would be helpful!

pierotofy, 2022-07-27:
I don't think you actually need to compute $\rho_x$ directly, to my understanding the white paper boils down to implementing:

[image]

`XCamera = dn_to_radiance code`
`pCam_X  = sensor gain adjustment from XMP`
`XLS * pLS_X = spectral_irradiance` (MicaSense calls this horizontal irradiance)

I've added some changes with #1513 and I'm running some tests to see if the results look plausible.

pierotofy, 2022-07-27:
And yes I don't think we want to use `Camera:RadiometricCalibration`.

Christoffer-Svenningsson, 2022-07-27:
> I don't think you actually need to compute ρx directly, to my understanding the white paper boils down to implementing:
> 
> [image]
> 
> `XCamera = dn_to_radiance code` `pCam_X = sensor gain adjustment from XMP` `XLS * pLS_X = spectral_irradiance` (MicaSense calls this horizontal irradiance, also read from XMP)

Yes this is what i tried to do aswell. (My code was horrible which is why i did not share it by the way). This makes the normalized indices such as NDVI look ok. But the individual bands still seems off. Maybe I am missing something here? Or do we simply ignore the individual bands?

pierotofy, 2022-07-27:
Can you post some screenshots?

I would also recommend to test #1513 and see how it compares to what you wrote.

Christoffer-Svenningsson, 2022-07-27:
I will attempt to connect to the computer in which I have the images and get them. Give me a few minutes!

pierotofy, 2022-07-27:
I also think the polynomial coefficients for the Phantom 4 are stored inverted compared to other sensors like the Micasense.

Christoffer-Svenningsson, 2022-07-27:
This is what  qgis "default" min/max colorthreshhold gives. Note that there does not seem to be a common ceil for the values? How can i even determine this when representing the image
[image]

This is what i get if i set the same colortreshhold.
[image]

This is with global seam leveling on.
[image]

It should be mentioned that this was a sunny day and pretty much all colors should be similar to the lighter patch (more or less) as seen in the first image

Christoffer-Svenningsson, 2022-07-27:
> I also think the polynomial coefficients for vignetting for the Phantom 4 are stored inverted compared to other sensors like the Micasense.

Yes, from my understanding we do not multiply by 1/V(x,y) but instead just V(x,y)

pierotofy, 2022-07-27:
Correct.

I'm getting decent results with #1513

[image]

[image]

I think I'll go ahead and merge it, that way more people can start testing it out.

closed 2022-07-27 by pierotofy

Christoffer-Svenningsson, 2022-07-27:
> Correct.
> 
> I'm getting decent results with #1513
> 
> [image]
> 
> [image]
> 
> I think I'll go ahead and merge it, that way more people can start testing it out. We can fix other issues as they come out.

This was a better example of the miscoloring for the RGB bands i was talking about, sorry i could not find a better example. Anyways, great job and thank you for fixing this again!

Christoffer-Svenningsson, 2022-07-28:
This is what i got with #1513. It's looking kind of strange. 
[image]

NDVI in grey scale
[image]

I dont know. Maybe this dataset is just bad for some reason

pierotofy, 2022-07-28:
Maybe. Can you share the images?

Christoffer-Svenningsson, 2022-07-28:
> Maybe. Can you share the images?

You mean the raw data right? Im just waiting for an OK to share it just to be certain

Christoffer-Svenningsson, 2022-07-28:
Hi, here is the dataset. It's my first time using dronedb. I hope i made it public so that you can view it!
The flight was automatic and with 75% side overlap and 80% front overlap.
[DroneDB P4M Test dataset](https://hub.dronedb.app/r/ffop/p4m-75_overlap-testdata)

pierotofy, 2022-07-29:
There's definitely some value intensity issue going on with this dataset, but I'm not sure if this is something that we're not correcting properly in ODM, or if the sensor just wasn't calibrated properly prior to data capture, or something else.

I've checked the code and we're doing everything that DJI documented to do, so I'm not sure.

pierotofy, 2022-07-29:
One thing I made sure to check is that mvs-texturing is not causing the artifacts, I manually overlaid the RGB images after radiometric-calibration and this is what came out of it:

[image]

It just doesn't look correct, but I'm not sure how to adjust it (especially given that other datasets seem to work).

pierotofy, 2022-07-29:
In particular, both sensor gain and exposure time between the good/bad shots are widely different:

```
DJI_0771.TIF
drone-dji:SensorGain="3.750"
drone-dji:ExposureTime="3040"

drone-dji:SensorGain="1.000"
drone-dji:ExposureTime="283"
```

I'm not sure why that is, but it's probably the cause.

Christoffer-Svenningsson, 2022-07-29:
> In particular, both sensor gain and exposure time between the good/bad shots are widely different:
> 
> ```
> DJI_0771.TIF
> drone-dji:SensorGain="3.750"
> drone-dji:ExposureTime="3040"
> 
> DJI_0311.TIF
> drone-dji:SensorGain="1.000"
> drone-dji:ExposureTime="283"
> ```
> 
> I'm not sure why that is, but it's probably the cause.

So the testdata i have been testing my changes to has been bad... Thats painful to hear. I wonder if the drone is broken. Thank you for this investigation anyway. Right now im out of ideas on what to investigate. But I will write here if anything comes up!

MendeBadra, 2025-04-16:
I don't get it. Is the camera+sun option supported for DJI Phantom 4 drones?

EvaGith, 2025-04-16:
In Pix4D I can select those options...but not sure how to make it work in ODM

[image]

Specifications: 
The DJI Phantom 4 Multispectral drone has a sunlight sensor that helps to improve the consistency of data collection at different times of the day

MendeBadra, 2025-04-16:
I am trying camera + sun for my dataset on ODM. Did you get Pix4D license by trial?

EvaGith, 2025-04-16:
Are you able to obtain reflectance values in the output of the processing with camera+sun? I never did in ODM. I got the chance of contrasting my attempts in ODM with some colleagues at the university who had access to an educational Pix4D licence. Please let me know about your findings! I have been wondering how to solve it in ODM for almost a year now!

MendeBadra, 2025-04-16:
I have finished processing. pc-quality: ultra was a bad idea, my pc memory filled. After that I got a patchy image but I believe it's due to `texturing-skip-global-seam-leveling` (I kinda read a lot of thread and mixed up which option contributes to which)

[image]

I will let you know about result without `texturing-skip-global-seam-leveling` tomorrow. (By the way one of my drone captures is producing this uneven lighting conditions no matter what (here is other image that I mosaicked)):

[image]

EvaGith, 2025-04-17:
I also managed mosaicking (different processing options, different outcomes). Visual inspection was ok. However, the histogram and the pixel value ranges were odd (not correct, histogram showed a weird distribution, and values were not in reflectances range (0-1) at all). :-( how about yours?

MendeBadra, 2025-04-17:
My computer froze multiple times. I have queued another run. I'll let you know the results.

EvaGith, 2025-04-18:
> I finished orthomosaicing and I don't know why in the middle, area of sudden bright spot appeared. Take a look:
>
> image.png (view on web)
>
> My parameters are:
> auto-boundary:true, feature-quality:ultra, mesh-octree-depth:12, mesh-size:600000, optimize-disk-space:true, orthophoto-resolution:0.001, pc-quality:high, radiometric-calibration:camera+sun, rolling-shutter:true
>
> I finished orthomosaicing and I don't know why in the middle, area of sudden bright spot appeared. Take a look:
>
> image.png (view on web)
>
> My parameters are:
> auto-boundary:true, feature-quality:ultra, mesh-octree-depth:12, mesh-size:600000, optimize-disk-space:true, orthophoto-resolution:0.001, pc-quality:high, radiometric-calibration:camera+sun, rolling-shutter:true

Did you check the histograms? What are the range values? It's not only a matter of visual inspection. You should obtain values between 0-1 in the pixels and a proper values distribution. I never managed though...trying different processing options for the phantom 4 multiespectral didn't render reflectances ...weird histograms all the time 

Yahoo Mail: Busca, organiza, conquista

MendeBadra, 2025-04-20:
Hello, there is the histogram values and some boxplots showing max and min: (I would say they're pretty normal looking histograms)

[image]

<div>
<style scoped>
    .dataframe tbody tr th:only-of-type {
        vertical-align: middle;
    }

    .dataframe tbody tr th {
        vertical-align: top;
    }

    .dataframe thead th {
        text-align: right;
    }
</style>
<table border="1" class="dataframe">
  <thead>
    <tr style="text-align: right;">
      <th></th>
      <th>Band</th>
      <th>Min</th>
      <th>Max</th>
      <th>Mean</th>
      <th>Median</th>
      <th>Std Dev</th>
      <th>Count</th>
    </tr>
  </thead>
  <tbody>
    <tr>
      <th>0</th>
      <td>Red (Band 1)</td>
      <td>0.000010</td>
      <td>0.801670</td>
      <td>0.111174</td>
      <td>0.104933</td>
      <td>0.038385</td>
      <td>72930440</td>
    </tr>
    <tr>
      <th>1</th>
      <td>Green (Band 2)</td>
      <td>0.000037</td>
      <td>0.657114</td>
      <td>0.103182</td>
      <td>0.098467</td>
      <td>0.032011</td>
      <td>72931495</td>
    </tr>
    <tr>
      <th>2</th>
      <td>Blue (Band 3)</td>
      <td>0.000153</td>
      <td>0.551923</td>
      <td>0.104978</td>
      <td>0.099838</td>
      <td>0.031800</td>
      <td>72931862</td>
    </tr>
    <tr>
      <th>3</th>
      <td>NIR (Band 4)</td>
      <td>0.000064</td>
      <td>0.819185</td>
      <td>0.197075</td>
      <td>0.187860</td>
      <td>0.060731</td>
      <td>72931525</td>
    </tr>
    <tr>
      <th>4</th>
      <td>RedEdge (Band 5)</td>
      <td>0.000005</td>
      <td>0.724430</td>
      <td>0.141161</td>
      <td>0.134643</td>
      <td>0.045160</td>
      <td>72930929</td>
    </tr>
    <tr>
      <th>5</th>
      <td>Band 6 (Band 6)</td>
      <td>255.000000</td>
      <td>255.000000</td>
      <td>255.000000</td>
      <td>255.000000</td>
      <td>0.000000</td>
      <td>72931873</td>
    </tr>
  </tbody>
</table>
</div>

[image]

EvaGith, 2025-04-21:
Yes, it looks fine indeed...however visually there are brightness heterogeneities. I am not sure how to correct for those...I'll give it a try using your processing parameters over one of my flights (also covering a shrubland area) to compare with the ones you are applying. I didn't even get reflectance values (and my histograms are shameful!). I thought the issue was some ODM developments incompatibility with the Phantom4MS, but you did manage (at least numerically...)
Maybe other users can shed some light into this? 
I'll be back when I manage to process the flight.
thanks for sharing! Cheers,
Eva
Yahoo Mail: Busca, organiza, conquista 

image.png (view on web)

|  | Band | Min | Max | Mean | Median | Std Dev | Count |
| 0 | Red (Band 1) | 0.000010 | 0.801670 | 0.111174 | 0.104933 | 0.038385 | 72930440 |
| 1 | Green (Band 2) | 0.000037 | 0.657114 | 0.103182 | 0.098467 | 0.032011 | 72931495 |
| 2 | Blue (Band 3) | 0.000153 | 0.551923 | 0.104978 | 0.099838 | 0.031800 | 72931862 |
| 3 | NIR (Band 4) | 0.000064 | 0.819185 | 0.197075 | 0.187860 | 0.060731 | 72931525 |
| 4 | RedEdge (Band 5) | 0.000005 | 0.724430 | 0.141161 | 0.134643 | 0.045160 | 72930929 |
| 5 | Band 6 (Band 6) | 255.000000 | 255.000000 | 255.000000 | 255.000000 | 0.000000 | 72931873 |

 
image.png (view on web)

image.png (view on web)

|  | Band | Min | Max | Mean | Median | Std Dev | Count |
| 0 | Red (Band 1) | 0.000010 | 0.801670 | 0.111174 | 0.104933 | 0.038385 | 72930440 |
| 1 | Green (Band 2) | 0.000037 | 0.657114 | 0.103182 | 0.098467 | 0.032011 | 72931495 |
| 2 | Blue (Band 3) | 0.000153 | 0.551923 | 0.104978 | 0.099838 | 0.031800 | 72931862 |
| 3 | NIR (Band 4) | 0.000064 | 0.819185 | 0.197075 | 0.187860 | 0.060731 | 72931525 |
| 4 | RedEdge (Band 5) | 0.000005 | 0.724430 | 0.141161 | 0.134643 | 0.045160 | 72930929 |
| 5 | Band 6 (Band 6) | 255.000000 | 255.000000 | 255.000000 | 255.000000 | 0.000000 | 72931873 |

 
image.png (view on web)

EvaGith, 2025-04-25:
I don't know what to say: after trying different processing options, I am not sure what's best. The reflectances values change for each one and I am not certain how to decide which is best. One of the combinations I used is: --auto-boundary--feature-quality ultra--optimize-disk-space--orthophoto-resolution 5--dem-resolution 1 --min-num-features 24000 --radiometric-calibration camera+sun--texturing-skip-global-seam-leveling --orthophoto-no-tiled --orthophoto-compression DEFLATE--use-exif--pc-quality high--texturing-data-term--mesh-octree-depth 12--mesh-size 600000
I see that using the command line in ODM instead of WebODM might help computationally?
About the artifacts, does anybody know how to smooth them somehow?

MendeBadra, 2025-05-01:
> Thanks for the very interesting thread. I need some guidance about how to use the orthomosaic output to derive NDVI and other spectral indices. The output is float32 reflectance values, but since they are not in the 0-1 values range, how should I proceed? Does the orthomosaic need some normalization or any other further processing? It's my first time processing drone imagery and I am trying to figure out a suitable processing pipeline. Thanks again!

I think you should proceed as is. I think normalization is not required because reflectance values should reflect measurement values. As for orthos that I've shown above, I tried Agisoft metashape and got similar results. But in metashape you can manually select which image should be taken into by hand. I manually excluded overexposed image and got a little better result. But still, the image looks like waves of sea.

MendeBadra, 2025-05-01:
I think DJI p4 's sun sensor values are very varied in my flight.

PR #1513 (2022-07-27): P4 Multispectral Support
Adds support for Phantom 4 Multispectral

Fixes #1483