OpenDroneMap/ODM #1490: not in G

sbonaime, 2022-07-01:
Debian 11.3
Docker image  2621cea2881e from 2022-06-25T00:07:59.16312362Z

I am processing more than ten batches of pictures from a recent field trip with a PH4RTK. For one batch I have the following error :
```
2022-07-01 09:02:15,150 INFO: Image 100_0099_0017.JPG could not be added (not enough overlap with other images)
2022-07-01 09:02:15,150 INFO: Image 100_0099_0146.JPG could not be added (not enough overlap with other images)
2022-07-01 09:02:15,150 INFO: Creating reconstruction poses from homography graph
Traceback (most recent call last):
  File "/code/SuperBuild/install/bin/opensfm/bin/opensfm_main.py", line 25, in <module>
    commands.command_runner(
  File "/code/SuperBuild/install/bin/opensfm/opensfm/commands/command_runner.py", line 38, in command_runner
    command.run(data, args)
  File "/code/SuperBuild/install/bin/opensfm/opensfm/commands/command.py", line 13, in run
    self.run_impl(data, args)
  File "/code/SuperBuild/install/bin/opensfm/opensfm/commands/reconstruct.py", line 14, in run_impl
    reconstruct.run_dataset(dataset)
  File "/code/SuperBuild/install/bin/opensfm/opensfm/actions/reconstruct.py", line 21, in run_dataset
    report, reconstructions = reconstruction.planar_reconstruction(
  File "/code/SuperBuild/install/bin/opensfm/opensfm/reconstruction.py", line 1967, in planar_reconstruction
    for im, graph_path in nx.single_source_shortest_path(graph, central_image).items():
  File "/usr/local/lib/python3.9/dist-packages/networkx/algorithms/shortest_paths/unweighted.py", line 333, in single_source_shortest_path
    raise nx.NodeNotFound(f"Source {source} not in G")
networkx.exception.NodeNotFound: Source 100_0099_0112.JPG not in G

===== Dumping Info for Geeks (developers need this to fix bugs) =====
Child returned 1
Traceback (most recent call last):
  File "/code/stages/odm_app.py", line 94, in execute
    self.first_stage.run()
  File "/code/opendm/types.py", line 347, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 347, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 347, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 328, in run
    self.process(self.args, outputs)
  File "/code/stages/run_opensfm.py", line 38, in process
    octx.reconstruct(args.rolling_shutter, self.rerun())
  File "/code/opendm/osfm.py", line 55, in reconstruct
    self.run('reconstruct')
  File "/code/opendm/osfm.py", line 34, in run
    system.run('"%s" %s "%s"' %
  File "/code/opendm/system.py", line 106, in run
    raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 1

===== Done, human-readable information to follow... =====

[91m[ERROR]   Uh oh! Processing stopped because of strange values in the reconstruction. This is often a sign that the input data has some issues or the software cannot deal with it. Have you followed best practices for data acquisition? See https://docs.opendronemap.org/flying/[0m
```

command line I use is 
```
/usr/bin/docker run  --rm -v  "$input_image_dir/:/code/images" \
-v "$output_dir/odm_georeferencing:/code/odm_georeferencing"  \
-v "$output_dir/odm_meshing:/code/odm_meshing"  \
-v "$output_dir/odm_dem:/code/odm_dem" \
-v "$output_dir/odm_orthophoto:/code/odm_orthophoto" \
-v "$output_dir/odm_texturing:/code/odm_texturing" \
-v "$output_dir/odm_report:/code/odm_report" \
opendronemap/odm --orthophoto-png --mesh-octree-depth 13 \
--auto-boundary    --pc-quality ultra --orthophoto-resolution 0.01 \
 --dsm   --orthophoto-kmz  --camera-lens brown  --orthophoto-cutline \
--sfm-algorithm planar --fast-orthophoto --matcher-neighbors 10\
 --mesh-size 300000  \
 --feature-quality ultra --max-concurrency 16  --use-hybrid-bundle-adjustment \
 --build-overviews --time --min-num-features 64000 --feature-type sift
```

I use this command line  on previous batches with success and I also used another odm script with those files without error

pierotofy, 2022-07-01:
> How can we reproduce this? (What steps did you do to trigger the problem? What parameters are you using for processing? If possible please include a copy of your dataset uploaded on Google Drive or Dropbox. Be detailed)

sbonaime, 2022-07-01:
Unfortunately I can't share the data...

pierotofy, 2022-07-26:
Closing as we cannot reproduce this.

closed 2022-07-26 by pierotofy