OpenDroneMap/ODM #1491: Garbled characters in tracks.csv

CosmosHua, 2022-07-04:
### How did you install ODM? (Docker, installer, natively, ...)?

Docker, version=2.8.6/2.8.5, Ubuntu 18.04, python 3.6

### What is the problem?

Can't read 'opensfm/tracks.csv' using python open(), because there are lots of garbled characters in it.

>   with open(odm+'/opensfm/tracks.csv', encoding='utf-8') as f: tracks = f.readlines()[1:]
>   File "/usr/lib/python3.6/codecs.py", line 321, in decode
>     (result, consumed) = self._buffer_decode(data, self.errors, final)
> UnicodeDecodeError: 'utf-8' codec can't decode byte 0xf2 in position 54: invalid continuation byte

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

In version=2.7.1, it is ok, but when I update ODM docker images to version=2.8.6/2.8.5, this problem occurs.

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

Just run ODM docker version=2.8.6/2.8.5, and try to read 'opensfm/tracks.csv', you will see it.

CosmosHua, 2022-07-04:
I found you have binarized the tracks file, but how should I decode tracks.csv? It can't be directly decoded by bytes.decode(). Could you supply some scripts, or tell me how to decode?

pierotofy, 2022-07-04:
Here's the code that decodes it https://github.com/OpenDroneMap/OpenSfM/blob/286/opensfm/src/map/src/tracks_manager.cc#L195

For all other questions, let's move the conversation over to https://community.opendronemap.org :pray:

closed 2022-07-04 by pierotofy