OpenDroneMap/ODM #1492: ODM doesn’t work when path has a special character

frankowskii, 2022-07-06:
I am having an issue running ODM when the install path has a ’ character in it. This is the output I get for the install path “C:\Temp\Tes’t\ODM”. I have a case where a Windows user name has that character, which makes it impossible to install ODM anywhere inside that user’s directory.

It seems to be similar to this thread ([Non ascii character in ODM Install Path 1](https://community.opendronemap.org/t/non-ascii-character-in-odm-install-path/11020)), but I guess that this specific issue hasn’t been fixed there.

OUTPUT:

```
/ ____ _____ __ __
| / __ \ | __ \ | / | |
| | | | | | | | | | \ / | |
| | | | | | | | | | |/| | |
| | || | | || | | | | | |
| _/ |/ || || |
__________________________/
version: 2.8.6

Traceback (most recent call last):
File “C:\Temp\Tes’t\ODM\run.py”, line 15, in
from opendm.utils import get_processing_results_paths, rm_r
File “C:\Temp\Tes’t\ODM\opendm\utils.py”, line 3, in
from opendm.photo import find_largest_photo_dims
File “C:\Temp\Tes’t\ODM\opendm\photo.py”, line 17, in
from opendm import get_image_size
File “C:\Temp\Tes’t\ODM\opendm\get_image_size.py”, line 2, in
import cv2
File “C:\Temp\Tes’t\ODM\venv\lib\site-packages\cv2_init_.py”, line 96, in
bootstrap()
File “C:\Temp\Tes’t\ODM\venv\lib\site-packages\cv2_init_.py”, line 59, in bootstrap
load_first_config([
File “C:\Temp\Tes’t\ODM\venv\lib\site-packages\cv2_init_.py”, line 53, in load_first_config
exec_file_wrapper(fpath, g_vars, l_vars)
File “C:\Temp\Tes’t\ODM\venv\lib\site-packages\cv2\load_config_py3.py”, line 8, in exec_file_wrapper
code = compile(f.read(), os.path.basename(fpath), ‘exec’)
File “config-3.8.py”, line 1
PYTHON_EXTENSIONS_PATHS = [r’C:\Temp\Tes’t\ODM\venv\lib\site-packages\cv2\python-3.8’] + PYTHON_EXTENSIONS_PATHS
^
SyntaxError: invalid syntax
```

Reference: https://community.opendronemap.org/t/odm-doesnt-work-when-path-has-a-special-character/12223/3

coumbs, 2022-08-21:
Using triple quotes for multiline should solve the issue I think.
I will do some tests and open a pull request during the week.

coumbsek, 2022-08-21:
Using triple quotes for multiline should solve the issue I think.
I will do some tests and open a pull request during the week.

closed 2022-08-25 by pierotofy