OpenDroneMap/ODM #1500: DJI Mini 3 pro has two rolling shutter times depending on shooting mode

ejrobinson, 2022-07-12:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For questions and discussion please open a topic on http://community.opendronemap.org/c/opendronemap. 

Please use the format below to report bugs and faults. 
****************************************

### How did you install ODM? (Docker, installer, natively, ...)?

Windows Native Installer

### What is the problem?

DJI Mini 3 Pro has two vastly different rolling shutter readout times in 12MP vs 48MP modes, but the 'camera model' is the same so WebODM currently has no way to differentiate 

12 MP is apporx 26 ms, 48 MP is approx 60 ms, Images attached
[image]
[image]

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

Apply two different corrections depending on the shooting resolution

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

I can't see anything else in the EXIF to check against that would work for this but I do think users should be either given a warning, or a choice when rolling shutter correction is enabled and the EXIF shows the images are from a Mini 3 Pro as the difference is significant.

Saijin-Naib, 2022-07-13:
Quad-bayer sensors, the gift that keep on giving -_-

Thank you for surfacing this! Something we need to look out for on all the other quad-bayer sUAS now.

pierotofy, 2022-07-21:
Mm, this is tricky, we can't even look at image size because the images could be resized (e.g. by WebODM).

pierotofy, 2022-07-21:
There's a `Megapixels` tag that keeps the megapixels information:

```
Megapixels                      : 48.8
```

```
Megapixels                      : 12.2
```

pierotofy, 2022-07-25:
So long as the width/height in the EXIF tags is left unaltered we can derive a megapixels estimate, from there #1511 addresses this with dynamic lookup values based on a function expression.

PR #1511 (2022-07-25): Dynamic rolling shutter readout db
Fixes #1500﻿
closed 2022-07-25 by pierotofy