OpenDroneMap/ODM #1506: Fails when processing ~2gb worth of imagery

fosteman, 2022-07-19:
How did you install OpenDroneMap? (Docker, natively, ...)?

Docker

What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Ubuntu 20.04 in Google Cloud with Nvidia Tesla T4

What is the problem?
- Stopping during sift

How can we reproduce this?
Using frontend, uploading ~700 files

[core-dump.txt](https://github.com/OpenDroneMap/ODM/files/9144996/core-dump.txt)

fosteman, 2022-07-19:
`[INFO] Initializing ODM 2.8.8 - Tue Jul 19 21:50:07 2022
[INFO] ==============
[INFO] 3d_tiles: False
[INFO] auto_boundary: False
[INFO] auto_boundary_distance: 0
[INFO] boundary: {}
[INFO] build_overviews: False
[INFO] camera_lens: auto
[INFO] cameras: {}
[INFO] cog: False
[INFO] copy_to: None
[INFO] crop: 3
[INFO] debug: False
[INFO] dem_decimation: 1
[INFO] dem_euclidean_map: False
[INFO] dem_gapfill_steps: 3
[INFO] dem_resolution: 5
[INFO] depthmap_resolution: 640
[INFO] dsm: False
[INFO] dtm: False
[INFO] end_with: odm_postprocess
[INFO] fast_orthophoto: False
[INFO] feature_quality: high
[INFO] feature_type: sift
[INFO] force_gps: False
[INFO] gcp: None
[INFO] geo: None
[INFO] gps_accuracy: 10
[INFO] ignore_gsd: False
[INFO] matcher_neighbors: 0
[INFO] matcher_type: flann
[INFO] max_concurrency: 16
[INFO] merge: all
[INFO] mesh_octree_depth: 11
[INFO] mesh_size: 200000
[INFO] min_num_features: 10000
[INFO] name: 78052c60-3f62-4a3d-b1cd-02f68325d0a0
[INFO] no_gpu: False
[INFO] optimize_disk_space: False
[INFO] orthophoto_compression: DEFLATE
[INFO] orthophoto_cutline: False
[INFO] orthophoto_kmz: False
[INFO] orthophoto_no_tiled: False
[INFO] orthophoto_png: False
[INFO] orthophoto_resolution: 5
[INFO] pc_classify: False
[INFO] pc_copc: False
[INFO] pc_csv: False
[INFO] pc_ept: False
[INFO] pc_filter: 2.5
[INFO] pc_geometric: False
[INFO] pc_las: False
[INFO] pc_quality: medium
[INFO] pc_rectify: False
[INFO] pc_sample: 0
[INFO] pc_tile: False
[INFO] primary_band: auto
[INFO] project_path: /var/www/data
[INFO] radiometric_calibration: none
[INFO] rerun: None
[INFO] rerun_all: False
[INFO] rerun_from: None
[INFO] resize_to: 2048
[INFO] rolling_shutter: False
[INFO] rolling_shutter_readout: 0
[INFO] sfm_algorithm: incremental
[INFO] skip_3dmodel: False
[INFO] skip_band_alignment: False
[INFO] skip_orthophoto: False
[INFO] skip_report: False
[... 126 lines cut ...]
File "/code/stages/odm_app.py", line 94, in execute
self.first_stage.run()
File "/code/opendm/types.py", line 347, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 347, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 347, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 328, in run
self.process(self.args, outputs)
File "/code/stages/run_opensfm.py", line 35, in process
octx.feature_matching(self.rerun())
File "/code/opendm/osfm.py", line 414, in feature_matching
self.match_features(rerun)
File "/code/opendm/osfm.py", line 419, in match_features
self.run('match_features')
File "/code/opendm/osfm.py", line 34, in run
system.run('"%s" %s "%s"' %
File "/code/opendm/system.py", line 106, in run
raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 134

===== Done, human-readable information to follow... =====

[ERROR] Uh oh! Processing stopped because of strange values in the reconstruction. This is often a sign that the input data has some issues or the software cannot deal with it. Have you followed best practices for data acquisition? See https://docs.opendronemap.org/flying/`

pierotofy, 2022-07-22:
Thanks for all the information.

I think based on the log SIFT_GPU is crashing at some point, which falls back to the CPU algorithm, however since the two are not interchangeable, we end up with half SIFT_GPU features and half SIFT, which the matcher doesn't really like.

I pushed a partial fix which at least cleans up any previous result from GPU processing, which gets the software to complete processing, although it's not a good solution, since we should really try to understand why SIFT GPU is crashing to begin with..

pierotofy, 2022-07-22:
It would be super-useful to get a copy of the core dump here (I'm unfortunately unable to reproduce this on my machine).

```
/code/SuperBuild/install/bin/opensfm/bin/opensfm: line 12: 66 Segmentation fault (core dumped) "$PYTHON" "$DIR"/opensfm_main.py "$@"
```

Johnny555555, 2022-07-22:
Ok here's my 19GB core dump file for you to download:
https://1drv.ms/u/s!AnbQHcX3KX2dhat3L0VbX0bgr1EPwg?e=nL9Z10

My system:
Ubuntu 20.4 LTS
HP Proliant DL380P
Dual Intel Xeon 20-core CPUs
512GB RAM
Nvidia K80 GPU (dual GPUs with 12GB RAM each)
1.5TB swap file
RAID (magnetic disks)

Data set:
Almost 7400 images at 20MPx (taken using a P4RTK)
90 GCPs

WebODM project task Settings:
auto-boundary: true, debug: true, dsm: true, dtm: true, smrf-threshold: 0.01, verbose: true

pierotofy, 2022-07-22:
Thanks @Johnny555555 :+1: can you push to docker hub your current docker image you used to generate the core dump? Or can you re-generate it by making sure to use this image? `opendronemap/odm:gpucore`

I tried to open the core file but I think I must be using a different build/version.

Johnny555555, 2022-07-22:
@pierotofy  
I downloaded the gpucore image but when I try to start it, it just exits immediately. Note that my server is running Ubuntu 20.4 LTS.

I am using this nodeodm:gpu image:
REPOSITORY                   TAG       IMAGE ID       CREATED         SIZE
opendronemap/nodeodm         gpu       05d97fae601d   2 months ago    5.88GB
which I installed using this command:
>docker run -dp 3001:3000 --gpus all --name nodeodmgpu opendronemap/nodeodm:gpu

I've pushed it to a public image in Docker Hub:
johnny555555/nodeodmgpu:nodeodmgpu_badzipfile_error

Let me know if you can access it.

pierotofy, 2022-07-23:
Got it, thank you.

I unfortunately wasn't able to open it. gdb keeps complaining. I also realized this is the core dump for the match_features command, which is not the root cause of the issue.

I think this is going to take too long to sort out this way. What would be needed is a small test case that can be reproduced quickly (and possibly, on multiple machines).

fosteman, 2022-07-23:
I'll dump the core and send it your way later today @pierotofy !

fosteman, 2022-07-23:
Here's the dataset of imagery I was trying to put through the standard settings of NodeODM:gpu

https://drive.google.com/drive/folders/1wCyE03UPRk5dwmQ35D20e9MrS4ClRkZt?usp=sharing

fosteman, 2022-07-23:
> [core-dump.txt](https://github.com/OpenDroneMap/ODM/files/9144996/core-dump.txt)

@pierotofy

pierotofy, 2022-07-24:
I was able to reproduce a segfault crash using a small module: https://gist.github.com/pierotofy/1277574c3a7fe4bab102a7dd4279f695

I was then able to fix the segmentation fault with the changes in https://github.com/OpenDroneMap/pypopsift/commit/96910582fa370eb2f166aff5d73a4e6c29dfdb57

I've committed https://github.com/OpenDroneMap/ODM/commit/55a7821058229cab735e2447daa83a86ff65e7cd which *should* fix this issue. If it works, there should be no warnings about failed GPU SIFT extraction and processing should continue to completion.

Please wait until the new GPU docker image is updated and try again? :pray:

fosteman, 2022-07-25:
Awesome, @pierotofy , the issue resolved. Though after the feature extraction is done, it takes superbly long time to actually reconstruct the images and it's only doing it 1 core at a time. I'm running a 16 core VM. Any way to improve that ? 
`Concurrency` is set to 16. 
Am I missing something ?

closed 2022-07-25 by fosteman

pierotofy, 2022-07-25:
Some parts of the pipeline are single threaded, other parts are multi-threaded. There's probably not much you can do. Glad the issue is fixed! Thanks for the reports and core dumps.

Johnny555555, 2022-07-25:
Thanks Piero, much appreciated for the fix!