OpenDroneMap/ODM #1523: Suspicious output in log files

minihop, 2022-08-09:
### How did you install ODM? (Docker, installer, natively, ...)?

docker-image latest

### What is the problem?

Suspicious output in log files 

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

I ran the latest docker container opendronemap/odm:latest. To download some data from aws I inserted my AWS session keys via exports:

export AWS_SECRET_ACCESS_KEY=....
export AWS_ACCESS_KEY_ID=....
export AWS_SESSION_TOKEN=.....

I downloaded all my data into the container using awscli
Afterwards i started the reconstruction process.

After undistorting the images i get the following output, which looks suspicious to me:
I seems like something is executed in AWS environment. Maybe some ressources are being created automatically???

Can someone can explain me whats going on there?
Thanks in advance
Christof

 
2022-08-09 17:46:18,857 DEBUG: Changing event name from creating-client-class.iot-data to creating-client-class.iot-data-plane
2022-08-09 17:46:18,859 DEBUG: Changing event name from before-call.apigateway to before-call.api-gateway
2022-08-09 17:46:18,861 DEBUG: Changing event name from creating-client-class.iot-data to creating-client-class.iot-data-plane
2022-08-09 17:46:18,862 DEBUG: Changing event name from creating-client-class.iot-data to creating-client-class.iot-data-plane
2022-08-09 17:46:18,864 DEBUG: Changing event name from creating-client-class.iot-data to creating-client-class.iot-data-plane
2022-08-09 17:46:18,865 DEBUG: Changing event name from request-created.machinelearning.Predict to request-created.machine-learning.Predict
2022-08-09 17:46:18,869 DEBUG: Changing event name from creating-client-class.iot-data to creating-client-class.iot-data-plane
2022-08-09 17:46:18,870 DEBUG: Changing event name from creating-client-class.iot-data to creating-client-class.iot-data-plane
2022-08-09 17:46:18,872 DEBUG: Changing event name from creating-client-class.iot-data to creating-client-class.iot-data-plane
2022-08-09 17:46:18,873 DEBUG: Changing event name from creating-client-class.iot-data to creating-client-class.iot-data-plane
2022-08-09 17:46:18,876 DEBUG: Changing event name from before-call.apigateway to before-call.api-gateway
2022-08-09 17:46:18,879 DEBUG: Changing event name from before-call.apigateway to before-call.api-gateway
2022-08-09 17:46:18,880 DEBUG: Changing event name from creating-client-class.iot-data to creating-client-class.iot-data-plane
2022-08-09 17:46:18,882 DEBUG: Changing event name from before-call.apigateway to before-call.api-gateway
2022-08-09 17:46:18,884 DEBUG: Changing event name from before-parameter-build.autoscaling.CreateLaunchConfiguration to before-parameter-build.auto-scaling.CreateLaunchConfiguration

pierotofy, 2022-08-09:
This doesn't look like our code (we have no support for AWS in ODM). If you made changes and have questions, let's move the conversation over to https://community.opendronemap.org :pray:

closed 2022-08-09 by pierotofy

seandburt, 2022-09-20:
+1 for this. No direct changes to the project. Similar situation and I cannot seem to find where these logs are coming from. They don't appear in `logs.json`