OpenDroneMap/ODM #1525: Add option to export OBJ using a single material

pierotofy, 2022-08-11:
Add a `--texturing-single-material` option that post-processes the 3D model output to use a single material (something similar to https://github.com/theFroh/imagepacker/blob/master/objuvpacker.py, but with support for multispectral TIFFs inputs, thus using rasterio).

theoway, 2022-08-13:
I'd like to help in this one. Please let me know which package are you planning to use for texturing. I'll integrate that in https://github.com/OpenDroneMap/ODM/blob/master/stages/odm_postprocess.py

pierotofy, 2022-08-14:
That would be awesome. :pray: 

I think you will need to modify https://github.com/theFroh/imagepacker/blob/master/objuvpacker.py to make it work with rasterio instead of PIL, write a module to bring into ODM (based on imagepacker), then call the module when required. It would be best to add it at the mvs_texturing step, replacing the multi-material model (we don't want two textured model outputs).

I doubt there's a pre-made package that does exactly what we need here.

theoway, 2022-08-14:
Please don't mind if it's a noob question, but why do we need rasterio instead of pillow? The module above uses `.obj` & `.mtl` file as input. It used `PIL` to extract images from `.mtl`. I'm assuming they'll be in jpg/png. Or will they be in `.tif` for our use case? 
Also if possible, please share a sample dataset so that I can fully understand the code and make necessary modifications.

pierotofy, 2022-08-14:
Because multispectral datasets use 16bit/32bit TIFFs, which are not supported by PIL.

pierotofy, 2022-08-14:
https://github.com/pierotofy/drone_dataset_micasense

(Pass `--radiometric-calibration camera` to generate 32bit TIFFs).

theoway, 2022-08-21:
I've created a public [repo](https://github.com/theoway/tiff-texture-packer) where I will add the code for the module. I'll get started in a day or two :)

theoway, 2022-08-21:
Please share a valid dataset. I'm not able to process this even with `min-num-features=32000`
```
[ERROR]   The program could not process this dataset using the current settings. 
Check that the images have enough overlap, that there are enough recognizable features 
and that the images are in focus. You could also try to increase the --min-num-features
parameter.The program will now exit.
```
Or it'd be awesome if someone can directly share the multispectral obj and mtl files.

pierotofy, 2022-08-22:
Mm, try this one otherwise: https://community.opendronemap.org/t/dji-phantom-4-pro-multipectral-w-o-rtk/4978 (or a subset of it, it's quite large for just testing).

theoway, 2022-08-29:
Created 3d models successfully. If anyone wants to use them, [download here](https://drive.google.com/drive/folders/1vUqmwus95xTtHvK8CS4Qkve-44C4BQDf?usp=sharing)
Now I can get started <3

theoway, 2022-09-09:
So far, everything is going well with rasterio integration. There's one thing that I noticed, the `.tif` texture files that I've generated from ODM don't have any `crs`, `transforms` etc. Is it normal??

pierotofy, 2022-09-09:
Yep! You can ignore rasterio's warning.

pierotofy, 2022-09-23:
Try:

```
big[:,x1:x2,y1:y2] = small
```

https://numpy.org/devdocs/user/basics.indexing.html

theoway, 2022-09-23:
@pierotofy Worked like charm! Now I can proceed further.

muratmaga, 2022-10-10:
Can I inquire about the status of this issue? We also very strongly need to be able to generate single texture map from ODM. Going through blender for this task is not a feasible solution for us. We are not developers but happy to help as much as can (testing feedback etc)...

theoway, 2022-10-23:
Just have to make minor changes, and it'll be ready for testing. I'll add instructions to run the code as well in the readme.

muratmaga, 2022-11-10:
We are getting ready to submit a paper that uses ODM and 3D Slicer to obtain 3D models of biological specimens. Since we can't use the multi-file texture maps in Slicer currently, we are converting the texture maps into vertex color in meshlab as a preprocessing step, which reduces their quality. Because of that step, textures look inferior to the ones generated by commercial photogrammetry software (which tpyically output a single texture file).  It will be great if this is incorporated prior to our paper (which is still some timeout), we stop doing the step that degrades, what is otherwise a great texture. 

Again happy to test and report back...

pierotofy, 2022-11-10:
@muratmaga if you need this in a hurry, perhaps you could offer to sponsor the development of this feature (and I'm sure @theoway would appreciate it).

muratmaga, 2022-11-11:
I would be happy to sponsor (to a certain amount), but unfortunately it is not as easy as doing a paypal or patreon or something like that as @smathermather can attest.

theoway, 2022-11-12:
Please feel free to test it out. I have added instructions on how to use. **Note:** This still under progress. Some small changes have to be made. I'll get to them in a few days. 
Please feel free to add issues if you see any bugs. Keep the issues detailed so it'd be easier for me to debug.

And I'd like to thank all of you for your patience and support! Waiting for your feedback.

muratmaga, 2022-11-13:
@theoway thank you very much. 

I tried with one of my samples, and while the command worked (i.e., produced an output), I have trouble loading the textures. I opened an issue at your repo. I also think that tiff file is not quite standard (it looks like a multichannel image, with three channels containing the identical RGB image). The GH issue contains the dataset I experimented and the commands and outputs.

pierotofy, 2023-01-11:
Thanks all for your help/contributions. 

This has now been implemented in #1582. :pray:

closed 2023-01-11 by pierotofy

smathermather, 2023-01-11:
@muratmaga -- check it out... .

muratmaga, 2023-01-13:
Works great. Thank you all. This has been a great addition for us...

smathermather, 2023-01-13:
I just really appreciate Murat's first example in slicermorph, including QR code "ground" control points for accurate measurements:
[image]

muratmaga, 2023-01-13:
It works even nicer and accurate, when the photography is done more carefully (this was just a quick test). Thanks again everyone, much appreciate this feature and the background removal (this is raw output from ODM, no post-processing beyond what's provided). 

[image]

PR #1582 (2023-01-11): Add --texturing-single-material
Closes #1525