OpenDroneMap/ODM #1535: Unable to process this dataset

ankitatcodingnebula, 2022-09-28:
### How did you install ODM? (Docker, installer, natively, ...)?

Docker

### What is the problem?

Hello ODM Team,

Thanks for the great project. I need your guidance with a dataset which is failing ortho creation. For the attached sample dataset I keep getting could not process error. I can see in logs that it finds keypoints and also able to find matches in images.

```
.
.
......
2022-09-28 06:58:59,334 DEBUG: Found 86954 points in 1.321786642074585s
2022-09-28 06:58:59,334 DEBUG: reducing threshold
2022-09-28 06:58:59,334 DEBUG: Computing sift with threshold 6.698730897528758e-05
2022-09-28 06:58:59,597 DEBUG: Found 88145 points in 1.280003309249878s
2022-09-28 06:58:59,597 DEBUG: reducing threshold
2022-09-28 06:58:59,597 DEBUG: Computing sift with threshold 6.698730897528758e-05
2022-09-28 06:58:59,797 DEBUG: Found 88017 points in 1.2654829025268555s
2022-09-28 06:58:59,797 DEBUG: reducing threshold
2022-09-28 06:58:59,797 DEBUG: Computing sift with threshold 6.698730897528758e-05
2022-09-28 06:59:00,240 DEBUG: Found 86613 points in 1.2735114097595215s
2022-09-28 06:59:00,240 DEBUG: done
2022-09-28 06:59:00,304 DEBUG: Found 88036 points in 1.2787768840789795s
2022-09-28 06:59:00,304 DEBUG: done
2022-09-28 06:59:00,311 DEBUG: Found 85793 points in 1.280247449874878s
2022-09-28 06:59:00,311 DEBUG: done
2022-09-28 06:59:00,363 DEBUG: Found 87497 points in 1.2632832527160645s
2022-09-28 06:59:00,363 DEBUG: done
2022-09-28 06:59:00,550 DEBUG: Found 85819 points in 1.249007225036621s
2022-09-28 06:59:00,551 DEBUG: done
2022-09-28 06:59:00,642 DEBUG: Found 86954 points in 1.3076930046081543s
2022-09-28 06:59:00,642 DEBUG: done
2022-09-28 06:59:00,855 DEBUG: Found 88145 points in 1.2577219009399414s
2022-09-28 06:59:00,855 DEBUG: done
2022-09-28 06:59:01,053 DEBUG: Found 88017 points in 1.255828857421875s
2022-09-28 06:59:01,053 DEBUG: done
2022-09-28 06:59:03,508 DEBUG: No segmentation for 20220916-180837-2836-emergent_det_original_rgbcrop.jpg, no features masked.
2022-09-28 06:59:03,599 DEBUG: No segmentation for 20220916-180837-2835-emergent_det_original_rgbcrop.jpg, no features masked.
2022-09-28 06:59:03,789 DEBUG: No segmentation for 20220916-180838-2841-emergent_det_original_rgbcrop.jpg, no features masked.
2022-09-28 06:59:03,789 DEBUG: No segmentation for 20220916-180838-2838-emergent_det_original_rgbcrop.jpg, no features masked.
2022-09-28 06:59:03,794 DEBUG: No segmentation for 20220916-180837-2834-emergent_det_original_rgbcrop.jpg, no features masked.
2022-09-28 06:59:03,905 DEBUG: No segmentation for 20220916-180838-2839-emergent_det_original_rgbcrop.jpg, no features masked.
2022-09-28 06:59:04,102 DEBUG: No segmentation for 20220916-180837-2837-emergent_det_original_rgbcrop.jpg, no features masked.
2022-09-28 06:59:04,309 DEBUG: No segmentation for 20220916-180838-2840-emergent_det_original_rgbcrop.jpg, no features masked.
[INFO] running "/code/SuperBuild/install/bin/opensfm/bin/opensfm" match_features "/var/www/data/07a80dbc-d138-4497-b407-4e95a94d39ad/opensfm"
2022-09-28 06:59:06,724 INFO: Matching 28 image pairs
2022-09-28 06:59:06,731 INFO: Computing pair matching with 16 processes
2022-09-28 06:59:07,152 DEBUG: No segmentation for 20220916-180837-2834-emergent_det_original_rgbcrop.jpg, no features masked.
2022-09-28 06:59:07,153 DEBUG: No segmentation for 20220916-180838-2841-emergent_det_original_rgbcrop.jpg, no features masked.
2022-09-28 06:59:07,154 DEBUG: No segmentation for 20220916-180838-2839-emergent_det_original_rgbcrop.jpg, no features masked.
2022-09-28 06:59:07,155 DEBUG: No segmentation for 20220916-180837-2837-emergent_det_original_rgbcrop.jpg, no features masked.
2022-09-28 06:59:07,156 DEBUG: No segmentation for 20220916-180837-2836-emergent_det_original_rgbcrop.jpg, no features masked.
2022-09-28 06:59:07,164 DEBUG: No segmentation for 20220916-180838-2840-emergent_det_original_rgbcrop.jpg, no features masked.
2022-09-28 06:59:07,185 DEBUG: No segmentation for 20220916-180838-2838-emergent_det_original_rgbcrop.jpg, no features masked.
2022-09-28 06:59:07,628 DEBUG: No segmentation for 20220916-180837-2835-emergent_det_original_rgbcrop.jpg, no features masked.
2022-09-28 06:59:12,235 DEBUG: Matching 20220916-180838-2840-emergent_det_original_rgbcrop.jpg and 20220916-180838-2838-emergent_det_original_rgbcrop.jpg. Matcher: FLANN (symmetric) T-desc: 5.470 T-robust: 0.019 T-total: 5.499 Matches: 7097 Robust: 7017 Success: True
2022-09-28 06:59:12,240 DEBUG: Matching 20220916-180838-2838-emergent_det_original_rgbcrop.jpg and 20220916-180837-2834-emergent_det_original_rgbcrop.jpg. Matcher: FLANN (symmetric) T-desc: 5.476 T-robust: 0.029 T-total: 5.506 Matches: 82 Robust: 12 Success: False
2022-09-28 06:59:12,265 DEBUG: Matching 20220916-180837-2837-emergent_det_original_rgbcrop.jpg and 20220916-180838-2840-emergent_det_original_rgbcrop.jpg. Matcher: FLANN (symmetric) T-desc: 5.508 T-robust: 0.019 T-total: 5.527 Matches: 105 Robust: 12 Success: False
2022-09-28 06:59:12,926 DEBUG: Matching 20220916-180837-2834-emergent_det_original_rgbcrop.jpg and 20220916-180837-2836-emergent_det_original_rgbcrop.jpg. Matcher: FLANN (symmetric) T-desc: 6.167 T-robust: 0.014 T-total: 6.190 Matches: 6302 Robust: 6213 Success: True
2022-09-28 06:59:12,954 DEBUG: Matching 20220916-180837-2837-emergent_det_original_rgbcrop.jpg and 20220916-180837-2834-emergent_det_original_rgbcrop.jpg. Matcher: FLANN (symmetric) T-desc: 6.198 T-robust: 0.018 T-total: 6.216 Matches: 96 Robust: 12 Success: False
2022-09-28 06:59:12,994 DEBUG: Matching 20220916-180838-2840-emergent_det_original_rgbcrop.jpg and 20220916-180837-2834-emergent_det_original_rgbcrop.jpg. Matcher: FLANN (symmetric) T-desc: 6.236 T-robust: 0.023 T-total: 6.259 Matches: 99 Robust: 12 Success: False
2022-09-28 06:59:13,024 DEBUG: Matching 20220916-180838-2841-emergent_det_original_rgbcrop.jpg and 20220916-180838-2839-emergent_det_original_rgbcrop.jpg. Matcher: FLANN (symmetric) T-desc: 6.258 T-robust: 0.020 T-total: 6.290 Matches: 7013 Robust: 6952 Success: True
2022-09-28 06:59:13,218 DEBUG: Matching 20220916-180838-2838-emergent_det_original_rgbcrop.jpg and 20220916-180838-2839-emergent_det_original_rgbcrop.jpg. Matcher: FLANN (symmetric) T-desc: 6.373 T-robust: 0.089 T-total: 6.482 Matches: 19804 Robust: 19757 Success: True
2022-09-28 06:59:13,409 DEBUG: Matching 20220916-180838-2840-emergent_det_original_rgbcrop.jpg and 20220916-180837-2835-emergent_det_original_rgbcrop.jpg. Matcher: FLANN (symmetric) T-desc: 6.622 T-robust: 0.049 T-total: 6.672 Matches: 93 Robust: 12 Success: False
2022-09-28 06:59:13,461 DEBUG: Matching 20220916-180838-2838-emergent_det_original_rgbcrop.jpg and 20220916-180837-2835-emergent_det_original_rgbcrop.jpg. Matcher: FLANN (symmetric) T-desc: 6.669 T-robust: 0.039 T-total: 6.724 Matches: 82 Robust: 10 Success: False
2022-09-28 06:59:13,578 DEBUG: Matching 20220916-180838-2840-emergent_det_original_rgbcrop.jpg and 20220916-180838-2841-emergent_det_original_rgbcrop.jpg. Matcher: FLANN (symmetric) T-desc: 6.573 T-robust: 0.139 T-total: 6.840 Matches: 20224 Robust: 20175 Success: True
2022-09-28 06:59:13,639 DEBUG: Matching 20220916-180838-2840-emergent_det_original_rgbcrop.jpg and 20220916-180838-2839-emergent_det_original_rgbcrop.jpg. Matcher: FLANN (symmetric) T-desc: 6.617 T-robust: 0.242 T-total: 6.902 Matches: 20268 Robust: 20215 Success: True
[... 12 lines cut ...]
2022-09-28 06:59:16,547 DEBUG: Matching 20220916-180837-2837-emergent_det_original_rgbcrop.jpg and 20220916-180837-2835-emergent_det_original_rgbcrop.jpg. Matcher: FLANN (symmetric) T-desc: 3.318 T-robust: 0.006 T-total: 3.328 Matches: 7112 Robust: 7034 Success: True
2022-09-28 06:59:16,575 DEBUG: Matching 20220916-180838-2840-emergent_det_original_rgbcrop.jpg and 20220916-180837-2836-emergent_det_original_rgbcrop.jpg. Matcher: FLANN (symmetric) T-desc: 2.911 T-robust: 0.009 T-total: 2.920 Matches: 82 Robust: 11 Success: False
2022-09-28 06:59:16,661 DEBUG: Matching 20220916-180837-2834-emergent_det_original_rgbcrop.jpg and 20220916-180837-2835-emergent_det_original_rgbcrop.jpg. Matcher: FLANN (symmetric) T-desc: 3.203 T-robust: 0.033 T-total: 3.246 Matches: 18707 Robust: 18657 Success: True
2022-09-28 06:59:16,725 DEBUG: Matching 20220916-180837-2837-emergent_det_original_rgbcrop.jpg and 20220916-180838-2841-emergent_det_original_rgbcrop.jpg. Matcher: FLANN (symmetric) T-desc: 3.179 T-robust: 0.009 T-total: 3.188 Matches: 95 Robust: 13 Success: False
[... cut ...]
### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

An orthomosaic

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

Link to the dataset that I am trying with. Pease guide. Thanks
https://www.dropbox.com/s/isf4c9d38j9gsv9/ortho_data.zip?dl=0

pierotofy, 2022-09-28:
How did you capture these images? I notice they have a suspicious filename convention.

Always use raw, unedited pictures coming straight from your drone/camera. If not, the internal camera parameters will likely not be recovered (and reconstruction will fail).

ankitatcodingnebula, 2022-09-28:
Thanks for the response @pierotofy 

This is from an emergent camera (https://emergentvisiontec.com) deployed on a Altax drone (https://freeflysystems.com/alta-x). The images are captured, processed, exif added and saved via c++ code on the drone payload.
I suspect the issue could be that that some mandatory exif data might be missing, is their a list of which exif data fields is used by ODM?

ankitatcodingnebula, 2022-09-28:
@pierotofy Forgot to mention in my last comment. The images are actually saved in bayer tiff 8 bit format. Those weren't working either with ODM so I converted the tiff to RGB jpg with exact exif copied (the dataset I attached)

ankitatcodingnebula, 2022-09-28:
@pierotofy In case this helps. I ran another similar dataset but with more images on developer mode and below is the error

```
Fused depth-maps 0 (100%, 0ms)
23:05:56 [App     ] Densifying point-cloud completed: 0 points (81ms)
23:05:56 [App     ] MEMORYINFO: {
23:05:56 [App     ] 	VmPeak:	 2160780 kB
23:05:56 [App     ] 	VmSize:	 1262256 kB
23:05:56 [App     ] } ENDINFO
[INFO]    running "/code/SuperBuild/install/bin/OpenMVS/DensifyPointCloud" --filter-point-cloud -1 -i "/datasets/mydataset/opensfm/undistorted/openmvs/scene_dense.mvs" -v 0 --cuda-device -2 --geometric-iters 0
23:05:56 [App     ] Build date: Sep 28 2022, 22:44:05
23:05:56 [App     ] CPU: 11th Gen Intel(R) Core(TM) i9-11900H @ 2.50GHz (16 cores)
23:05:56 [App     ] RAM: 62.51GB Physical Memory 2.00GB Virtual Memory
23:05:56 [App     ] OS: Linux 5.15.0-48-generic (x86_64)
23:05:56 [App     ] Disk: 
23:05:56 [App     ] SSE & AVX compatible CPU & OS detected
23:05:56 [App     ] Command line: DensifyPointCloud --filter-point-cloud -1 -i /datasets/mydataset/opensfm/undistorted/openmvs/scene_dense.mvs -v 0 --cuda-device -2 --geometric-iters 0
23:05:56 [App     ] error: no valid point-cloud for the ROI estimation

===== Dumping Info for Geeks (developers need this to fix bugs) =====
Child returned 1
Traceback (most recent call last):
  File "/code/stages/odm_app.py", line 88, in execute
    self.first_stage.run()
  File "/code/opendm/types.py", line 378, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 378, in run
    self.next_stage.run(outputs)
  File "/code/opendm/types.py", line 378, in run
    self.next_stage.run(outputs)
  [Previous line repeated 1 more time]
  File "/code/opendm/types.py", line 359, in run
    self.process(self.args, outputs)
  File "/code/stages/openmvs.py", line 206, in process
    system.run('"%s" %s' % (context.omvs_densify_path, ' '.join(config + gpu_config + extra_config)))
  File "/code/opendm/system.py", line 109, in run
    raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 1

===== Done, human-readable information to follow... =====

[ERROR]   Uh oh! Processing stopped because of strange values in the reconstruction. This is often a sign that the input data has some issues or the software cannot deal with it. Have you followed best practices for data acquisition? See https://docs.opendronemap.org/flying/
```

ankitatcodingnebula, 2022-09-30:
> None of the EXIF fields are mandatory, (only the GPS lat/lon ones if you want georeferencing). https://github.com/OpenDroneMap/ODM/blob/master/opendm/photo.py#L195
> 
> You mentioned images are `captured, processed...`. What kind of processing are you performing?

@pierotofy In the processing pipeline I just add custom exif data in JSON format. Nothing else.

pierotofy, 2022-09-30:
Mm, I'm not sure then. Have you tried reconstructing a well textured scene (e.g. a checkerboard) with the same sensor? Did it work?

ankitatcodingnebula, 2022-10-04:
@pierotofy I have only tried with the high GSD images from the camera. Do you have any suggestion where should I look in the code for the focal length database entry? I am setup with the development environment and can debug. Thanks

pierotofy, 2022-10-17:
[image]

Several things to point out:

 - Overlap is probably insufficient (seems like you flew in a straight line, without sidelap?)
 - Images don't have many features (it's a field?). You either need much more overlap, or fly higher.
 - Is the sensor probably is not in the focal database, but that shouldn't be a problem if the other factors are OK. You can find the full list on the `sensor_data.json` https://github.com/OpenDroneMap/OpenSfM/blob/291/opensfm/data/sensor_data.json

In short, probably not a bug. But feel free to open a PR to add the proper focal length value if you find that it helps in the OpenSfM repository. :+1:

closed 2022-10-17 by pierotofy