OpenDroneMap/ODM #1536: Volume mesurment as in WebODM

IuliuNovac, 2022-09-28:
Hi @pierotofy, 

Could you give some pointers or sources on how the volume measurement was implemented in WebODM? I would like to run ODM directly but have an option of measuring the volume of a stockpile implemented in ODM.

I could do it directly by looking into WebODM code, but to save time&progress faster I would be grateful for some pointers. 

### How did you install ODM? (Docker, installer, natively, ...)?

Not Applicable.

### What is the problem?

Not Applicable.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

Not Applicable.

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

Not Applicable.

IuliuNovac, 2022-09-28:
@pierotofy 

I can see the script for the execution is in [calc_volume](https://github.com/OpenDroneMap/WebODM/blob/master/coreplugins/measure/calc_volume.py), but how exactly to pass the inputs is rather new to me. Could you give some example of files used below?

```
    Module("v.import", input=opts['area_file']<-THIS, output="polygon_area", overwrite=True)
    Module("v.import", input=opts['points_file']<-THIS, output="polygon_points", overwrite=True)
    Module("v.buffer", input="polygon_area"<-THIS, s=True, type="area", output="region", distance=1, minordistance=1, overwrite=True)
```

Saijin-Naib, 2022-09-28:
We are making calls to GRASS GIS modules:  
https://grass.osgeo.org/grass82/manuals/index.html

closed 2022-09-28 by pierotofy