OpenDroneMap/ODM #1542: When there’s RTK GCP’a isn’t used??

APOS80, 2022-10-15:
### How did you install ODM? (Docker, installer, natively, ...)?

WebODM on Windows

### What is the problem?

When processing images with RTK and GCP’s, GCP’s doesn’t seem to be used.

hights in resulting pointcloud is of by about 20 meters.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

As with no RTK and just GCP, accurate coordinates.

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

Try and see

pierotofy, 2022-10-15:
> How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

closed 2022-10-15 by pierotofy

APOS80, 2022-10-15:
I do not have this problem wen I don’t use rtk.

it’s not the first time this happens.

How is ODM supposed to handle it?

APOS80, 2022-10-15:
I’ve removed the xmp rtk information from the photos and doing a new attempt of reconstruction.