OpenDroneMap/ODM #1545: ODM can’t handle more than 3 GCP’s

APOS80, 2022-10-22:
### How did you install ODM? (Docker, installer, natively, ...)?

WebODM, windows

### What is the problem?

I’ve tried using more than 3 GCP’s before and had problems with the geo referencing being of by decimetres and now when I had to use more GCP’s because of a large dataset and split-merge I had to try it again. It if course failed with geo reference being of by 10-50 centimetres at some points.

trying a part of the dataset and just using three GCP gave me a perfect cloud but it’s not the whole thing I wished fore.

I know that you can make an image-group file but didn’t know how but today I found a way using QGis so I’m going to try creating groups with just three GCP’s for each submodel and hope that it works.

i would like to know though how the geo reference in ODM uses more than three GCP’s as of now because I think something is wrong.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

If more than three GCP’s is present an average georeference should be computed.

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

Just try it with a high quality dataset. Like something in an urban environment with a 80/75 overlap and GSD at 10mm.

pierotofy, 2022-10-22:
ODM supports an arbitrary number of GCPs. 

If you think there's an issue with georeferencing, or if you find any other issue with the software related to processing, you MUST include a copy of your dataset. :pray: 

Please, for the future, do not open further bug reports about problems related to processing without sharing your dataset. I will automatically close such reports in the future.

closed 2022-10-22 by pierotofy

APOS80, 2022-10-23:
But it’s not related to a specific dataset.