OpenDroneMap/ODM #1547: RuntimeError: Unable to cast Python instance to C++ type

teiminryu, 2022-10-27:
Hi, experts. Thank you in advance for reading my question.

Recently, I am running ODM to process multiple image datasets continuously, which means there is no stop from one dataset to the next. However, ODM sometimes shows me the following error when running dataset with more than 500 images:

opensfm/reconstruction.py", line 108, in bundle
report = pysfm.BAHelpers.bundle(
RuntimeError: Unable to cast Python instance to C++ type (compile in debug mode for details)

Is there anyone who knows the root cause and how to solve it? 
I am sure that ODM runs well for relatively smaller dataset, but the above error pops up in a sudden. (I guess it is caused by insufficient memory or too many caches) 

I would appreciate for guiding me how to resolve this issue.  (I've tried some ways, but useless)

Saijin-Naib, 2022-10-27:
Sorry for the trouble! Could you please start with giving me a bit more information about your system?

For instance:  

1) Operating System and Version
 eg: Windows 11, MacOS 12.4, Ubuntu Linux 22.04LTS, etc...
2) Hardware Specifications
 eg: 32GB RAM, i7-6700k, NVIDIA GTX 1050TI OC, 1TB SSD, etc...
3) WebODM Install Method 
 eg: Native installer, Docker, Snap, GitHub download, compiled from source
4) WebODM/ODM Version
 eg: WebODM v1.9.14 Build 64, ODM v2.8.7, etc... 
5) Procesing Node
 eg: Automatic, Lightning, local (node-odm-1), etc...
6) Screenshots demonstrating the issue/behavior/error messages

teiminryu, 2022-10-30:
Hi, thank you so much for your response. Here is my processing info,

1. OS: Ubuntu Linux 20.04
2. Hardware Specifications: 64GB RAM
3. I installed through Docker with GitHub download
4. ODM 2.9.0
5. Automatic processing 
Btw, sometimes two parallel ODM processes are implemented at the same time on a local computer. Might it be the cause?

Thank you in advance for your solutions.   
7.

pierotofy, 2022-12-13:
Closing, as we cannot reproduce this. Feel free to re-open by sharing a sample dataset. :pray:

closed 2022-12-13 by pierotofy