OpenDroneMap/ODM #1548: Canceling may not be working properly with multiple projects in queue

6monkeyrs, 2022-10-31:
### How did you install ODM? (Docker, installer, natively, ...)?

Window 10 installer

### What is the problem?

When I have multiple project on queue, and attempt to cancel the bigger one, canceling does not work properly.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

When I click on cancel, the status of the project goes back to a review mode, before submitting again.

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset uploaded to Google Drive or Dropbox (otherwise we cannot reproduce this).

Load multiple projects, start one, then start the others to place on queue. Once the reconstruction starts on the first, cancel it. See images attach of what I was experiencing. When I close the WebODM dashboard, and restarted, the next project started, but the first still remain in a canceling status state. It was then the other project was started, which was waiting on queue.

It was then when the cancel status was showing "Canceled"

I am prepare to give access to screenshots and projects images.

Saijin-Naib, 2022-11-01:
What version of WebODM is it? When you start WebODM, the blue splash screen will pop up with the version and build number on the bottom in white text.

6monkeyrs, 2022-11-10:
v1.9.16 build 73