OpenDroneMap/ODM #1549: Generate oblique orthophotos

chriswilley, 2022-11-04:
This would be an enhancement to ODM, the idea being that a user could generate one or more oblique orthomosaics (aka: "2.5D") of a scene provided (possibly) that the flight was conducted with sufficient oblique-ness. An example is here: https://oblique.sanborn.com/25sample_new/ (click the compass at the bottom to see the oblique versions of the scene).

A discussion of this feature is on [the Community Forum](https://community.opendronemap.org/t/creating-2-5d-oblique-orthophoto/13579).  In the example above it appears four orthos were created, one for each cardinal point; I think this would be the ideal way to have the feature work in ODM as well. Then as suggested by @smathermather in the Forum, create a nice way to view the oblique orthos in WebODM.

Looking for help/guidance on how to do this as I have never contributed code to this project, but I'm game to give it a go.

pierotofy, 2022-11-04:
This would be a really cool improvement for sure.

You would probably need to write a new module, similar to https://github.com/OpenDroneMap/odm_orthophoto, to render oblique imagery. You might find the help of a rendering engine library helpful here (something like https://github.com/Zielon/CPURasterizer), but maybe you can write it from scratch.. 

Once you can render oblique shots, you will need to render multiple shots from different areas and angles, but that's a second step.

chriswilley, 2022-11-04:
Yeah, you're confirming what I was thinking after I peeked at the code. The mesh-to-ortho is done in C++ which is outside my wheelhouse for sure (not to mention the geometry involved). Manipulating the mesh in Python before handing it to `odm_orthophoto` seems like it would require one or more big dependencies that I assume you would want to avoid; plus, not sure it would matter since my noob reading of the C++ indicates it's doing some 3D transforms to align the model before writing the TIFF out.

I agree it would be a nice feature; however, I think it's beyond my skills for the moment. Maybe another intrepid volunteer will pick this up.

coumbsek, 2022-12-13:
I just had a quick look. In a first time wouldn't it be possible to cheat by rotating the .obj with the blender `bpy` package already including (and the pointcloud if needed with pdal) and using this has the input of the actual `odm_orthophoto` stage ?

pierotofy, 2023-01-09:
I've looked a bit more into this, I guess the `odm_orthophoto` is sufficient for the task (I thought it wouldn't work):

[image]

The cleanest way to do this would be to add a parameter in `odm_orthophoto` to rotate the model around its center prior to rendering; the next steps would be to add the necessary lines of code to ODM to generate multiple orthophotos, crop, etc.

coumbsek, 2023-03-02:
Looking at this featur, I am wondering how to simply rotate the model.
Should we make our own function based on one of the version of `load_obj` or look for a lib that handle obj files and model and allow to rotate it ?

jakecoppinger, 2023-03-12:
As a workaround (and to start from a `.obj` file) I've written about a method of creating oblique orthophotos by rotating in Blender and then running `odm_orthophoto`, expanding on this forum comment: https://community.opendronemap.org/t/creating-2-5d-oblique-orthophoto/13579/4

I agree adding flags to `odm_orthophoto` to set rotation axis and rotation degrees would be optimal. Would a PR into https://github.com/OpenDroneMap/odm_orthophoto be the next step or is that a fork of the logic in this repo?

https://jakecoppinger.com/2023/03/generating-aerial-imagery-with-your-iphones-lidar-sensor/