OpenDroneMap/ODM #1564: Possible Bug: Qhull precision error: Initial simplex is flat

Saijin-Naib, 2022-12-12:
### How did you install ODM? (Docker, installer, natively, ...)?
Docker, Alpine Linux Latest-Stable
### What is the problem?
On a few (seemingly random) datasets, reconstruction will fail with an error in QHull about degenerate hyperplane(s). Each time there is nothing obviously wrong with the data, on the contrary, they're usually exceptionally well-captured data. The solution always is to completely strip EXIF Geolocation metadata, and then the reconstruction will succeed.
### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.
There should not be this error... I can't figure out exactly what is happening, but it seems like something with co-incident camera positions.
### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).
Example dataset:  
https://hub.dronedb.app/r/roofergaming/r4gzbklfs8ym8mjo
Example dataset thread:  
https://community.opendronemap.org/t/cannot-process-dataset-other-projects-without-problems/13268?u=saijin_naib
Something of note:  
```
I removed the GPS data from every 6th picture and the calculation was done without any problems.
```
Reconstruction with no EXIF Geolocation Metadata:  
[image]

Saijin-Naib, 2022-12-12:
Other instances:  
https://community.opendronemap.org/t/when-things-go-wrong/11005/5?u=saijin_naib
https://community.opendronemap.org/t/i-had-3-degenerate-hyperplanes-but-now-have-a-3d-model-after-removing-gps-from-exif/13071?u=saijin_naib

pierotofy, 2022-12-12:
The linked dataset does not have GPS, so this cannot be reproduced.

Saijin-Naib, 2022-12-12:
> Note: some images from the linked dataset do not have GPS (is this expected?)

Yes, the user modified them to drop GPS every 6th frame and that made it reconstruct.

I'll upload the unmodified images shortly, sorry.

Saijin-Naib, 2022-12-13:
Original images with full EXIF Geolocation Metadata:  
https://hub.dronedb.app/r/saijinnaib/roofer_gaming-13268

Saijin-Naib, 2022-12-13:
Another dataset: 
https://hub.dronedb.app/r/saijinnaib/gordon-14038

PR #1565 (2022-12-14): Automatic outputs alignment via CODEM
This PR adds the ability to automatically register a reconstruction (and resulting orthophotos, point clouds, 3D models, etc.) to a reference DSM (GeoTIFF, LAS or LAZ) by using CODEM (https://github.com/NCALM-UH/CODEM, credits to Craig Glennie and the group at CRREL). 

The details of the method are available at https://github.com/NCALM-UH/CODEM/blob/main/docs/details.md

Usage:

* Pass a file via `--align <path-to-tif-or-las-laz>`.
* Place a `align.tif`, or `align.laz` or `align.las` in the project root (similar to GEO and GCP files).

Sample output repor [... cut]
closed 2022-12-14 by pierotofy