OpenDroneMap/ODM #1569: Windows installation in non-default path causes error

pierotofy, 2022-12-15:
> installed at a non-default path (in my user dir) and tried to run run.bat but it failed because it couldn't find geos_c.dll at ./venv/Library/bin/geos_c.dll .  Copying it from the shapely site-packages to that location solved it though

coumbsek, 2023-02-17:
Strange, what command/dataset did you use ? Here we constantly instal ODM inside the AppData folder (so inside user dir) and we have not yet encounter this error.

pierotofy, 2023-02-17:
Just `run.bat` without a dataset. 

It's possible that this might have been related to something else (thus the "possible bug") label. Glad it's working in your case.

coumbsek, 2023-02-17:
I will try to reproduce next week

HeDo88TH, 2023-02-17:
My Windows user is called `IT 2` and I tried to install ODM in `C:\Users\IT 2\ODM`. It gave me this error on startup:

`Unable to create process using 'C:\Users\IT 2\ODM\\python38\python.exe -X utf8 "C:\Users\IT 2\ODM\\run.py" --help'`

When trying to run a reconstruction:

```
(venv) C:\Users\IT 2\ODM>run D:\datasets\brighton-2
Unable to create process using 'C:\Users\IT 2\ODM\\python38\python.exe -X utf8 "C:\Users\IT 2\ODM\\run.py" D:\datasets\brighton-2'
```

I think it's something with the building of the path.

When trying to install in `Program files` the installer lacks sufficient privileges:

[image]

It's at least an inconvenient to close the nasty error message and then right click on the installer -> run as administrator. I think the setup should be able to re-launch itself if the destination folder needs elevated privileges

seanrl, 2023-02-28:
Replication steps for this issue:
- Install ODM using the latest Windows binary release (I installed to ~/Programs/ODM)
- Open a powershell or cmd prompt window
- Execute "~/Programs/ODM/run.bat"

coumbsek, 2023-03-01:
> Replication steps for this issue in 3.0.4:
> 
>     * Install ODM using the latest Windows binary release (I installed to ~/Programs/ODM)
> 
>     * Open a powershell or cmd prompt window
> 
>     * Execute "~/Programs/ODM/run.bat"
> 
> 
> Its worth noting that the ODM Console doesn't have this issue. It only crops up if you use your own console to run the batch file. In the error, it lands on this line as being the culprit ` _lgeos = CDLL(os.path.join(sys.prefix, 'Library', 'bin', 'geos_c.dll'))`
> 
> A quick fix for me is to copy 2 files _geos.dll_ and _geos_c.dll_ from .\venv\Lib\site-packages\shapely\DLLs to .\venv\Library\bin (which has to be created because it doesn't exist)

This is normal, the ODM Console loads at lot of environment variables and configuration via the `winrunt.bat`and `win32env.bat`, if you want to use your own console you should have a look at it.

coumbsek, 2023-03-01:
> Just `run.bat` without a dataset.
> 
> It's possible that this might have been related to something else (thus the "possible bug") label. Glad it's working in your case.

Tried to reproduce it but everything seems fine. Even run a complete dataset with no issue while ODM is intall inside : `C:\Users\MyUser\Documents\ODM>`

coumbsek, 2023-03-01:
> My Windows user is called `IT 2` and I tried to install ODM in `C:\Users\IT 2\ODM`. It gave me this error on startup:
> 
> `Unable to create process using 'C:\Users\IT 2\ODM\\python38\python.exe -X utf8 "C:\Users\IT 2\ODM\\run.py" --help'`
> 
> I think it's something with the building of the path.

I tried with a space in path because I though it could be it but it worked. Are you sure the user launching ODM Console has read/wright access to the folder ?

To test if the issue comes from the building of the path just replace in `winrun.bat` the line 6 with this one:
`python -X utf8 "%ODMBASE%run.py" %*`
i.e. removing the extra `\`

PR #1627 (2023-04-01): Fix strange python launcher error when folder has spaces
This PR fixes the behavior of innosetup to use the windows short path when it contains spaces.
The issue seems to be related to https://github.com/pypa/setuptools/issues/216 and several others. I tested the same exact setup with newer python versions and they work fine. Unfortunately we are stuck with python 3.8 due to breaking changes in newer versions.

Bonus: added `vcpkg`, `venv`, `python38`, `dist` and `innosetup` folders to gitignore.

Closes #1569
closed 2023-04-01 by pierotofy