OpenDroneMap/ODM #1570: WedODM1.9.17(Engine3.0.2):"Cannot process dataset"

horsehaircrab, 2022-12-16:
### How did you install ODM? (Docker, installer, natively, ...)?

Manual Install(Docker Version 18.06.1-ce-win73 )
Windows10pro 

### What is the problem?

I've installed and used WebODM for 2years with no problem. 
yesterday,my laptop suddenly could not have a connection "./webodm.sh start" to localhost:8000 and log in.

So, I've done ./webodm.sh start && ./webodm.sh resetadminpassword newpass and had log in.
Then, I've never seen Error "Cannot process dataset" with below Task Output:

Traceback (most recent call last):
File "/code/run.py", line 18, in <module>
from stages.odm_app import ODMApp
File "/code/stages/odm_app.py", line 14, in <module>
from stages.odm_georeferencing import ODMGeoreferencingStage
File "/code/stages/odm_georeferencing.py", line 22, in <module>
from opendm.align import compute_alignment_matrix, transform_point_cloud, transform_obj
File "/code/opendm/align.py", line 4, in <module>
import codem
File "/usr/local/lib/python3.9/dist-packages/codem/__init__.py", line 5, in <module>
from codem.main import apply_registration
File "/usr/local/lib/python3.9/dist-packages/codem/main.py", line 28, in <module>
from codem.preprocessing.preprocess import GeoData
File "/usr/local/lib/python3.9/dist-packages/codem/preprocessing/preprocess.py", line 36, in <module>
import pdal
File "/code/SuperBuild/install/lib/python3.8/dist-packages/pdal/__init__.py", line 8, in <module>
inject_pdal_drivers()
File "/code/SuperBuild/install/lib/python3.8/dist-packages/pdal/drivers.py", line 66, in inject_pdal_drivers
drivers = libpdalpython.getDrivers()
RuntimeError: filesystem error: cannot make canonical path: Operation not permitted [.]

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

I want to revert to previous stable state.

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

another my WebODM1.8.1 installed desktop computer can process dataset. 
trigger is  WedODM1.9.17(Engine3.0.2)?

pierotofy, 2022-12-16:
Hi @horsehaircrab :wave: thanks for the report. 

Can you provide some information about the docker setup? Are you using the WSL 2 based engine?

horsehaircrab, 2022-12-20:
Thank you for your response.

my Docker Version 18.06.1-ce-win73 is different from  the WSL 2 based engine.
It's too old Docker ver.?

Therfore,I installed the latest Docker ver. that is probablely based on the WSL2.
So, I could fix it.

But, new WebODM1.9.17(engine3.0.2) has just a few  different points from my previous one that is based on WebODM1.8.1(engine2.4.x).  

At first, it takes much time, about three times(2hours) to make orthophoto with same images than previous one(just only 40 minuites).

the second,I feel 3D image preciseness of previous is just a little better than new(but almost no difference).

the 3rd,./webodm.sh start command in Git Bash could not stop and make pop-up another command prompt. 
I colud not help Pressing Ctrl+C instead of typing ./webodm.sh stop command.
Is it Ok?
[image]

Saijin-Naib, 2022-12-20:
You can fork the start of WebODM into the background by issuing the command like this:  
```
./webodm.sh start &
```

That way you don't have to worry about closing it with CTRL+C to get input back in your prompt.

pierotofy, 2023-01-03:
Should be fixed with #1577