OpenDroneMap/ODM #1571: Possible regression in OpenMVS, geometrically consistent views.

smathermather, 2022-12-17:
Here is the failure I am seeing. It's not the most enjoyable test dataset with 2k camera positions, but happy to share as needed.

Ubuntu 20.04, docker install, FWIW.

```
Geometric-consistent estimated depth-maps 1448 (72.11%, 27m13s, ETA 10m)...
Geometric-consistent estimated depth-maps 2008 (100%, 27m13s529ms)

===== Dumping Info for Geeks (developers need this to fix bugs) =====
Child returned 1
Traceback (most recent call last):
File "/code/stages/odm_app.py", line 81, in execute
self.first_stage.run()
File "/code/opendm/types.py", line 386, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 386, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 386, in run
self.next_stage.run(outputs)
[Previous line repeated 1 more time]
File "/code/opendm/types.py", line 365, in run
self.process(self.args, outputs)
File "/code/stages/openmvs.py", line 117, in process
raise e
File "/code/stages/openmvs.py", line 103, in process
run_densify()
File "/code/stages/openmvs.py", line 99, in run_densify
system.run('"%s" "%s" %s' % (context.omvs_densify_path,
File "/code/opendm/system.py", line 109, in run
raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 1

===== Done, human-readable information to follow... =====

[ERROR]   Uh oh! Processing stopped because of strange values in the reconstruction. This is often a sign that the input data has some issues or the software cannot deal with it. Have you followed best practices for data acquisition? See https://docs.opendronemap.org/flying/
```

pierotofy, 2022-12-17:
Would be interesting to patch openmvs.py and disable the geometric consistency to see if it processes through.

Also would be really helpful to narrow down a smaller subset of images that causes the crash.

smathermather, 2022-12-17:
Currently running through a local split-merge with `--split 200` to get sub 500-image submodels. We'll see if there are particular ones that fail.

Saijin-Naib, 2022-12-17:
> Currently running through a local split-merge with `--split 200` to get sub 500-image submodels. We'll see if there are particular ones that fail.
Is this stable enough to work almost like a binary bisect of datasets? If so, this needs to go into my mental toolkit...

smathermather, 2022-12-17:
The split is as solid as anything else. The biggest issue with split-merge is it needs rewritten to be more modular. As it is, it's really brittle for large datasets because anything going wrong in the submodels is going to cause failures globally. So we need more resilience for these failures largely through re-tries that are more atomic.

But yet: split works. Everything ends up symlinked in a tidy little submodels directory that can be copied elsewhere for testing. For now, I'm just letting it run through because that will also test all those submodels for me and help find the culprit(s).

smathermather, 2022-12-18:
> Also would be really helpful to narrow down a smaller subset of images that causes the crash.

At first blush, it seems it's not a pocket of images that trigger this. I'm wondering if it's an OOM or race condition with the full dataset:
```
[INFO]    11 valid DEM rasters to merge
[INFO]    Output bounds: 707154.5049 410352.0225 708429.2714 412409.1397250001
[INFO]    Output width: 25496, height: 41143
[INFO]    Adjusted bounds: 707154.5049 410351.98972500005 708429.3048999999 412409.1397250001
[INFO]    Cropping /datasets/kakuma_mission_17a_flight_01/odm_dem/dsm.tif
[INFO]    running gdalwarp -cutline /datasets/kakuma_mission_17a_flight_01/odm_georeferencing/odm_georeferenced_model.bounds.gpkg -crop_to_cutline -co TILED=YES -co COMPRESS=DEFLATE -co BLOCKXSIZE=512 -co BLOCKYSIZE=512 -co BIGTIFF=IF_SAFER -co NUM_THREADS=64  /datasets/kakuma_mission_17a_flight_01/odm_dem/dsm.original.tif /datasets/kakuma_mission_17a_flight_01/odm_dem/dsm.tif --config GDAL_CACHEMAX 48.3%
Creating output file that is 25495P x 41142L.
Processing /datasets/kakuma_mission_17a_flight_01/odm_dem/dsm.original.tif [1/1] : 0Using internal nodata values (e.g. -9999) for image /datasets/kakuma_mission_17a_flight_01/odm_dem/dsm.original.tif.
Copying nodata values from source /datasets/kakuma_mission_17a_flight_01/odm_dem/dsm.original.tif to destination /datasets/kakuma_mission_17a_flight_01/odm_dem/dsm.tif.
...10...20...30...40...50...60...70...80...90...100 - done.
[INFO]    Created /datasets/kakuma_mission_17a_flight_01/odm_dem/dsm.tif
[INFO]    Merging 11 shots.geojson files
[INFO]    Finished merge stage
[INFO]    Running odm_postprocess stage
[INFO]    Post Processing
[INFO]    Finished odm_postprocess stage
[INFO]    No more stages to run
[INFO]    MMMMMMMMMMMNNNMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMNNNMMMMMMMMMMM
[INFO]    MMMMMMdo:..---../sNMMMMMMMMMMMMMMMMMMMMMMMMMMNs/..---..:odMMMMMM
[INFO]    MMMMy-.odNMMMMMNy/`/mMMMMMMMMMMMMMMMMMMMMMMm/`/hNMMMMMNdo.-yMMMM
[INFO]    MMN/`sMMMMMMMMMNNMm/`yMMMMMMMMMMMMMMMMMMMMy`/mMNNMMMMMMMMNs`/MMM
[INFO]    MM/ hMMMMMMMMNs.+MMM/ dMMMMMMMMMMMMMMMMMMh +MMM+.sNMMMMMMMMh +MM
[INFO]    MN /MMMMMMNo/./mMMMMN :MMMMMMMMMMMMMMMMMM: NMMMMm/./oNMMMMMM: NM
[INFO]    Mm +MMMMMN+ `/MMMMMMM`-MMMMMMMMMMMMMMMMMM-`MMMMMMM:` oNMMMMM+ mM
[INFO]    MM..NMMNs./mNMMMMMMMy sMMMMMMMMMMMMMMMMMMo hMMMMMMMNm/.sNMMN`-MM
[INFO]    MMd`:mMNomMMMMMMMMMy`:MMMMMMMNmmmmNMMMMMMN:`hMMMMMMMMMdoNMm-`dMM
[INFO]    MMMm:.omMMMMMMMMNh/  sdmmho/.`..`-``-/sddh+  /hNMMMMMMMMdo.:mMMM
[INFO]    MMMMMd+--/osss+:-:/`  ```:- .ym+ hmo``:-`   `+:-:ossso/-:+dMMMMM
[INFO]    MMMMMMMNmhysosydmNMo   /ds`/NMM+ hMMd..dh.  sMNmdysosyhmNMMMMMMM
[INFO]    MMMMMMMMMMMMMMMMMMMs .:-:``hmmN+ yNmds -:.:`-NMMMMMMMMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMMMMMMMN.-mNm- //:::. -:://: +mMd`-NMMMMMMMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMMMMMMM+ dMMN -MMNNN+ yNNNMN :MMMs sMMMMMMMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMMMMMMM`.mmmy /mmmmm/ smmmmm``mmmh :MMMMMMMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMMMMMMM``:::- ./////. -:::::` :::: -MMMMMMMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMMMMMMM:`mNNd /NNNNN+ hNNNNN .NNNy +MMMMMMMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMMMMMMMd`/MMM.`ys+//. -/+oso +MMN.`mMMMMMMMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMMMMMMMMy /o:- `oyhd/ shys+ `-:s-`hMMMMMMMMMMMMMMMMMM
[INFO]    MMMMMMMMNmdhhhdmNMMM`  +d+ sMMM+ hMMN:`hh-  sMMNmdhhhdmNMMMMMMMM
[INFO]    MMMMMms:::/++//::+ho    .+- /dM+ hNh- +/`   -h+:://++/::/smMMMMM
[INFO]    MMMN+./hmMMMMMMNds-  ./oso:.``:. :-``.:os+-  -sdNMMMMMMmy:.oNMMM
[INFO]    MMm-.hMNhNMMMMMMMMNo`/MMMMMNdhyyyyhhdNMMMM+`oNMMMMMMMMNhNMh.-mMM
[INFO]    MM:`mMMN/-sNNMMMMMMMo yMMMMMMMMMMMMMMMMMMy sMMMMMMMNNs-/NMMm`:MM
[INFO]    Mm /MMMMMd/.-oMMMMMMN :MMMMMMMMMMMMMMMMMM-`MMMMMMMo-./dMMMMM/ NM
[INFO]    Mm /MMMMMMm:-`sNMMMMN :MMMMMMMMMMMMMMMMMM-`MMMMMNs`-/NMMMMMM/ NM
[INFO]    MM:`mMMMMMMMMd/-sMMMo yMMMMMMMMMMMMMMMMMMy sMMMs-/dMMMMMMMMd`:MM
[INFO]    MMm-.hMMMMMMMMMdhMNo`+MMMMMMMMMMMMMMMMMMMM+`oNMhdMMMMMMMMMh.-mMM
[INFO]    MMMNo./hmNMMMMMNms--yMMMMMMMMMMMMMMMMMMMMMMy--smNMMMMMNmy/.oNMMM
[INFO]    MMMMMms:-:/+++/:-+hMMMMMMMMMMMMMMMMMMMMMMMMMNh+-:/+++/:-:smMMMMM
[INFO]    MMMMMMMMNdhhyhdmMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMmdhyhhmNMMMMMMMM
[INFO]    MMMMMMMMMMMMMMMNNNNNMMMMMMNNNNNNMMMMMMMMNNMMMMMMMNNMMMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMMh/-...-+dMMMm......:+hMMMMs../MMMMMo..sMMMMMMMMMMMM
[INFO]    MMMMMMMMMMMM/  /yhy-  sMMm  -hhy/  :NMM+   oMMMy   /MMMMMMMMMMMM
[INFO]    MMMMMMMMMMMy  /MMMMN`  NMm  /MMMMo  +MM: .` yMd``` :MMMMMMMMMMMM
[INFO]    MMMMMMMMMMM+  sMMMMM:  hMm  /MMMMd  -MM- /s `h.`d- -MMMMMMMMMMMM
[INFO]    MMMMMMMMMMMs  +MMMMM.  mMm  /MMMMy  /MM. +M/   yM: `MMMMMMMMMMMM
[INFO]    MMMMMMMMMMMN-  smNm/  +MMm  :NNdo` .mMM` oMM+/yMM/  MMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMNo-    `:yMMMm      `:sNMMM` sMMMMMMM+  NMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMMMMNmmNMMMMMMMNmmmmNMMMMMMMNNMMMMMMMMMNNMMMMMMMMMMMM
[INFO]    ODM app finished - Sun Dec 18 17:48:21  2022
```

I'll take a closer look at the data and logs this afternoon.

smathermather, 2022-12-18:
Confirmed. Works fine with split merge, so it is likely a combination of geometric consistency and super big dataset. I'll test next with removing geometric consistency.

smathermather, 2022-12-18:
> As it is, it's really brittle for large datasets because anything going wrong in the submodels is going to cause failures globally.

This might be more true for distributed split-merge than for sequential split-merge. I'm going to play more with the sequential version to discern whether this is the case.

smathermather, 2022-12-18:
Now rerunning the full dataset in 2.9.2. Should be done in a day or two.

smathermather, 2022-12-20:
Confirmed it completes in 2.9.2 in addition to split merge:
```
Fused depth-maps 1978 (98.60%, 2m41s, ETA 2s)...
Fused depth-maps 1981 (98.75%, 2m41s, ETA 2s)...
Fused depth-maps 1984 (98.90%, 2m41s, ETA 1s)...
Fused depth-maps 1987 (99.05%, 2m41s, ETA 1s)...
Fused depth-maps 1989 (99.15%, 2m42s, ETA 1s)...
Fused depth-maps 1993 (99.35%, 2m42s, ETA 1s)...
Fused depth-maps 1995 (99.45%, 2m42s, ETA 894ms)...
Fused depth-maps 1999 (99.65%, 2m42s, ETA 568ms)...
Fused depth-maps 2002 (99.80%, 2m42s, ETA 324ms)...
Fused depth-maps 2004 (99.90%, 2m42s, ETA 162ms)...
Fused depth-maps 2006 (100%, 2m42s663ms)           
14:57:40 [App     ] Densifying point-cloud completed: 71446070 points (25m51s792ms)
````

This appears to be a regression for larger datasets. I've had other 2-5k image datasets all fail on latest.

smathermather, 2022-12-23:
Here is a small dataset that succeeds in 2.9.2 and fails now:
https://drive.google.com/drive/folders/1lDYH7klSqDos-ptXPO9d19XYVt83Ot8r?usp=share_link

26 images. Quite reasonable. :smiley:

smathermather, 2022-12-23:
Hmm. Ran another update and re-ran and that one succeeds now. I'll try re-running on the big stuff.

smathermather, 2022-12-23:
Possibly related: https://community.opendronemap.org/t/having-issues-lately-segmentation-fault/14145

pierotofy, 2023-02-13:
Closing since we now have `pc-skip-geometric`.

closed 2023-02-13 by pierotofy