OpenDroneMap/ODM #1584: Feature request, increase default resolution to 1cm

smathermather, 2023-01-17:
Per this conversation:
https://community.opendronemap.org/t/output-resolution-of-orthos-in-webodm-seems-to-be-fixed-and-not-a-result-of-image-resolution/14404

I think it makes some sense to increase the default resolution, given we now calculate GSD and don't go above it unless unwisely requested to do so. 1cm seems like a good compromise these days.

pierotofy, 2023-01-17:
The compromise is always quality vs speed (and in most cases, memory usage).

I'm not so sure bumping quality to the highest is the way to go (especially being a default).

For example, I've already seen some posts from confused users, that upon selecting "Ultra Quality" either run out of memory or never finish processing (e.g. https://community.opendronemap.org/t/processing-time-over-22-hrs-now-i-would-like-some-clarification/14223/12) 

Defaults should provide sensible values to start an investigation of a dataset. First and foremost, I think they should 1) Finish in a reasonable amount of time 2) Actually finish! Running out of memory is the worst outcome, finishing in days of processing a close second worst case scenario.

smathermather, 2023-01-17:
Is orthophoto the RAM bottleneck? I haven't profiled in a loooong time.

pierotofy, 2023-01-17:
It is not currently (but it might become it?). The thing is, `orthophoto-resolution` doesn't just impact the orthophoto rendering part, it affects the size of the undistorted images, which affects texturing, which is the second bottleneck overall.

smathermather, 2023-01-17:
> which is the second bottleneck overall.

What's number 1 currently? SfM?

pierotofy, 2023-01-17:
Depthmap fusion.

smathermather, 2023-01-17:
Cool. It's funny how and where the memory bottlenecks are depending on dataset size. I've rarely OOM'd on depthmap fusion on larger datasets.

Building a test VM, since often the OOM cases are swift and deadly, so better to put a hard cap on usage to discern whether turning the knobs exceeds the current bottleneck. I'm making it with the last 4-8GB of memory on my host, so hopefully the virtualization is well behaved... .

smathermather, 2023-01-17:
I'm going to close this issue and the related pull request. Having tested at 4GB RAM, our stated minimum, there is no space for parameter change. If we decide to increase this value to 8GB (which is probably a kinder minimum) then we can revisit this.

I made the mistake of not resizing before processing 75 images at 4GB and wow: that was painful.