OpenDroneMap/ODM #1586: Automatically extract .SRT subtitles from video files when possible

pierotofy, 2023-01-25:
Some video files (e.g. those made by newer DJI drones) have a textual track optionally saved along with video files. One can extract the corresponding .SRT file via:

```
fmpeg -i video.mp4 -o video.srt
```

It would be nicer if ODM could do this automatically *without* calling/adding ffmpeg as a dependency, we already compile ffmpeg within OpenCV, so it should be a matter of finding out which routines to call within the compiled ffmpeg library to extract the SRT data.

pierotofy, 2023-05-08:
No way to do this directly with OpenCV on Windows, as the binary shipped with Windows does not expose anything other than the OCV FFMPEG Capture interface code.

On Linux we already have avcodec and avformat so we could build something like this https://gist.github.com/pierotofy/12038d4b1674444b1707be880c8ce4c8 to extract the SRT, although it requires more code to be stable and imitate the ffmpeg binary.

On Windows we could build a separate binary, but it would still weight a few dozen MB.

Including the ffmpeg executable (a minimal build perhaps) might be the way to go.

vinsonliux, 2023-06-13:
hi,I couldn't export the subtitle file with the ffmpeg command you provided, I ended up using the following code
`ffmpeg -i DJI_0289.MOV -map 0:s:0 DJI_0289.srt`