OpenDroneMap/ODM #1594: Feature request: Return meshes from split merge

smathermather, 2023-02-02:
Per https://community.opendronemap.org/t/offer-a-single-textured-mesh-as-output-in-split-merge-processes/14608 , long term, returning a mesh from split merge would be beneficial. In the interim, returning the meshes that are produced may be an acceptable alternative.

Sibling issue here:
https://github.com/OpenDroneMap/NodeODM/issues/199

smathermather, 2023-02-15:
@pierotofy -- the links you reference in the NodeODM issue, would that be something similar to how multispectral datasets are handled with symlinks or something as simple as what we see in `stages/splitmerge.py` with submodel paths for e.g. `all_orthos_and_ortho_cuts`?

pierotofy, 2023-02-15:
* Scan the submodel folders for textured model files (similar to `all_orthos_and_ortho_cuts`)
 * Create the necessary folder structure via symlinks / hardlinks in the root `odm_texturing` folder. (e.g. `odm_texturing/submodel_0000/odm_textured_model_geo.obj`, `...`)

Symlinks are not available on Windows (you need to use hardlinks, which only work for files, not directories). Using links is the proper way to implement this imo. Naive copies are both slow and explode disk space.

Starting from https://github.com/OpenDroneMap/ODM/blob/bf824d35833783fde6afc6d9b5a95affec53a5c2/opendm/system.py#L149 , extend it to support files and folders, recursively. :warning: it's not hard, but with anything FS related it's not super trivial either!