OpenDroneMap/ODM #1595: How to apply custom keypoint detector in ODM?

Atypical-Programmer, 2023-02-04:
In recent years, there have been many new key point detectors, some of which may be better than SIFT. I want to add new methods to ODM, but the ODM source code is complex. Is there a simple way to add custom methods? Thank you.

Saijin-Naib, 2023-02-04:
Welcome!

Are these keypoint detectors in OpenCV, or are they separate libraries with dependencies we'll need to integrate and manage?

Atypical-Programmer, 2023-02-06:
These methods are not only in OpenCV, but also in various conferences (e.g CVPR,ICCV,ECCV) in recent years. I have used Colmap before, and I can directly and simply import custom keypoints and descriptors for matching and adjustment. But I don't know there is a similar way in ODM. Thank you.