OpenDroneMap/ODM #1597: Error during Build Docker Images From Source

peigon, 2023-02-08:
### How did you install ODM? (Docker, installer, natively, ...)?
I tried "Build Docker Images From Source"
https://github.com/OpenDroneMap/ODM

docker desktop Version 4.3.1 (72247)

### What is the problem?
An error occurred in opencv during build.

#12 664.0 -- Installing: /code/SuperBuild/install/share/opencv4/lbpcascades/lbpcascade_silverware.xml
#12 664.0 [ 55%] Completed 'opencv'
#12 664.1 [ 55%] Built target opencv
------
executor failed running [/bin/sh -c bash configure.sh install]: exit code: 2

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

The logs show an error when opencv is the build target after installing to SuperBuild.
Does the response to this commit work correctly on ubuntu?
commit:2a526f112021355e3dbfc5f4621b9af148569ab9

Thank you in advance for your cooperation.

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

The following is a description of the environment in which the test was conducted.
Hardware :
Apple : MacBook Pro (13-inch, M1, 2020) 16gb ram, 512gb SSD
Ubuntu : Running Virtual machine on Mac M1

peigon, 2023-02-09:
Thanks for the reply.
I will send you the log.
Best regards.

[fulllog.txt](https://github.com/OpenDroneMap/ODM/files/10691250/fulllog.txt)

pierotofy, 2023-02-09:
There it is:

```
#10 3837.5 g++: fatal error: Killed signal terminated program cc1plus
```

You ran out of memory. So you might need more RAM (or swap) to build the image. Not a software fault. 🙏

closed 2023-02-09 by pierotofy

peigon, 2023-02-10:
I gave the docker memory.
I was able to run it with the following code.
`docker build --squash -t my_odm_image .`

I see that I need more than 20GB memory to Build.

Best regards.