OpenDroneMap/ODM #1599: The 3D Model Generated by ODM With --3d-tiles Always Navigate to Italy in Cesium

PpYyRrAa, 2023-02-13:
Appears to work fine with the 3D Tiles of the PointCloud, but the 3D model ends up in Milano, Italy:
image
image
1920×1080 115 KB

ppyyrraa-01
Created on:	2/10/2023, 11:06:48 PM
Processing Node:	node-odm-1 (manual)
Options:	3d-tiles: true, auto-boundary: true, dsm: true
Average GSD:	2.43 cm
Area:	79,360.96 m²
Reconstructed Points:	1,917,311
WebODM current/latest, ODM v3.0.4, previewed in Cesium ION Online

image
image
1920×1080 127 KB
ppyyrraa-02

Created on:	2/10/2023, 11:07:11 PM
Processing Node:	node-odm-1 (manual)
Options:	3d-tiles: true, auto-boundary: true, dsm: true
Average GSD:	3.54 cm
Area:	80,098.1 m²
Reconstructed Points:	1,527,007
WebODM current/latest, ODM v3.0.4, previewed in Cesium ION Online

Native Dataset：
https://1drv.ms/u/s!AhZ8njnn54ZkkC2YC4TL4F4V5y1w?e=e9Zu6j

pierotofy, 2023-02-13:
Could you copy/paste the full task output log? 🙏 

The default location for the obj2tiles program is Milan, Italy, so for some reason it looks like the program is not being invoked with the proper geographic center.

PpYyRrAa, 2023-02-14:
[log.zip](https://github.com/OpenDroneMap/ODM/files/10728449/log.zip)
This the log.json file output by ODM.

PpYyRrAa, 2023-02-14:
> Could you copy/paste the full task output log? 🙏
> 
> The default location for the obj2tiles program is Milan, Italy, so for some reason it looks like the program is not being invoked with the proper geographic center.

This the log.json file output by ODM.
[log.zip](https://github.com/OpenDroneMap/ODM/files/10728484/log.zip)

pierotofy, 2023-02-14:
Thanks for the log; this should be fixed with #1603 :pray:

PR #1603 (2023-02-14): Update obj2tiles, fix lat/lon/alt
Fixes #1599