OpenDroneMap/ODM #1620: Possible Bug: Rolling Shutter Correction unable to handle non-Geolocated Imagery

Saijin-Naib, 2023-03-16:
### How did you install ODM? (Docker, installer, natively, ...)?
Docker

### What is the problem?
Consistent failure to process a sample dataset with RSC enabled, though it processes fine with same settings otherwise (and RSC disabled). Sensor readout is accurate and present in our database.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.
If RSC can't work with non-geolocated imagery, a graceful fallback or fixed up error message that is more meaningful would be helpful.

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset AND task output log, uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).
Console Log:  
```
[INFO]    Initializing ODM 3.0.5 - Thu Mar 16 05:06:28  2023
[INFO]    ==============
[INFO]    3d_tiles: False
[INFO]    align: None
[INFO]    auto_boundary: True
[INFO]    auto_boundary_distance: 0
[INFO]    bg_removal: True
[INFO]    boundary: {}
[INFO]    build_overviews: False
[INFO]    camera_lens: auto
[INFO]    cameras: {}
[INFO]    cog: True
[INFO]    copy_to: None
[INFO]    crop: 3
[INFO]    dem_decimation: 1
[INFO]    dem_euclidean_map: False
[INFO]    dem_gapfill_steps: 3
[INFO]    dem_resolution: 5
[INFO]    dsm: False
[INFO]    dtm: False
[INFO]    end_with: odm_postprocess
[INFO]    fast_orthophoto: False
[INFO]    feature_quality: ultra
[INFO]    feature_type: hahog
[INFO]    force_gps: False
[INFO]    gcp: None
[INFO]    geo: None
[INFO]    gltf: True
[INFO]    gps_accuracy: 10
[INFO]    ignore_gsd: False
[INFO]    matcher_neighbors: 64
[INFO]    matcher_type: bow
[INFO]    max_concurrency: 8
[INFO]    merge: all
[INFO]    mesh_octree_depth: 11
[INFO]    mesh_size: 200000
[INFO]    min_num_features: 128000
[INFO]    name: d6e61ec5-946e-42b5-92c0-39f6944f3c32
[INFO]    no_gpu: False
[INFO]    optimize_disk_space: False
[INFO]    orthophoto_compression: DEFLATE
[INFO]    orthophoto_cutline: False
[INFO]    orthophoto_kmz: False
[INFO]    orthophoto_no_tiled: False
[INFO]    orthophoto_png: False
[INFO]    orthophoto_resolution: 5
[INFO]    pc_classify: False
[INFO]    pc_copc: False
[INFO]    pc_csv: False
[INFO]    pc_ept: True
[INFO]    pc_filter: 0.0
[INFO]    pc_las: False
[INFO]    pc_quality: ultra
[INFO]    pc_rectify: False
[INFO]    pc_sample: 0
[INFO]    pc_skip_geometric: False
[INFO]    pc_tile: False
[INFO]    primary_band: auto
[... 176 lines cut ...]
File "/code/opendm/types.py", line 397, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 397, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 397, in run
self.next_stage.run(outputs)
File "/code/opendm/types.py", line 376, in run
self.process(self.args, outputs)
File "/code/stages/run_opensfm.py", line 38, in process
octx.reconstruct(args.rolling_shutter, self.rerun())
File "/code/opendm/osfm.py", line 73, in reconstruct
self.run('rs_correct')
File "/code/opendm/osfm.py", line 34, in run
system.run('"%s" %s "%s"' %
File "/code/opendm/system.py", line 110, in run
raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 1

===== Done, human-readable information to follow... =====

[ERROR]   Uh oh! Processing stopped because of strange values in the reconstruction. This is often a sign that the input data has some issues or the software cannot deal with it. Have you followed best practices for data acquisition? See https://docs.opendronemap.org/flying/
```

Imagery:  
Upon request (Contains PII)

pierotofy, 2023-03-16:
Yep, we shouldn't allow RSC if there's no geographic reference (it's not doable).

Saijin-Naib, 2023-03-16:
Which is a shame because the velocities it resolved actually make a ton of sense and seem correct.

Is there really no way to have the cake and eat it too here?

closed 2023-03-21 by pierotofy