OpenDroneMap/ODM #1630: you cant type anything in the console on win11

goofyseeker311, 2023-04-07:
instructions say to type in the console after you run it, but you cant type anything in the console, it just shows the help text, nothing happens

pierotofy, 2023-04-08:
```
Please use the format below to report bugs and faults. 
****************************************

### How did you install ODM? (Docker, installer, natively, ...)?

[Type answer here]

### What is the problem?

[Type answer here]

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

[Type answer here]

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset AND task output log, uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

[Type answer here]

```

closed 2023-04-08 by pierotofy

goofyseeker311, 2023-04-08:
very disrespectful, will not endorse your project in any way