OpenDroneMap/ODM #1635: GLTF models can't be ingested by Sketchfab. I've tried two very different models so I'm concluding this is an ODM issue not something particular to one model.

TimeScience, 2023-04-20:
GLTF models can't be ingested by Sketchfab. I've tried two very different models so I'm concluding this is an ODM issue not something particular to one model.
They also all fail when opening in Meshlab with the error: _"accessor[0] invalid bufferView"_
@pierotofy, can you have a look?
See for example this model: https://asdc.cloud.edu.au/public/task/6b947e2e-ea34-4fb9-b8ac-3e837cff3758/3d/

_Originally posted by @TimeScience in https://github.com/OpenDroneMap/ODM/issues/1587#issuecomment-1515922484_

pierotofy, 2023-04-20:
The model is fine.

https://github.khronos.org/glTF-Validator/

[image]

It's MeshLab that lacks support (probably for the Draco compression extension), so if anything a ticket should be opened in MeshLab.

[image]

https://gltf-viewer.donmccurdy.com/

One can use OBJs if they need to open them in MeshLab.

closed 2023-04-20 by pierotofy

TimeScience, 2023-04-24:
Could we maybe add a feature to turn draco compression off and on (assuming that is the issue)? If two of the most commonly used 3D platforms don't support the models ODM creates then that immediately adds a major roadblock for ODM users to work with the files, and the goal of adding GLTF exports was to make it easier to get ODM-generated objects into other platforms like meshlab and sketchfab.
For example, the obj export is 300mb which is too big to upload to sketchfab even for my paid account (>200mb requires a pro accounts at $70/mo), hence the utility of having ODM export gltf format.
Likewise, meshlab is one of the most commonly used open source modelling apps and there still aren't that many Win apps that support gltf software so if people can't use meshlab we are limiting what they can do.

Also, there's still something not totally right with the glb files. For example, if I import them into spatial.io, all gltf models from ODM come in turned on their side and with a weird rotational axis so they can't be rotated to be in the correct orientation. See [video here](https://user-images.githubusercontent.com/3067588/233913005-dbbecc38-8a3a-4a24-9d35-c2c244cb7f93.MP4) comparing an ODM model vs a model made by polycam. This doesn't seem to be related to the model having a huge bounding box due to outlyer data since the obj model (which I can view in meshlab), loads fine and has a normal sized bounding box. And since all ODM-produced glb models have this issue, it seems likely to be an ODM issue not a spatial issue. 
Since I can't view the glb model in either meshlab or sketchfab, I can't provide further information about it.

pierotofy, 2023-04-24:
Let's discuss via e-mail.