OpenDroneMap/ODM #1647: Easy way to modify the code and test it

joelitreview, 2023-05-06:
### How did you install ODM? (Docker, installer, natively, ...)?
- Cloned https://github.com/OpenDroneMap/ODM
- Ran with the following command:
```bash
docker run -ti --rm -v /home/youruser/datasets:/datasets --gpus all opendronemap/odm:gpu --project-path /datasets project
```
- Or using the dev environment:
```bash
DATA=/path/to/datasets ./start-dev-env.sh
```
### What is the problem?
I cannot test my opensfm code changes. I tried forking from the opensfm repo and pushed changes to the new repo and referencing that new repo from the cmake file SuperBuild/cmake/External-OpenSfM.cmake, but it doesn't take.
I was wondering what is the easiest way to modify opensfm code, preferably locally, (Specifically the opensfm/src/map/src/tracks_manager.cc) and get the changes into the build. I tried it also in the dev env.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

Changes made to the code are actually compiled.

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset AND task output log, uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

- fork opensfm repo and push code changes 
- reference the repo form the cmake file
- run code

pierotofy, 2023-05-06:
Definitely use `DATA=/path/to/datasets ./start-dev-env.sh` and not docker.

From the dev environment I typically navigate to `/code/SuperBuild/install/bin/opensfm` and create a new branch (`git checkout -b newbranch`), make changes and then run ODM as normal.

If I need to modify the C++ modules, after editing the files in OpenSfM I navigate to `/code/SuperBuild/build/opensfm` and issue `make -j$(nproc)`.

This feels like a question, not really a bug; perhaps this conversation should be moved to the forum.

closed 2023-05-06 by pierotofy

smathermather, 2023-05-07:
@joelitreview -- if you see a place to clarify https://github.com/OpenDroneMap/ODM/blob/master/README.md, please open a pull request. Thanks!