OpenDroneMap/ODM #1652: error when docker build -f gpu.Dockerfile: No module named 'cv2'

rexliuser, 2023-05-17:
### How did you install ODM? (Docker, installer, natively, ...)?

Trying to build gpu.Dockerfile.
Host system: ubuntu 18.04.6 LTS bionic, docker 23.0.1, RTX3090

### What is the problem?

Error during docker build. 
Command: docker build -f gpu.Dockerfile -t odm_new_gpu . --no-cache

Error message: 
-----------------------------------
#0 200.2 Setting up python3-pip (20.0.2-5ubuntu1.8) ...
#0 201.2 Requirement already satisfied: pip in /usr/local/lib/python3.8/dist-packages (23.1.2)
#0 201.6 WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
#0 202.3 Requirement already satisfied: shyaml in /usr/local/lib/python3.8/dist-packages (0.6.2)
#0 202.6 Requirement already satisfied: pyyaml in /usr/local/lib/python3.8/dist-packages (from shyaml) (5.1)
#0 202.9 WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
#0 203.0 Ubuntu: Release:       20.04, good!
#0 203.0 Installing Required Requisites
#0 312.5 E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/g/gcc-10/libgfortran5_10.3.0-1ubuntu1~20.04_amd64.deb  Hash Sum mismatch
#0 312.5    Hashes of expected file:
#0 312.5     - SHA512:b6484bdcade5bd21da0d6e4b4975e40e0782815b7439dfdbb717bd36fc8659e0b4567ef6ccfeb1df9abd67332908bfcadf8545141741e9e66b7aa5a7ed0bc2b5
#0 312.5     - SHA256:1404ae20993f3cd24ec71e29445e5c5d44eadfd8456e0d55bf1edc72c9271d4b
#0 312.5     - SHA1:3a69e4f01fe2aa96159af1038aea25756375ab45 [weak]
#0 312.5     - MD5Sum:e84641d06b5196cc3b542b2c91a4c462 [weak]
#0 312.5     - Filesize:736412 [weak]
#0 312.5    Hashes of received file:
#0 312.5     - SHA512:0ff99d58890f38a0951f75109e3e1521fdd0ae3d8f9a3ac63bda2ec9cf29012fc9487fefecb7db6dc48b3bf245f39d2e505e62c4b9936da95ae08475c6ef13f4
#0 312.5     - SHA256:d43c289f98c1ba22e726a667b15550d1049261bb55581ce5c219f99b49fdfa89
#0 312.5     - SHA1:e1d8b1a9b453b4e0eaf774615f2f766e092e6440 [weak]
#0 312.5     - MD5Sum:7aaa092df19c750e96e7bffba5902d03 [weak]
#0 312.5     - Filesize:736412 [weak]
#0 312.5    Last modification reported: Mon, 14 Jun 2021 19:28:55 +0000
#0 312.5 E: Failed to fetch http://ppa.launchpad.net/ubuntugis/ubuntugis-unstable/ubuntu/pool/main/p/proj/proj-data_8.2.0-1~focal0_all.deb  Connection timed out [IP: 185.125.190.52 80]
#0 312.5 E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
#0 312.5 Installing OpenCV Dependencies
#0 418.7 E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/universe/libv/libva/libva2_2.7.0-2_amd64.deb  Hash Sum mismatch
#0 418.7    Hashes of expected file:
#0 418.7     - SHA256:3fb6c7ca117cc4889215769001e512cb839971498dae06ef1ac3769815d304d3
#0 418.7     - SHA1:e96e732e1e1e240921c00ba0e39b62097364681d [weak]
#0 418.7     - MD5Sum:a604eb8550c8a927823b8b558ffd5d3c [weak]
#0 418.7     - Filesize:53532 [weak]
#0 418.7    Hashes of received file:
#0 418.7     - SHA256:f488501dc914590fb99d09ddbd0ff57ef457055aee31493350ce0470324dbc13
#0 418.7     - SHA1:165b061bf1bcf9eb9930cd265444ea9fae08e5a8 [weak]
#0 418.7     - MD5Sum:0cf47e377595ac51a1f5cc30b1a41508 [weak]
#0 418.7     - Filesize:53532 [weak]
#0 418.7    Last modification reported: Thu, 16 Apr 2020 16:03:42 +0000
#0 418.7 E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/universe/libv/libva/libva-x11-2_2.7.0-2_amd64.deb  Hash Sum mismatch
#0 418.7    Hashes of expected file:
#0 418.7     - SHA256:e69a69c87d1608fb4df750788d57a0eb46e62dc5763602a6f3f7fb5680c1cba1
#0 418.7     - SHA1:55712efbdb57db31f434ee99f412870e570a6491 [weak]
#0 418.7     - MD5Sum:9d28e8a093d652b9120f04205a542db0 [weak]
#0 418.7     - Filesize:11896 [weak]
#0 418.7    Hashes of received file:
#0 418.7     - SHA256:21605b89f70bf5ffd006a47f82c01c096dd763bab04aea14b624b52238f65133
#0 418.7     - SHA1:3ff1c4939b2bcf3988541c6edecb73febb4aa1b0 [weak]
#0 418.7     - MD5Sum:24e8033492d20f590e0cb045cbf2f672 [weak]
#0 418.7     - Filesize:11896 [weak]
#0 418.7    Last modification reported: Thu, 16 Apr 2020 16:03:43 +0000
#0 418.7 E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/libv/libvdpau/libvdpau1_1.3-1ubuntu2_amd64.deb  Bad header line Bad header data [IP: 185.125.190.39 80]
#0 418.7 E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/universe/c/codec2/libcodec2-0.9_0.9.2-2_amd64.deb  Hash Sum mismatch
#0 418.7    Hashes of expected file:
#0 418.7     - SHA256:afb548df7b5ac98ac74bbef26a7930257076f95a29a09081d539f57e90f5a632
#0 418.7     - SHA1:a30eaabeb2128dcb77775db6c8e6e2dbfc8dabe4 [weak]
#0 418.7     - MD5Sum:076a5df32a06dd23e5c4bb9d99b11c25 [weak]
#0 418.7     - Filesize:7885840 [weak]
#0 418.7    Hashes of received file:
#0 418.7     - SHA256:d7f1b9069a166fa70e3fc7e98b330b72f95881a56c835133cda1fcb63f82e268
#0 418.7     - SHA1:14dc3b93738ec14a8623208b7378b2ea50e5c81b [weak]
#0 418.7     - MD5Sum:607330c70298f3f9e476476601738d31 [weak]
#0 418.7     - Filesize:7885840 [weak]
#0 418.7    Last modification reported: Mon, 23 Dec 2019 05:33:16 +0000
#0 418.7 E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
#0 418.7 Installing OpenSfM Dependencies
...
#0 439.0 Unpacking libboost-system1.71.0:amd64 (1.71.0-6ubuntu6) ...
#0 439.3 Setting up libboost-iostreams1.71.0:amd64 (1.71.0-6ubuntu6) ...
#0 439.5 Setting up libboost-serialization1.71.0:amd64 (1.71.0-6ubuntu6) ...
#0 439.6 Setting up libboost-system1.71.0:amd64 (1.71.0-6ubuntu6) ...
#0 439.8 Setting up libboost-program-options1.71.0:amd64 (1.71.0-6ubuntu6) ...
#0 439.9 Processing triggers for libc-bin (2.31-0ubuntu9.9) ...
#0 440.5 Traceback (most recent call last):
#0 440.5   File "/code/run.py", line 15, in <module>
#0 440.5     from opendm.utils import get_processing_results_paths, rm_r
#0 440.5   File "/code/opendm/utils.py", line 5, in <module>
#0 440.5     from opendm.photo import find_largest_photo_dims
#0 440.5   File "/code/opendm/photo.py", line 17, in <module>
#0 440.5     from opendm import get_image_size
#0 440.5   File "/code/opendm/get_image_size.py", line 2, in <module>
#0 440.5     import cv2
#0 440.5 ModuleNotFoundError: No module named 'cv2'
------
gpu.Dockerfile:44
--------------------
  43 |     # Also run a smoke test on ODM and OpenSfM
  44 | >>> RUN bash configure.sh installruntimedepsonly \
  45 | >>>   && apt-get clean \
  46 | >>>   && rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/* \
  47 | >>>   && bash run.sh --help \
  48 | >>>   && bash -c "eval $(python3 /code/opendm/context.py) && python3 -c 'from opensfm import io, pymap'"
  49 |     # Entry point
--------------------
ERROR: failed to solve: process "/bin/sh -c bash configure.sh installruntimedepsonly   && apt-get clean   && rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/*   && bash run.sh --help   && bash -c \"eval $(python3 /code/opendm/context.py) && python3 -c 'from opensfm import io, pymap'\"" did not complete successfully: exit code: 1
-----------------------------------
### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

I want to build ODM gpu docker with custom modification.

### How can we reproduce this? What steps did you do to trigger the problem? If this is an issue with processing a dataset, YOU MUST include a copy of your dataset AND task output log, uploaded on Google Drive or Dropbox (otherwise we cannot reproduce this).

Command: docker build -f gpu.Dockerfile -t odm_new_gpu . --no-cache

Saijin-Naib, 2023-05-17:
Can you please try on 20.04LTS or newer?

rexliuser, 2023-05-18:
Success on 20.04. Been stuck a long time. Thanks

closed 2023-05-18 by rexliuser