OpenDroneMap/ODM #1668: AnsiEsc codes in logs

sbonaime, 2023-06-21:
Hi,
As I am using ODM with SLURM on a clsuter, my logs go to text file and I read them with vim. But AnsiEsc are quite annoying because there are not supported by default. 
I would be interested by having a --no-ansiesc flag. 

And I think I can do it ! Any chance that this futur PR will be merge ? 
Thanks

pierotofy, 2023-06-21:
Mm, I think log cleaning should be done by client apps (perhaps via a filter). A simple regex would do the trick. See https://github.com/OpenDroneMap/NodeODM/blob/26d10d990ccd9b04f3a591fd3cb2ac515e928d45/libs/Task.js#L601

closed 2023-06-21 by pierotofy

pierotofy, 2023-06-21:
I should mention, it could be OK to have an environment variable toggle ansiesc codes. (Just not a command line flag).

See also https://github.com/powerman/vim-plugin-AnsiEsc