OpenDroneMap/ODM #1675: docker build with gpu: cuda image no longer available?

rexliuser, 2023-07-14:
Trying to build docker with gpu. However cuda image "docker.io/nvidia/cuda:11.2.0-devel-ubuntu20.04" seems no longer available:

os: ubuntu-22.04
command: docker build -f gpu.Dockerfile -t test . --no-cache

output error:
-------------------------------------------------------------------------------------------------------------------
[+] Building 4.4s (4/4) FINISHED                                                                   docker:default
 => [internal] load build definition from gpu.Dockerfile                                                     0.4s
 => => transferring dockerfile: 1.78kB                                                                       0.1s
 => [internal] load .dockerignore                                                                            0.6s
 => => transferring context: 372B                                                                            0.0s
 => CANCELED [internal] load metadata for docker.io/nvidia/cuda:11.2.0-runtime-ubuntu20.04                   2.0s
 => ERROR [internal] load metadata for docker.io/nvidia/cuda:11.2.0-devel-ubuntu20.04                        2.0s
------
 > [internal] load metadata for docker.io/nvidia/cuda:11.2.0-devel-ubuntu20.04:
------
gpu.Dockerfile:1
--------------------
   1 | >>> FROM nvidia/cuda:11.2.0-devel-ubuntu20.04 AS builder
   2 |
   3 |     # Env variables
--------------------
**ERROR: failed to solve: nvidia/cuda:11.2.0-devel-ubuntu20.04: docker.io/nvidia/cuda:11.2.0-devel-ubuntu20.04: not found**

pierotofy, 2023-07-14:
Mm, yes it looks like they pulled the `11.2.0` image. 

We should update the docker image to use [11.2.2-devel-ubuntu20.04](https://hub.docker.com/layers/nvidia/cuda/11.2.2-devel-ubuntu20.04/images/sha256-2d3ebf94cfe21fe434f35d1033d63c09bd0a79e70f18d7a926264f3f877a19e4?context=explore) instead.

Interested in opening a PR @rexliuser ? 🙏

rexliuser, 2023-07-15:
opened PR. 🙏

closed 2023-07-15 by rexliuser