OpenDroneMap/ODM #1679: Can not  set up the required third party libraries after the docker build

sbonaime, 2023-07-19:
Can not  set up the required third party libraries after the docker build on Debian  BullEyes 5.10.179-2.
I install Miniconda for python and also try installing libpng-dev  libjpeg-dev libz3-dev
Here is the (long) error message avec launching ``` bash configure.sh reinstall```

````
-- Installing: /code/SuperBuild/install/share/opencv4/lbpcascades/lbpcascade_silverware.xml
[ 59%] Completed 'opencv'
[ 59%] Built target opencv
In file included from Src/Ply.h:56,
                 from Src/VertexStream.h:33,
                 from Src/FEMTree.h:51,
                 from Src/PoissonRecon.cpp:47:
Src/CoredMesh.h: In constructor ‘CoredFileMeshData<Index, VertexFactory>::CoredFileMeshData(const VertexFactory&, const char*) [with Index = int; VertexFactory = VertexFactory::Factory<float, VertexFactory::PositionFactory<float, 3>, VertexFactory::NormalFactory<float, 3>, VertexFactory::ValueFactory<float>, VertexFactory::EmptyFactory<float> >]’:
Src/CoredMesh.h:156:67: warning: ‘XXXXXX’ directive writing 6 bytes into a region of size between 1 and 1024 [-Wformat-overflow=]
  156 |    if( fileHeader && strlen(fileHeader) ) sprintf( _fileName , "%sXXXXXX" , fileHeader );
      |                                                                   ^~~~~~
In file included from /usr/include/stdio.h:866,
                 from Src/PoissonRecon.cpp:40:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:38:34: note: ‘__builtin___sprintf_chk’ output between 7 and 1030 bytes into a destination of size 1024
   38 |   return __builtin___sprintf_chk (__s, __USE_FORTIFY_LEVEL - 1,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   39 |       __glibc_objsize (__s), __fmt,
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   40 |       __va_arg_pack ());
      |       ~~~~~~~~~~~~~~~~~
In file included from Src/Ply.h:56,
                 from Src/VertexStream.h:33,
                 from Src/FEMTree.h:51,
                 from Src/PoissonRecon.cpp:47:
Src/CoredMesh.h: In constructor ‘CoredFileMeshData<Index, VertexFactory>::CoredFileMeshData(const VertexFactory&, const char*) [with Index = int; VertexFactory = VertexFactory::Factory<float, VertexFactory::PositionFactory<float, 3>, VertexFactory::DynamicFactory<float> >]’:
Src/CoredMesh.h:156:67: warning: ‘XXXXXX’ directive writing 6 bytes into a region of size between 1 and 1024 [-Wformat-overflow=]
  156 |    if( fileHeader && strlen(fileHeader) ) sprintf( _fileName , "%sXXXXXX" , fileHeader );
      |                                                                   ^~~~~~
In file included from /usr/include/stdio.h:866,
                 from Src/PoissonRecon.cpp:40:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:38:34: note: ‘__builtin___sprintf_chk’ output between 7 and 1030 bytes into a destination of size 1024
   38 |   return __builtin___sprintf_chk (__s, __USE_FORTIFY_LEVEL - 1,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   39 |       __glibc_objsize (__s), __fmt,
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   40 |       __va_arg_pack ());
      |       ~~~~~~~~~~~~~~~~~
In file included from Src/Ply.h:56,
                 from Src/VertexStream.h:33,
                 from Src/FEMTree.h:51,
                 from Src/PoissonRecon.cpp:47:
Src/CoredMesh.h: In constructor ‘CoredFileMeshData<Index, VertexFactory>::CoredFileMeshData(const VertexFactory&, const char*) [with Index = int; VertexFactory = VertexFactory::Factory<float, VertexFactory::PositionFactory<float, 3>, VertexFactory::NormalFactory<float, 3>, VertexFactory::DynamicFactory<float> >]’:
Src/CoredMesh.h:156:67: warning: ‘XXXXXX’ directive writing 6 bytes into a region of size between 1 and 1024 [-Wformat-overflow=]
  156 |    if( fileHeader && strlen(fileHeader) ) sprintf( _fileName , "%sXXXXXX" , fileHeader );
      |                                                                   ^~~~~~
In file included from /usr/include/stdio.h:866,
                 from Src/PoissonRecon.cpp:40:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:38:34: note: ‘__builtin___sprintf_chk’ output between 7 and 1030 bytes into a destination of size 1024
   38 |   return __builtin___sprintf_chk (__s, __USE_FORTIFY_LEVEL - 1,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   39 |       __glibc_objsize (__s), __fmt,
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   40 |       __va_arg_pack ());
      |       ~~~~~~~~~~~~~~~~~
In file included from Src/Ply.h:56,
                 from Src/VertexStream.h:33,
                 from Src/FEMTree.h:51,
                 from Src/PoissonRecon.cpp:47:
Src/CoredMesh.h: In constructor ‘CoredFileMeshData<Index, VertexFactory>::CoredFileMeshData(const VertexFactory&, const char*) [with Index = int; VertexFactory = VertexFactory::Factory<float, VertexFactory::PositionFactory<float, 3>, VertexFactory::ValueFactory<float>, VertexFactory::DynamicFactory<float> >]’:
Src/CoredMesh.h:156:67: warning: ‘XXXXXX’ directive writing 6 bytes into a region of size between 1 and 1024 [-Wformat-overflow=]
  156 |    if( fileHeader && strlen(fileHeader) ) sprintf( _fileName , "%sXXXXXX" , fileHeader );
      |                                                                   ^~~~~~
In file included from /usr/include/stdio.h:866,
                 from Src/PoissonRecon.cpp:40:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:38:34: note: ‘__builtin___sprintf_chk’ output between 7 and 1030 bytes into a destination of size 1024
[... 297 lines cut ...]
/usr/include/x86_64-linux-gnu/bits/stdio2.h:38:34: note: ‘__builtin___sprintf_chk’ output between 5 and 1028 bytes into a destination of size 1024
   38 |   return __builtin___sprintf_chk (__s, __USE_FORTIFY_LEVEL - 1,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   39 |       __glibc_objsize (__s), __fmt,
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   40 |       __va_arg_pack ());
      |       ~~~~~~~~~~~~~~~~~
Src/PoissonRecon.cpp:375:82: warning: ‘PR_’ directive writing 3 bytes into a region of size between 1 and 1024 [-Wformat-overflow=]
  375 |   if( tempPath[ strlen( tempPath )-1 ]==FileSeparator ) sprintf( tempHeader , "%sPR_" , tempPath );
      |                                                                                  ^~~
In file included from /usr/include/stdio.h:866,
                 from Src/PoissonRecon.cpp:40:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:38:34: note: ‘__builtin___sprintf_chk’ output between 4 and 1027 bytes into a destination of size 1024
   38 |   return __builtin___sprintf_chk (__s, __USE_FORTIFY_LEVEL - 1,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   39 |       __glibc_objsize (__s), __fmt,
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   40 |       __va_arg_pack ());
      |       ~~~~~~~~~~~~~~~~~
[ 60%] Performing install step for 'poissonrecon'
[ 60%] Completed 'poissonrecon'
[ 60%] Built target poissonrecon
make: *** [Makefile:103: all] Error 2
(odmdev)[bonaime:/code] 12m13s 2 $ 
```

pierotofy, 2023-07-19:
You need to check the full log for errors (which sometimes are not at the end). Make sure you're not running out of memory!

sbonaime, 2023-07-19:
Ok the first Error I have is 
```CMake Error at cmake_install.cmake:125 (file):
  file INSTALL cannot copy file
  "/code/SuperBuild/build/gflags/gflags-registry-entry" to
  "/home/bonaime/.cmake/packages/gflags/c872b97259e15f7dbf4904fb8eb2bb74":
  Permission denied.
````

But I dont have a /home/bonaime/.cmake directory !

sbonaime, 2023-07-20:
I try to create a new user on my Debian with the same result. 

Next idea was to change the gflags version in the make file from 2.1.2 to 2.2.2. 
And now I don't have any compilation issue !  :-) 

@pierotofy  Is there some reason to use gflags 2.1.2 from Mar 24, 2015 instead of 2.2.2 from Nov 11, 2018  ?

Thanks

closed 2023-07-21 by pierotofy