OpenDroneMap/ODM #1685: Can’t build GPU docker for Nvidia Jetson devices

Dennis-Johnson, 2023-08-09:
### How did you install ODM? (Docker, installer, natively, ...)?

Docker: tried using the original gpu.Dockerfile and a modified dockerfile (included below) as well. 

### What is the problem?
I can't build ODM on a Nvidia Jetson (Tegra SoC) device using the default gpu.Dockerfile.

### What should be the expected behavior? If this is a feature request, please describe in detail the changes you think should be made to the code, citing files and lines where changes should be made, if possible.

I want to test ODM on Nvidia Jetson devices. These devices have an integrated GPU on the SoC unlike the PCIe connected GPUs on x86 desktops. This implies that ```nvidia-smi``` does not work to detect presence of a GPU. Typically ```tegrastats``` util is used to find GPU and CUDA info. 

My use case is to accelerate SIFT feature detection on the GPU. Is the OpenCV dependency needed for this in the first place? Can someone suggest parts of the code-base to modify to test whether the SIFT bits have been built with CUDA support properly?

### How can we reproduce this? What steps did you do to trigger the problem? 

On trying to build the original gpu.Dockerfile, this error is shown.
```
# Original dockerfile with GPU options.
docker build -t test_odm -f gpu.Dockerifle . 

...
Setting up libboost-program-options1.71.0:arm64 (1.71.0-6ubuntu6) ...
Processing triggers for libc-bin (2.31-0ubuntu9.9) ...
Traceback (most recent call last):
  File "/code/run.py", line 15, in <module>
    from opendm.utils import get_processing_results_paths, rm_r
  File "/code/opendm/utils.py", line 5, in <module>
    from opendm.photo import find_largest_photo_dims
  File "/code/opendm/photo.py", line 17, in <module>
    from opendm import get_image_size
  File "/code/opendm/get_image_size.py", line 2, in <module>
    import cv2
  File "/code/SuperBuild/install/lib/python3.8/dist-packages/cv2/__init__.py", line 96, in <module>
    bootstrap()
  File "/code/SuperBuild/install/lib/python3.8/dist-packages/cv2/__init__.py", line 86, in bootstrap
    import cv2
ImportError: libavcodec.so.58: cannot open shared object file: No such file or directory
The command '/bin/sh -c bash configure.sh installruntimedepsonly   && apt-get clean   && rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/*   && bash run.sh --help   && bash -c "eval $(python3 /code/opendm/context.py) && python3 -c 'from opensfm import io, pymap'"' returned a non-zero code: 1
```

I tried apt install  ```libavutil-dev``` before this step in the dockerfile, but it's not found. 

Since the CUDA version is coupled with the OS for the Tegra devices, I installed what I assumed is a similar base image which has the required CUDA runtime and libraries. 
```
# Modified gpu.Dockerfile
FROM nvcr.io/nvidia/l4t-cuda:11.4.19-runtime

# Env variables
ENV DEBIAN_FRONTEND=noninteractive \
    PYTHONPATH="$PYTHONPATH:/code/SuperBuild/install/lib/python3.9/dist-packages:/code/SuperBuild/install/lib/python3.8/dist-packages:/code/SuperBuild/install/bin/opensfm" \
    LD_LIBRARY_PATH="$LD_LIBRARY_PATH:/code/SuperBuild/install/lib"

# Prepare directories
WORKDIR /code

# Copy everything
COPY . ./

RUN PORTABLE_INSTALL=YES GPU_INSTALL=YES bash configure.sh install

# Install shared libraries that we depend on via APT, but *not*
# the -dev packages to save space!
# Also run a smoke test on ODM and OpenSfM
RUN bash configure.sh installruntimedepsonly \
  && apt-get clean \
  && rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/*
RUN bash run.sh --help
RUN bash -c "eval $(python3 /code/opendm/context.py) && python3 -c 'from opensfm import io, pymap'"

# Entry point
ENTRYPOINT ["python3", "/code/run.py"]
```

Still no luck, I got a different error during the OpenCV build step using this modified Dockerfile.
```
....
Error: 
-- Installing: /code/SuperBuild/install/share/opencv4/lbpcascades/lbpcascade_frontalface_improved.xml
-- Installing: /code/SuperBuild/install/share/opencv4/lbpcascades/lbpcascade_profileface.xml
-- Installing: /code/SuperBuild/install/share/opencv4/lbpcascades/lbpcascade_silverware.xml
[ 68%] Completed 'opencv'
[ 68%] Built target opencv
make: *** [Makefile:84: all] Error 2
The command '/bin/sh -c PORTABLE_INSTALL=YES GPU_INSTALL=YES bash configure.sh install' returned a non-zero code: 2
```

### System Info
```
$ cat /etc/lsb-release

DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.5 LTS"

# Info from jtop utility
Machine: aarch64 Jetson Xavier AGX
Release: 5.10.104-tegra
System Python version: 3.8.10

Linux4Tegra : v35.1.0
Jetpack: 5.0.2 
```

pierotofy, 2023-08-25:
Strange that the build fails, in any case, to test stuff you shouldn't need to rebuild the image, use `start-dev-env.sh` and pass the `IMAGE=opendronemap/odm:gpu GPU=YES` parameters to the script.

grunlab, 2025-03-08:
Hi,

I'm also very interested to get/test ODM GPU image on ARM64 (turing pi RK1)
I've tried to build the image myself (https://gitlab.com/grunlab/opendronemap/-/blob/main/build/gitlab-ci.yml#L37) but I'm getting the same error during opencv installation step (no issue with libavcodec):

```
...
...
-- Installing: /code/SuperBuild/install/share/opencv4/haarcascades/haarcascade_licence_plate_rus_16stages.xml
-- Installing: /code/SuperBuild/install/share/opencv4/haarcascades/haarcascade_lowerbody.xml
-- Installing: /code/SuperBuild/install/share/opencv4/haarcascades/haarcascade_profileface.xml
-- Installing: /code/SuperBuild/install/share/opencv4/haarcascades/haarcascade_righteye_2splits.xml
-- Installing: /code/SuperBuild/install/share/opencv4/haarcascades/haarcascade_russian_plate_number.xml
-- Installing: /code/SuperBuild/install/share/opencv4/haarcascades/haarcascade_smile.xml
-- Installing: /code/SuperBuild/install/share/opencv4/haarcascades/haarcascade_upperbody.xml
-- Installing: /code/SuperBuild/install/share/opencv4/lbpcascades/lbpcascade_frontalcatface.xml
-- Installing: /code/SuperBuild/install/share/opencv4/lbpcascades/lbpcascade_frontalface.xml
-- Installing: /code/SuperBuild/install/share/opencv4/lbpcascades/lbpcascade_frontalface_improved.xml
-- Installing: /code/SuperBuild/install/share/opencv4/lbpcascades/lbpcascade_profileface.xml
-- Installing: /code/SuperBuild/install/share/opencv4/lbpcascades/lbpcascade_silverware.xml
[ 64%] Completed 'opencv'
[ 64%] Built target opencv
make: *** [Makefile:84: all] Error 2
Error: building at STEP "RUN PORTABLE_INSTALL=YES GPU_INSTALL=YES bash configure.sh install": while running runtime: exit status 2
```

Full build logs are available here:
https://gitlab.com/grunlab/opendronemap/-/blob/main/build/build-odm-gpu.log

It would be really nice if this topic could be resumed.
If I can help (testing builds or whatever), please let me know.

Note: for people interested by ODM deployment on kubernetes (full stack: webodm, worker, redis, postgis, clusterodm and a pool of nodeodm), here is the way I did deploy it: https://gitlab.com/grunlab/opendronemap/-/tree/main/deploy/kustomize ... it is working like a charm (using CPU processing for the moment)

Thank you

grunlab, 2025-03-12:
Hi,

FYI, I was finally able to build the odm-gpu image on ARM64 ... I will be able to test it now :-)

In the initial issue description, @Dennis-Johnson did report 2 issues:
- The 1st issue was related to libavcodec : for me this issue is solved ... I didn't have any error at this level.
- Then a 2nd issue related to opencv : I've also faced this issue and I think I have now an explanation and a way mitigate it.

After having a deeper look into the build logs, I realize that the exit error 2 (that can make you think the issue is at opencv step level) is in fact linked to another error that occur much more before in the build process --> error during poissonrecon build step:

```
...
...[ 29%] Building CXX object CMakeFiles/pdal_base.dir/filters/OptimalNeighborhoodFilter.cpp.o
[ 19%] Building CXX object 3rdparty/carotene/hal/carotene/CMakeFiles/carotene_objs.dir/src/warp_affine.cpp.o
[ 19%] Building CXX object 3rdparty/carotene/hal/carotene/CMakeFiles/carotene_objs.dir/src/warp_perspective.cpp.o
[ 19%] Built target carotene_objs
Scanning dependencies of target ittnotify
[ 19%] Building C object 3rdparty/ittnotify/CMakeFiles/ittnotify.dir/src/ittnotify/ittnotify_static.c.o
g++_real: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[3]: *** [Makefile:186: Bin/Linux/PoissonRecon.o] Error 1
make[2]: *** [CMakeFiles/poissonrecon.dir/build.make:113: PoissonRecon/src/poissonrecon-stamp/poissonrecon-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:236: CMakeFiles/poissonrecon.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 19%] Building C object 3rdparty/ittnotify/CMakeFiles/ittnotify.dir/src/ittnotify/jitprofiling.c.o
Scanning dependencies of target ade
[ 19%] Building CXX object modules/CMakeFiles/ade.dir/__/3rdparty/ade/ade-0.1.1f/sources/ade/source/alloc.cpp.o
[ 19%] Building CXX object modules/CMakeFiles/ade.dir/__/3rdparty/ade/ade-0.1.1f/sources/ade/source/assert.cpp.o
[ 19%] Building CXX object modules/CMakeFiles/ade.dir/__/3rdparty/ade/ade-0.1.1f/sources/ade/source/check_cycles.cpp.o
...
...
<< ~2000 lines of logs before the exit error 2 >>
...
...
-- Installing: /code/SuperBuild/install/share/opencv4/haarcascades/haarcascade_licence_plate_rus_16stages.xml
-- Installing: /code/SuperBuild/install/share/opencv4/haarcascades/haarcascade_lowerbody.xml
-- Installing: /code/SuperBuild/install/share/opencv4/haarcascades/haarcascade_profileface.xml
-- Installing: /code/SuperBuild/install/share/opencv4/haarcascades/haarcascade_righteye_2splits.xml
-- Installing: /code/SuperBuild/install/share/opencv4/haarcascades/haarcascade_russian_plate_number.xml
-- Installing: /code/SuperBuild/install/share/opencv4/haarcascades/haarcascade_smile.xml
-- Installing: /code/SuperBuild/install/share/opencv4/haarcascades/haarcascade_upperbody.xml
-- Installing: /code/SuperBuild/install/share/opencv4/lbpcascades/lbpcascade_frontalcatface.xml
-- Installing: /code/SuperBuild/install/share/opencv4/lbpcascades/lbpcascade_frontalface.xml
-- Installing: /code/SuperBuild/install/share/opencv4/lbpcascades/lbpcascade_frontalface_improved.xml
-- Installing: /code/SuperBuild/install/share/opencv4/lbpcascades/lbpcascade_profileface.xml
-- Installing: /code/SuperBuild/install/share/opencv4/lbpcascades/lbpcascade_silverware.xml
[ 64%] Completed 'opencv'
[ 64%] Built target opencv
make: *** [Makefile:84: all] Error 2
Error: building at STEP "RUN PORTABLE_INSTALL=YES GPU_INSTALL=YES bash configure.sh install": while running runtime: exit status 2
```

I also found this following thread talking about this issue and explaining that at least 4GB of memory is needed to build the image:
https://community.opendronemap.org/t/odm-developer-build-always-fails-with-opencv/21237/4

I can confirm that if you don't have enough memory, the poissonrecon build step is failing (without a clear "out of memory" error message)

I don't know if it is specific to ARM64 but I've observed that poissonrecon build step is effectively consuming a lot of memory --> it consumed ~15GB of memory (so much more than 4GB)

I was able to build the image on a turing pi RK1 having 16GB of memory by doing this:
- headless OS started without any application running (to have as much as possible memory available)
- modify the ```gpu.Dockerfile``` to disable parallel build processes (the idea was to build the packages one by one in order to have the max memory available during the poissonrecon build step)

So replacing this line :
```
RUN PORTABLE_INSTALL=YES GPU_INSTALL=YES bash configure.sh install
```
by:
```
RUN PORTABLE_INSTALL=YES GPU_INSTALL=YES bash configure.sh install 1
```

Extract of configure.sh help:
```
    echo "Usage:"
    echo "bash configure.sh <install|update|uninstall|installreqs|help> [nproc]"
    echo "Subcommands:"
    echo "  install"
    echo "    Installs all dependencies and modules for running OpenDroneMap"
...
...
    echo "  help"
    echo "    Displays this message"
    echo "[nproc] is an optional argument that can set the number of processes for the make -j tag. By default it uses $(nproc)"
```

By default, configure.sh is parallelizing build processes (one build process launched per available cpu thread) --> this default configuration makes the build faster but is also consuming a bit more memory ... which was in my case an issue.

For people interested, you can pull the image from here:
- https://gitlab.com/grunlab/opendronemap/container_registry/
- registry.gitlab.com/grunlab/opendronemap/odm-gpu:3.5.5-arm64
- registry.gitlab.com/grunlab/opendronemap/odm-gpu:latest-arm64

So conclusion, to build odm-gpu on an ARM64 machine, you need at least 16GB of memory.
I think this issue can certainly be closed now !?

pierotofy, 2025-03-12:
Yes you need quite a bit of memory to build ODM 👍 thanks for posting your steps.

grunlab, 2026-02-05:
@piaoyaoi @brandonwheat 
OK, I will try to rebuild it ... keep you informed